PiSCSI/piscsi #228: 504 Gateway Time-out when trying to upload images

nsputnik, 2021-09-04:
# Info

- Which version of Pi are you using: 3B+
- Which github revision of software: version 21.05 --DEVELOPMENT BUILD
- Which board version: 2.4a
- Which computer is the RaSCSI connected to: none yet

# Describe the issue

When trying to upload .iso images that was 350MB (and planing to upload many more image files that will be what appears to be the limit of 2048MB) through the web interface I get 504 Gateway Time-out after a few seconds.  It has been suggested that I need to change some settings in nginx.conf and I tried that, not totally knowing what I was doing but then it will not run nginx.  Please let me know the exact settings I should change.

rdmark, 2021-09-08:
@nsputnik Can you provide logs leading up to the error? 
On the Pi's command line: "sudo journalctl -f -n 100"

nsputnik, 2021-09-09:
This error is no longer happening.  However, I still cannot upload .iso images.  It acts like its and I can see network traffic when I start the upload but so far only one .iso image I have uploaded is listed on the web interface.  I have tried a handful of other .iso images and they will never list under Image File Management.  And when I try to mount it it says it is mounted at the top in the green area but then it does not appear on the list.  Only 600MB.hda will appear there.

I can post my log but it is not logging anything when I try to upload or mount, just all of the info regarding startup.

By the way these are Emu sampler CD .iso files.

rdmark, 2021-09-09:
@nsputnik Can you post what you get when doing this on the command line of your Pi?
`ls -l ~/images/`

The file list in the web ui will only list files in ~/images/ that have file endings recognized as compatible with RaSCSI, so the files you download may be stored with unsupported file endings.

nsputnik, 2021-09-10:
I discovered they should be renamed .hda and not .iso.  So the images that have made it I have renamed to .hda using mv.  BUt I can't reliably upload images and get them to save through the web interface.

total 1546920
-rw-r--r-- 1 pi   pi   629145600 Sep  2 21:59 600MB.hda
-rw-r--r-- 1 root root 349624320 Sep  3 20:22 EmuEmulatorStandardsVol1.hda
-rw-r--r-- 1 root root 605261312 Sep  8 23:37 EmuEmulatorVol5WorldInstr.hda

rdmark, 2021-09-10:
@nsputnik What version of RaSCSI are you running, and how did install it? Manually or using the easyinstall.sh script?
You can find the git revision of your copy of RaSCSI at the bottom of the web ui page, to the left of 'Linux'.

The reason for asking is that there's definitely something wrong with your setup.  The latest stable version of RaSCSI should recognize '.iso' as a CD-ROM image and be able to list it as a mountable file.

If you rename an iso file to hda you will not get the outcome that you want. RaSCSI will treat it as a hard drive image and not be able to read the contents of the CD image.

nsputnik, 2021-09-10:
Hi @rdmark I used the easyinstall.sh script.
Linux raspberrypi 5.10.60-v7+ #1449 SMP Wed Aug 25 15:00:01 BST 2021 armv7l GNU/Linux

nsputnik, 2021-09-10:
I ended up installing SMB and now I can at least monitor the upload progress.  Maybe I was not waiting long enough on the web interface.

rdmark, 2021-09-11:
@nsputnik Ah that is interesting. I'm not a sampler user, but that wiki page was a good read. If block sizing is a challenge for Akai samplers, the upcoming version of the RaSCSI software will be helpful for you I think, since it allows for more flexibility in mounting images using certain block sizes. See https://github.com/akuker/RASCSI/pull/203

nsputnik, 2021-09-11:
I got the 504 Gateway Time-out error again, this time while making a 2GB .hda image through the web interface.  But I could see it was still working since I could see the file size growing through my SMB connection.

rdmark, 2021-09-15:
I will look into making the file upload code in the webapp more resilient to timeouts in this ticket.

rdmark, 2021-09-16:
One bizarre issue that I've seen just now (develop 8a3642b) is that files uploaded via the web app gets a few hundred bytes too large.  This is from Windows 10 / Firefox to the Pi.  Does not happen over ssh between the same machines.

I think it's worth introducing another method of file transfer that's more reliable.

rdmark, 2021-09-19:
At least part of the problem seems to be that the http header gets embedded in the resulting file.

> pi@rpi3b:~/images $ head foobar2.hda
> ------WebKitFormBoundaryNiHxx1Z6DggbdeWD
> Content-Disposition: form-data; name="file"; filename="foobar2.hda"
> Content-Type: application/octet-stream
> 
> 
> ------WebKitFormBoundaryNiHxx1Z6DggbdeWD--
>

rdmark, 2021-09-24:
@nsputnik These issues should have been resolved in the latest develop. Please try again when you have the chance.

closed 2021-09-24 by rdmark