PiSCSI/piscsi #232: The webapp does not generate correct NEC hard drive images

rdmark, 2021-09-09:
The rascsi-webapp does not generate correct NEC hard drive images of the Anex86 (hdi) and T98Next (nhd) types. The current implementation simply uses 'truncate' to generate an empty image file. However, the code in devices/scsihd_nec.cpp expects certain headers to be present. Some logic to modify the image file to the NEC spec is required.

rdmark, 2021-09-09:
Commenting out these two drive types in the web ui until a solution is found. https://github.com/akuker/RASCSI/pull/229

uweseimet, 2021-09-09:
@rdmark As mentioned elsewhere, this header is actually the first sector (root sector) of the drive. I cannot prove it, but it is likely that the fact that a special root sector content for NEC drives is expected is just a broken feature. I bet that just like you have to partition/format (terminology depends on the platform) a DOS/Windows/Mac/Atari drive, you have to do this for the NEC drive. As part of this procedure the software used for initializing the drive writes some geometry data into the first sector.
DOS also writes this kind of information to the root sector, by the way, and even Windows might still do it. But for SCSI drives (in contrast to old IDE drives without LBA support) the geometry is not relevant, because you access sectors by their absolute sector number, not by cylinder, head and sector on the cylinder.

rdmark, 2021-09-22:
Once piece of information I found out researching this is that the Anex86 (hdi) and T98Next (nhd) file formats are the proprietary formats of the PC98 emulators of the same names. One venue to explore would be to acquire the emulators and attempt to generate disk image files with their tooling, and then see if RaSCSI would accept them.

More and more, I believe we should not have the facility in RaSCSI itself to generate these particular file formats. From reading the Japanese RaSCSI docs, I believe the original usecase was for users of those particular PC98 emulators to be able to use images they've prepped in the emulator on a real PC98 via RaSCSI.

rdmark, 2021-09-22:
Founds a test case for the Anex86 (hdi) format: https://archive.org/details/ritokatsugathpc98
Mounting the file in the Anex86 emulator works, and it can load the software.
RaSCSI is able to attach the hdi file without error, although I of course have no PC98 machine to test it further.

rdmark, 2021-09-22:
BTW found this library for reading/writing the headers of these images: https://github.com/tsudoko/98imgtools

rdmark, 2021-09-22:
Using this version of T98-Next https://www.zophar.net/pc98/t98-next.html I was able to generate an 80MB 'ndh' file, which RaSCSI happily attached.

So now we know that at a surface level, the logic of reading the boot sector data on these proprietary image formats works as expected. Beyond that, we need someone with a real PC98 machine + SCSI adapter to test. I don't think we should invest further effort in trying to recreate such images in rascsi-web.

@uweseimet Anything else you think should be addressed here?

uweseimet, 2021-09-23:
@rdmark I don't think so.

closed 2021-09-24 by rdmark