PiSCSI/piscsi #233: CD-ROM device does not get eject command when rebooting a Mac

rdmark, 2021-09-09:
# Info

- Which version of Pi are you using: 3 B
- Which github revision of software: uweseimet_develop c163dcd 
- Which board version: 2.4a
- Which computer is the RaSCSI connected to: Mac SE / Performa 476

# Describe the issue
When a Mac is rebooted, it should be sending an eject command to removable media. With a genuine Apple CD-ROM drive on the same machines, I can confirm that a physical CD-ROM gets ejected upon reboot. However in RaSCSI, the CD-ROM media stays mounted through a reboot cycle.
Note: A shutdown of the Mac *does* lead to RaSCSI ejecting CD-ROM media.

uweseimet, 2021-09-09:
@rdmark Can you please add a trace log (of the relevant reboot sequence only), so that one can see the SCSI commands sent by the Mac?

rdmark, 2021-09-09:
@uweseimet Does this help? From attaching a device, to system booting, to reboot and system starting up again.

> Sep 09 13:05:30 rpi3b RASCSI[20641]: [2021-09-09 13:05:30.670] [info] Validating: operation=ATTACH, device id=3, unit=0, type=SCCD, device params='/home/pi/images/brix_v102.sit.iso', vendor='', product='', revision='', block size=0
> Sep 09 13:05:30 rpi3b RASCSI[20641]: [2021-09-09 13:05:30.674] [info] Executing: operation=ATTACH, device id=3, unit=0, type=SCCD, device params='/home/pi/images/brix_v102.sit.iso', vendor='', product='', revision='', block size=0
> Sep 09 13:05:30 rpi3b RASCSI[20641]: [2021-09-09 13:05:30.675] [info] Attached read-only SCCD device, ID 3, unit 0
> Sep 09 13:05:30 rpi3b RASCSI[20641]: [2021-09-09 13:05:30.704] [info] +----+----+------+-------------------------------------
> Sep 09 13:05:30 rpi3b RASCSI[20641]: [2021-09-09 13:05:30.704] [info] | ID | UN | TYPE | DEVICE STATUS
> Sep 09 13:05:30 rpi3b RASCSI[20641]: [2021-09-09 13:05:30.704] [info] +----+----+------+-------------------------------------
> Sep 09 13:05:30 rpi3b RASCSI[20641]: [2021-09-09 13:05:30.704] [info] |  3 |  0 | SCCD | /home/pi/images/brix_v102.sit.iso (WRITEPROTECT)
> Sep 09 13:05:30 rpi3b RASCSI[20641]: [2021-09-09 13:05:30.704] [info] +----+----+------+-------------------------------------
> Sep 09 13:06:03 rpi3b RASCSI[20641]: [2021-09-09 13:06:03.694] [warning] virtual void Disk::ModeSense6(SASIDEV*) Not supported MODE SENSE page $30
> Sep 09 13:06:03 rpi3b RASCSI[20641]: [2021-09-09 13:06:03.694] [warning] virtual void Disk::ModeSense6(SASIDEV*) Not supported MODE SENSE page $30
> Sep 09 13:06:03 rpi3b RASCSI[20641]: [2021-09-09 13:06:03.695] [warning] virtual void Disk::ModeSense6(SASIDEV*) Not supported MODE SENSE page $30
> Sep 09 13:06:34 rpi3b sudo[20975]:       pi : TTY=pts/0 ; PWD=/home/pi/dev/RASCSI/src/web ; USER=root ; COMMAND=/usr/local/bin/rasctl -g trace
> Sep 09 13:06:34 rpi3b sudo[20975]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
> Sep 09 13:06:34 rpi3b sudo[20975]: pam_unix(sudo:session): session closed for user root
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=3 (with device)
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] void SCSIDEV::MsgOut() ID 3
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] Message Out Phase
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] void SCSIDEV::MsgOut() ID 3
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] virtual void SCSIDEV::Receive()
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] virtual void SCSIDEV::Receive() length was != 0
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] void SCSIDEV::MsgOut() ID 3
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] virtual void SCSIDEV::Receive()
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] virtual void SCSIDEV::Receive() ctrl.phase: 10 (msgout)
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] Message code IDENTIFY $80
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] virtual void SASIDEV::Command() Command Phase
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] virtual void SCSIDEV::Execute() Execution phase command $1E
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $1E
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] virtual bool SCSICD::Dispatch(SCSIDEV*) Calling base class for dispatching $1E
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing PreventAllowMediumRemoval ($1E)
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [debug] bool Disk::CheckReady() Disk is ready!
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] void SASIDEV::Status() Status phase
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] void SASIDEV::Status() Status Phase $00
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] void SASIDEV::MsgIn() Starting Message in phase
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] void SASIDEV::MsgIn() Transitioning to Send()
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] void SASIDEV::MsgIn() Transitioning to Send()
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.265] [trace] virtual void SCSIDEV::BusFree() Bus free phase
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.266] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=3 (with device)
[... cut ...]
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.393] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.393] [trace] void SASIDEV::MsgIn() Transitioning to Send()
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.393] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
> Sep 09 13:07:00 rpi3b RASCSI[20641]: [2021-09-09 13:07:00.393] [trace] virtual void SCSIDEV::BusFree() Bus free phase

uweseimet, 2021-09-09:
@rdmark Yes, this helps, but I'm afraid I need yet another log of the same scenario, based on a logging update in the latest uweseimet_develop branch.

rdmark, 2021-09-09:
@uweseimet using 5e266e2

> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.442] [trace] virtual void SCSIDEV::Receive() length was != 0
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.442] [trace] void SCSIDEV::MsgOut() ID 3
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.442] [trace] virtual void SCSIDEV::Receive()
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.442] [trace] virtual void SCSIDEV::Receive() ctrl.phase: 10 (msgout)
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.442] [trace] Message code IDENTIFY $80
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.442] [trace] virtual void SASIDEV::Command() Command Phase
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.442] [trace] virtual void SCSIDEV::Execute() Execution phase command $28
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.442] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.442] [trace] virtual bool SCSICD::Dispatch(SCSIDEV*) Calling base class for dispatching $28
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.442] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Read10 ($28)
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.442] [debug] virtual void Disk::Read10(SASIDEV*) READ(10) command record=$   123EB blocks=1
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.442] [debug] bool Disk::CheckReady() Disk is ready!
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.442] [debug] virtual int Disk::Read(const DWORD*, BYTE*, uint64_t)
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.442] [debug] bool Disk::CheckReady() Disk is ready!
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.442] [trace] BOOL DiskTrack::Read(BYTE*, int) const reading sector: 235
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.442] [trace] void Disk::Read(SASIDEV*, uint64_t) ctrl.length is 2048
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.442] [trace] void SASIDEV::DataIn() Going into Data-in Phase
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.442] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 2048
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.444] [trace] virtual void SCSIDEV::Send() Move to next phase datain (6)
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.444] [trace] void SASIDEV::Status() Status phase
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.444] [trace] void SASIDEV::Status() Status Phase $00
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.444] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.445] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.445] [trace] void SASIDEV::MsgIn() Starting Message in phase
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.445] [trace] void SASIDEV::MsgIn() Transitioning to Send()
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.445] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.445] [trace] void SASIDEV::MsgIn() Transitioning to Send()
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.445] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
> Sep 09 13:40:56 rpi3b RASCSI[29179]: [2021-09-09 13:40:56.445] [trace] virtual void SCSIDEV::BusFree() Bus free phase
> Sep 09 13:40:57 rpi3b RASCSI[29179]: [2021-09-09 13:40:57.207] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=3 (with device)
> Sep 09 13:40:57 rpi3b RASCSI[29179]: [2021-09-09 13:40:57.207] [trace] void SCSIDEV::MsgOut() ID 3
> Sep 09 13:40:57 rpi3b RASCSI[29179]: [2021-09-09 13:40:57.207] [trace] Message Out Phase
> Sep 09 13:40:57 rpi3b RASCSI[29179]: [2021-09-09 13:40:57.207] [trace] void SCSIDEV::MsgOut() ID 3
> Sep 09 13:40:57 rpi3b RASCSI[29179]: [2021-09-09 13:40:57.207] [trace] virtual void SCSIDEV::Receive()
> Sep 09 13:40:57 rpi3b RASCSI[29179]: [2021-09-09 13:40:57.207] [trace] virtual void SCSIDEV::Receive() length was != 0
> Sep 09 13:40:57 rpi3b RASCSI[29179]: [2021-09-09 13:40:57.207] [trace] void SCSIDEV::MsgOut() ID 3
> Sep 09 13:40:57 rpi3b RASCSI[29179]: [2021-09-09 13:40:57.207] [trace] virtual void SCSIDEV::Receive()
> Sep 09 13:40:57 rpi3b RASCSI[29179]: [2021-09-09 13:40:57.207] [trace] virtual void SCSIDEV::Receive() ctrl.phase: 10 (msgout)
> Sep 09 13:40:57 rpi3b RASCSI[29179]: [2021-09-09 13:40:57.207] [trace] Message code IDENTIFY $80
> Sep 09 13:40:57 rpi3b RASCSI[29179]: [2021-09-09 13:40:57.207] [trace] virtual void SASIDEV::Command() Command Phase
> Sep 09 13:40:57 rpi3b RASCSI[29179]: [2021-09-09 13:40:57.207] [trace] virtual void SCSIDEV::Execute() Execution phase command $1E
> Sep 09 13:40:57 rpi3b RASCSI[29179]: [2021-09-09 13:40:57.207] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $1E
> Sep 09 13:40:57 rpi3b RASCSI[29179]: [2021-09-09 13:40:57.207] [trace] virtual bool SCSICD::Dispatch(SCSIDEV*) Calling base class for dispatching $1E
> Sep 09 13:40:57 rpi3b RASCSI[29179]: [2021-09-09 13:40:57.207] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing PreventAllowMediumRemoval ($1E)
[... cut ...]
> Sep 09 13:40:57 rpi3b RASCSI[29179]: [2021-09-09 13:40:57.313] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
> Sep 09 13:40:57 rpi3b RASCSI[29179]: [2021-09-09 13:40:57.313] [trace] void SASIDEV::MsgIn() Transitioning to Send()
> Sep 09 13:40:57 rpi3b RASCSI[29179]: [2021-09-09 13:40:57.313] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
> Sep 09 13:40:57 rpi3b RASCSI[29179]: [2021-09-09 13:40:57.313] [trace] virtual void SCSIDEV::BusFree() Bus free phase

uweseimet, 2021-09-09:
@rdmark I just updated the load/eject handling. Can you please re-try with the latest codebase? If it still fails, please provide a log once more.

rdmark, 2021-09-09:
@uweseimet Testing with 688255b I still don't see the device getting ejected on reboot. Also attaching logs from shutdown for comparison.

REBOOT
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.201] [trace] virtual void SCSIDEV::Receive() length was != 0
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.201] [trace] void SCSIDEV::MsgOut() ID 3
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.201] [trace] virtual void SCSIDEV::Receive()
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.201] [trace] virtual void SCSIDEV::Receive() ctrl.phase: 10 (msgout)
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.201] [trace] Message code IDENTIFY $80
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.201] [trace] virtual void SASIDEV::Command() Command Phase
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.201] [trace] virtual void SCSIDEV::Execute() Execution phase command $28
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.201] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.201] [trace] virtual bool SCSICD::Dispatch(SCSIDEV*) Calling base class for dispatching $28
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.201] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Read10 ($28)
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.201] [debug] virtual void Disk::Read10(SASIDEV*) READ(10) command record=$     6FF blocks=1
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.201] [debug] bool Disk::CheckReady() Disk is ready!
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.201] [debug] virtual int Disk::Read(const DWORD*, BYTE*, uint64_t)
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.201] [debug] bool Disk::CheckReady() Disk is ready!
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.201] [trace] BOOL DiskTrack::Read(BYTE*, int) const reading sector: 255
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.201] [trace] void Disk::Read(SASIDEV*, uint64_t) ctrl.length is 2048
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.201] [trace] void SASIDEV::DataIn() Going into Data-in Phase
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.202] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 2048
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.204] [trace] virtual void SCSIDEV::Send() Move to next phase datain (6)
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.204] [trace] void SASIDEV::Status() Status phase
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.204] [trace] void SASIDEV::Status() Status Phase $00
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.204] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.204] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.204] [trace] void SASIDEV::MsgIn() Starting Message in phase
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.204] [trace] void SASIDEV::MsgIn() Transitioning to Send()
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.204] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.204] [trace] void SASIDEV::MsgIn() Transitioning to Send()
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.204] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.204] [trace] virtual void SCSIDEV::BusFree() Bus free phase
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.868] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=3 (with device)
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.868] [trace] void SCSIDEV::MsgOut() ID 3
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.868] [trace] Message Out Phase
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.868] [trace] void SCSIDEV::MsgOut() ID 3
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.868] [trace] virtual void SCSIDEV::Receive()
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.868] [trace] virtual void SCSIDEV::Receive() length was != 0
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.868] [trace] void SCSIDEV::MsgOut() ID 3
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.868] [trace] virtual void SCSIDEV::Receive()
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.868] [trace] virtual void SCSIDEV::Receive() ctrl.phase: 10 (msgout)
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.868] [trace] Message code IDENTIFY $80
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.868] [trace] virtual void SASIDEV::Command() Command Phase
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.868] [trace] virtual void SCSIDEV::Execute() Execution phase command $1E
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.868] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $1E
> Sep 09 15:52:44 rpi3b RASCSI[17707]: [2021-09-09 15:52:44.868] [trace] virtual bool SCSICD::Dispatch(SCSIDEV*) Calling base class for dispatching $1E
[... cut ...]
> Sep 09 15:56:30 rpi3b RASCSI[17707]: [2021-09-09 15:56:30.938] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
> Sep 09 15:56:30 rpi3b RASCSI[17707]: [2021-09-09 15:56:30.938] [trace] void SASIDEV::MsgIn() Transitioning to Send()
> Sep 09 15:56:30 rpi3b RASCSI[17707]: [2021-09-09 15:56:30.938] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
> Sep 09 15:56:30 rpi3b RASCSI[17707]: [2021-09-09 15:56:30.938] [trace] virtual void SCSIDEV::BusFree() Bus free phase

rdmark, 2021-09-09:
One observation in the current state of the code, is that at the moment the removable media drive gets accessed by the system (when the driver loads, I think) it gets the LOCKED property in RaSCSI.  This did not happen until very recently.
In parallel I was playing around with Iomega removable devices, trying to use a virtual Zip drive. I got as far as the Iomega driver loading and recognizing the device as a Zip drive, and triggering the OS formatting modal for a 100MB zip disk. However since RaSCSI locked the device, the formatting fails immediately.

uweseimet, 2021-09-10:
@rdmark rasctl does not support unlocking:

```
CMD := {attach|detach|insert|eject|protect|unprotect|show}
```

I fixed issues with the load/eject handling, please re-try.

Regarding the ZIP, does the Mac need a special driver? As long as one does not want to use special ZIP features (which are not supported and also not really useful for RaSCSI) the ZIP is just a standard removable media drive.

rdmark, 2021-09-10:
@uweseimet Thanks, I'll keep testing the ejection functionality today after work.

I have to use the Iomega driver for the Mac treat the Zip device as a removable storage device. Without it, it gets identified as a non-removable hard drive device (despite the SCRM type) and can be initialized as such by Apple's SCSI drive formatting tool. Anyhow, I'll keep playing around with this and see if there's some other way to get the Mac to treat a SCRM device as removable. It's more of a minor use case and not really in the scope of this ticket. I brought it up since both SCRM and SCCD devices get inexplicably (?) Locked when the system starts accessing them.

uweseimet, 2021-09-10:
@rdmark I see. Looks as if the regular Mac driver does not evaluate the bit in the INQUIRY data that says that a device supports removable media. The ZIP (like any other removable media drive, e.g. SyQuest, but also MO, LS-120, CD-ROM, DVD-RAM...) all set this bit, because that's what the standard demands. Since the Mac knows (at least I think it does) that MOs are removable it seems as if the standard driver looks at the device type to make a decision, not at the INQUIRY data. If this is the case, this is not very intelligent.

Is there any source where I can download this Iomega driver?

rdmark, 2021-09-10:
@uweseimet This is the driver that I'm testing with right now: https://macintoshgarden.org/apps/iomegatools5

uweseimet, 2021-09-10:
Just for the record: ZIP and JAZ use proprietary SCSI commands (opcodes $06 and $0C) to manage the ZIP/JAZ specific password-based write protection. From the rascsi perspective this proprietary feature is quite useless, of course. The software also wants to access the unsupported mode pages $2F and $30.

rdmark, 2021-09-10:
@uweseimet Strangely enough, with df8a70 the CD-ROM doesn't get ejected even upon Shutdown anymore...  Tested on both my Macs with various CD-ROM drivers... If Macs have a non-standard way to signal the eject upon system shutdown I don't think it's end of the world that removable media stays mounted though.  It's better than having bugs like the one Akuker found last week where media ejected by a reboot led to an unusable 'empty' drive in RaSCSI...

uweseimet, 2021-09-10:
@rdmark The logging of previous versions (commits) was wrong. The current logging should be correct and reflect what the Mac really does. I don't think that it is possible to do something special in order to eject, because there is only one way to eject media. And this is the way the Finder uses.

Is there any difference now to how a real CD-ROM drive behaves?

rdmark, 2021-09-10:
@uweseimet Yeah that's the thing, when I hook up a physical Apple branded CD-ROM drive to the same machine/system/driver, any inserted CD-ROM gets ejected the moment you issue the Reboot command in the OS.

uweseimet, 2021-09-10:
@rdmark OK. I will double-check this again, most likely tomorrow.

rdmark, 2021-09-10:
@uweseimet Maybe I haven't given you big enough snippets of logs and some context is left out?  Let me grab one using the latest code.

uweseimet, 2021-09-10:
@rdmark I assume that my Mac does exactly the same thing as yours, so no need for new logs, at least not at the moment.

uweseimet, 2021-09-10:
I just checked this issue once more today ;-). For the testing branch "testing_eject" I set logging for START/STOP UNIT to level error, so that I only see these messages in the log.
This is what I get when shutting down my Mac:

```
[2021-09-10 20:33:49.047] [error] $00  START: 0  LOEJ: 0
[2021-09-10 20:33:49.047] [error] Stopping unit
[2021-09-10 20:33:49.162] [error] $03  START: 1  LOEJ: 1
[2021-09-10 20:33:49.162] [error] Loading medium
```

The hex value is byte 4 of the command block, with the START and LOEJ bits. The specification says:

```
If the load eject (LOEJ) bit is set to zero and the POWER CONDITION field is
set to zero, then the logical unit shall take no action regarding loading or
ejecting the medium. If the LOEJ bit is set to one and the POWER CONDITION
field is set to zero, then the logical unit shall unload the medium if the  START
bit is set to zero. If the LOEJ bit is set to one, the POWER CONDITION field is
set to zero, and the START bit is set to one, then the logical unit shall load the
medium.
```

RaSCSI does not care for the POWER CONDITIION field, so only the other two fields are relevant.

This means that for EJECT the START bit must be 0 and the LOEJ bit 1. Correct me if I am wrong. The Mac does not send this command.

With a reboot instead of a shutdown only one command is sent:

```
[2021-09-10 20:32:41.053] [error] $00  START: 0  LOEJ: 0
[2021-09-10 20:32:41.053] [error] Stopping unit
```

Just a STOP UNIT in this case.

@rdmark All in all the log shows that my Mac never sends an EJECT. Would be great if you could verify my results with the testing_eject branch. The log on error level should reveal everything.
Can it be that it makes a difference whether the inserted CD is a bootable CD or not? In my case it was just a random data CD, with no Mac specific data on it, with SCSI ID 5.

rdmark, 2021-09-11:
Thanks for the technical breakdown, Uwe. I'll run a few tests with testing_eject first thing tomorrow morning. I've been testing with random non-bootable disks, both in the physical Apple drive and with RaSCSI.

uweseimet, 2021-09-11:
@rdmark I tried to run tests with the real CD-ROM drive in my Performa, but even though the drive is detected during a SCSI device scan MacOS does not seem to support it. I'm not sure if this is the original drive (it's a Matsushita one) or whether it is a third-party drive that requires additional drivers. I also tried an external Plextor CD-ROM drive, but this one was also not accepted by MacOS, even though it is displayed by a SCSI device scan.

rdmark, 2021-09-11:
@uweseimet Have you tried using the various drivers described here? https://github.com/akuker/RASCSI/wiki/Drive-Setup#mounting-cd-iso-or-mo-images

rdmark, 2021-09-11:
@uweseimet Regarding your eject problem, I have not been encountering that on my Macs when using RaSCSI. When I eject a CD (by putting it in the trash :) ) the icon disappears from the desktop, and I can insert another CD image. What system version & CD driver have you been using?

rdmark, 2021-09-11:
@uweseimet Testing with the testing_eject branch, I made a little bit of a breakthrough:  The commands that are sent (and behaviors) on reboot and shutdown are completely driver dependent. Test results from my Performa 476:

> APPLE CD-ROM
> REBOOT
> Result: not ejected
> 
> [2021-09-11 05:05:59.081] [error] $00  START: 0  LOEJ: 0
> [2021-09-11 05:05:59.081] [error] Stopping unit
> 
> APPLE CD-ROM
> SHUTDOWN
> Result: not ejected
> 
> [2021-09-11 05:06:52.671] [error] $00  START: 0  LOEJ: 0
> [2021-09-11 05:06:52.671] [error] Stopping unit
> [2021-09-11 05:06:52.787] [error] $03  START: 1  LOEJ: 1
> [2021-09-11 05:06:52.787] [error] Loading medium
> 
> CD SUNRISE
> REBOOT
> Result: ejected
> [2021-09-11 05:13:08.334] [error] $02  START: 0  LOEJ: 1
> [2021-09-11 05:13:08.335] [error] Ejecting medium
> 
> SHUTDOWN
> Result: ejected
> [2021-09-11 05:19:21.883] [error] $02  START: 0  LOEJ: 1
> [2021-09-11 05:19:21.883] [error] Ejecting medium
> 
> CD/DVD TUNEUP
> REBOOT
> Result: not ejected
> [no output]
> 
> SHUTDOWN
> Result: not ejected
> [no output]
> 
> RAPIDCD
> REBOOT
> Result: ejected
> [2021-09-11 05:22:48.086] [error] $02  START: 0  LOEJ: 1
> [2021-09-11 05:22:48.086] [error] Ejecting medium
> 
> SHUTDOWN
> Result: ejected
> [2021-09-11 05:23:47.863] [error] $02  START: 0  LOEJ: 1
> [2021-09-11 05:23:47.863] [error] Ejecting medium

uweseimet, 2021-09-11:
@rdmark This confirms that the command implementation as such is correct, and if a CD is not ejected the reason is not that the eject command was not properly executed, but that simply no eject command was sent.
Looks as if this issue is resolved? The uweseimet_develop branch contains the same code, but without the special logging, which I just added for testing.

Regarding my system, I am using the regular Mac drivers, i.e. I did not install any special CD-ROM drivers. I did not move my drive into the trash, but used the eject option in the Finder's menu.

rdmark, 2021-09-11:
@uweseimet Yeah let me close this issue. Thanks for all your efforts troubleshooting this one. I guess Apple must have some non-standard mechanism for telling its 'genuine' CD-ROM drives to eject in these scenarios.

Do you want help figuring out what's going on with your Mac there?  Macs definitely have this weird handling of removable storage media where it keeps a 'shadow' of ejected disks file systems in memory in certain cases, such as when an application on that disk is still running in memory. You get that modal as soon as the system or running application tries to access data on the ejected disk again. This was particularly common/useful on pre-hard drive Macs where the system software ran off of floppy. I remember juggling floppies in and out of a Macintosh Plus all the time for the most basic tasks back in the days. :)

uweseimet, 2021-09-11:
@rdmark My guess is that this is not a non-standard mechanism, but that the Mac driver "knows" something about their own CD-ROMs (e.g. an extended status), and that as a result the eject handling is different. But what this special thing might be would be hard to find. It could also be an issue in how RaSCSI handles the drive status in general, prompting the Mac driver to react in a different way than usual. I would not be surprised if the issue disappears with potential further refinements of the drive status handling.
At least we verified that the actual "Eject" implementation is very likely correct.

Thank you for offering to help me with my Mac issue, but I also think that this may be caused by the Mac caching information in memory. This is not an urgent issue for me, and I guess sooner or later I'll figure out the details.
The Finder appears to cache quite a lot. When renaming a file on a RaSCSI drive, for instance, it takes quite some time until the WRITE SCSI command is sent.

closed 2021-09-11 by rdmark