PiSCSI/piscsi #235: The RaSCSI logs do not have entries for disk image creation/deletion

stinkerton18, 2021-09-13:
# Info

- Raspberry Pi 3A+, 3B
- Latest and Pull request #229
- v2.3 Full Featured
- N/A

When using the web ui to create or delete an image, there is no feedback logged on the operation itself or whether it completed successfully. In the case of the RPi's file system becoming read only or full, the user would not receive any indication why the image creation or deletion failed. 

If logging could be added along the lines of the ones present when mounting or dismounting an image, that would satisfy this issue.

uweseimet, 2021-09-15:
@rdmark Would you agree that the image creation should be a part of rascsi, instead of being done by a client? I don't know if the web UI has alogfile of its own, but if rascsi were creating the images it could simply log any issues to the regular logfile.

rdmark, 2021-09-15:
@uweseimet I think this is a great feature to have in rascsi.  I don't know if you've kept up with #rascsi-dev on Discord, but I've been working on expanding image creation functionality of the frondend, together with a 'sidecar' configuration file format that goes with the image so that you can create, say, a 100MB "Iomega Zip" image which when attached will get the right product and block size properties loaded automatically from the sidecar file.  I have a rudimentary implementation in place in the webapp code in develop, and would love your thoughts on this approach. In particular, do you think it makes sense to do more of this on the rascsi backend, perhaps the ability to pass the config file as a parameter?

uweseimet, 2021-09-15:
@rdmark IMHO rascsi should essentially provide more or less basic functionality, typically atomic operations. More complex functions (e.g. combinations of different rasci operations) should better be handled on the client side.

rdmark, 2021-09-15:
@uweseimet Agreed this is the correct design principle.

uweseimet, 2021-09-16:
@rdmark Please note that I updated the protobuf interface in the feature_image_creation branch to use maps with named parameters instead of parameter lists. The initial approach is too error prone and also less explicit when you have operations that expect several parameters. In addition, the error handling can be more precise.
Please see the .proto file (PbOperation message) for the parameter names supported by each operation. The functionality as such has not changed.

rdmark, 2021-09-16:
@uweseimet I think that would be a cleaner solution, and make better use of parameters which is a bit underutilized right now.

uweseimet, 2021-09-16:
@rdmark One should not forget, though, that the parameters are not type-safe, in constrast to explicit protobuf message fields.

rdmark, 2021-09-16:
@uweseimet Well noted, I'll make sure the Python code does extra thorough type validation when using the parameters. (The woes of a weakly typed language.)

uweseimet, 2021-09-17:
@rdmark After sleeping a night on it I decided not to convert more explicit parameters to named parameters, due to the loss of type-safety, which also results in more and more string to integer (and back) conversions in the code. If you convert too many parameters to strings you end up with a string-based free-text interface as it was before using protobuf ;-).

Note that I added two interface operations for making image files read-only or writable. I added these because the R/W status of an image file is checked in the rascsi code, so it should be possible to control these attributes via the interface.

After some more testing I will create a PR for the changes discussed here.

rdmark, 2021-09-17:
@uweseimet Sounds good, then I don't have to rewrite as much of the web ui code.  I'll start integrating your feature branch in my feature branch now to start the integration of the new protobuf and image creation functionality.

rdmark, 2021-09-17:
@uweseimet BTW did you have a particular usecase in mind when you implemented the 'read_only' property for created images? I'm just wondering how it could be leveraged in the UI, if there is a common use case for protecting your images.

uweseimet, 2021-09-17:
@rdmark rasci (also the old code) checks image file permissions, and because the permissions are relevant from the rascsi perspectiveI added the means to modify them.

rdmark, 2021-09-17:
> For attaching/inserting image files the parameter name is "file", for SCBR/SCDP interfaces it is "interfaces". Please see the .proto file (PbOperation message) for the other parameter names for each operation.

@uweseimet  I think the proto file is missing inline documentation for the "file" and "interfaces" named parameters (for ATTACH in particular, and I assume INSERT should accept the "file" parameter as well)

uweseimet, 2021-09-17:
@rdmark Indeed, thank you. I just updated the comments.

rdmark, 2021-09-17:
@uweseimet Thanks for the quick fix!

Now before I go too far into the rabbit hole, is my reading of the code correct in that the named parameter format is any number of strings containing name value pairs with equal signs as separator?

I.e. "file=harddrive.hda"

uweseimet, 2021-09-17:
@rdmark It's a protobuf map (key/value pairs), see the definition in the .proto file.

rdmark, 2021-09-17:
@uweseimet I see. I'm trying to wrap my head around the Python representation of protobuf maps, but I'm really struggling so far. The documentation isn't much help https://developers.google.com/protocol-buffers/docs/reference/python-generated#map-fields

I'll keep trying for a bit longer, but I very much prefer the previous implementation. ;)

uweseimet, 2021-09-17:
@rdmark Looks like a good opportunity to learn something new about python and protobuf ;-). It was the same for me with C++ and Kotlin, where I had never used protobuf before. (Before only with Java.)

rdmark, 2021-09-17:
Alright, alright, I'll keep trying. I think I figured out at least the basic usecase. You're supposed to use the name of the parameter as an "undefined key" of the map dictionary, and then assign the value to that.

uweseimet, 2021-09-17:
@rdmark I'm afraid I cannot really comment on the Python mapping, because I hardly know Python. How some protobuf features are mapped may depend a lot on the programming language used. The C++ mapping is quite complicated, mainly because of memory managment requirements. The Kotlin mapping is disappointing, because it does not support protobuf enums, which means it's useless for me. I had to switch to the Java mapping instead, which can also be used with Kotlin without issues.

rdmark, 2021-09-18:
@uweseimet I think there's a bug now where if you attach a daynaport device with rasctl it gets the 'file' parameter instead of the 'interfaces' parameter.  Are you seeing the same behavior?  Using the protobuf interface directly is fine, so I assume the bug is in rasctl.

> pi@rpi3b:~ $ rasctl -i 5 -c attach -t scdp -f wlan0:10.10.20.1/24
> pi@rpi3b:~ $ rasctl -i 5 -c show
>   5:0  SCDP  Dayna:SCSI/Link:1.4a  file=wlan0:10.10.20.1/24

rdmark, 2021-09-18:
Finished implementing file creation and deletion using the new interface. PR in https://github.com/akuker/RASCSI/pull/242

uweseimet, 2021-09-18:
@rdmark Fixed.

rdmark, 2021-09-18:
@uweseimet Thanks for the fix! I've tested it in my environment. What do you say we get this merged to develop now?

closed 2021-09-19 by uweseimet