PiSCSI/piscsi #236: easyinstall.sh assumes code is checked out in /home/pi/RASCSI

rdmark, 2021-09-15:
To support easy management of multiple RASCSI working directories on one system, improve easyinstall.sh to detect the directory it is in, and adjust paths dynamically in the script.

Current suggested workaround is to keep checkouts elsewhere, and symlink to /home/pi/RASCSI

akuker, 2021-09-15:
Another alternative would be to 'install' the web service to a different location (like /usr/local or something) Then it will always be at a fixed location. (Similar to how the rascsi/rasctl tools are installed)

uweseimet, 2021-09-15:
Wouldn't it make sense to make this even more flexible by using GNU autoconf or cmake?

rdmark, 2021-09-16:
I like the autoconf idea! In addition, it could take over some of the operations that the easyinstall.sh script does right now, such as compiling the protobuf library.

rdmark, 2021-09-16:
Alternatively, use Python's native package management. Some ideas:
https://packaging.python.org/tutorials/packaging-projects/
https://docs.python.org/3/distutils/setupscript.html#

Perhaps skip the step of publishing it on PyPI though. ;)

uweseimet, 2021-09-16:
@rdmark autoconf only configures, but does not create parts of the final executables. That's still managed by the Makefile created by autoconf. I don't know how autoconf relates to python, I only know it as a tool for C/C++ based projects.

rdmark, 2021-09-26:
A quick test suggests that there is nothing hindering rascsi-web from running from a directory such as /usr/local/rascsi-web/
Let me pull together a PR with the install script updates needed..

karrots, 2021-10-03:
Generally `/opt/` is more accepted these days for this kind of thing. So `/opt/rascsi-web`

rdmark, 2021-10-06:
@karrots Good progress improving on the script here! Do you plan to do some more work on this at this time? We're getting ready to cut a new release shortly, and it would be great to get these improvements merged before that.

karrots, 2021-10-06:
I have been testing for bugs. Mostly dealing with forgetting sudo or
adjusting permissions that didn't need to be special before. Not sure I'll
get much time until the weekend to test more.

rdmark, 2021-10-06:
@karrots Do you mind if I cherry-pick the improvements that are not to do with installing the webapp to /opt ? The latter needs more testing I think so let's aim for the next release, while the other tweaks and improvements would be helpful right now.

rdmark, 2021-10-07:
@karrots See my "fork" of your PR here: https://github.com/akuker/RASCSI/pull/305

rdmark, 2021-10-08:
@karrots I realized I could do it myself, so I made those tweaks and merged the PR.

karrots, 2021-10-08:
Ok sounds good.

rdmark, 2021-11-07:
PR ready to address the portability of easyinstall and rascsi-web: https://github.com/akuker/RASCSI/pull/426

closed 2021-11-09 by rdmark