PiSCSI/piscsi #238: The rasctl -f can be used to pass arbitrary parameters

rdmark, 2021-09-15:
The manpage and cursory reading of the code suggests that rasctl -f is expected to take only a file path, but in reality can take an arbitrary device parameter. Ex:

> pi@rpi3b:~/RASCSI $ rasctl -i 6 -c attach -t scdp -f wlan1:10.10.20.1/24
> pi@rpi3b:~/RASCSI $ rasctl -i 6 -c show
>   6:0  SCDP  Dayna:SCSI/Link:1.4a  wlan1:10.10.20.1/24

rdmark, 2021-09-15:
@uweseimet I couldn't see another rasctl option that *should* be used to pass arbitrary parameters, so it's actually quite helpful that you can do the above. However, there's a gap with the manpage at the very least.

uweseimet, 2021-09-15:
@rdmark A filename is just a string, and any string is accepted. Filenames can have any character in them, at least on most platforms. I don't see how you would be able to filter anything. Are you thinking of something in particular that should be done?

rdmark, 2021-09-15:
@uweseimet I think the current implementation is good, so perhaps consider a documentation update? Call out that -f takes a parameter string that can be a file name or something else, depending on the device type. Thoughts?

uweseimet, 2021-09-15:
@rdmark Yes, I agree. I just updated the manpage and help for rasctl. rasctl and rascti are soon running out of available letters for options ;-).

rdmark, 2021-09-15:
Thanks, this resolves my concern. I'll close out the ticket once your code gets merged.

closed 2021-09-19 by uweseimet