PiSCSI/piscsi #245: Big files make the service crash, value too large for datatype

Aapjedotinfo, 2021-09-19:
# Info

- Which version of Pi are you using: Pi4
- Which github revision of software: develop
- Which board version: 2.4a fullspec
- Which computer is the RaSCSI connected to: Macintosh Classic II but its turned off for this case.

Develop branch crashes on a big file when opening the web interface:
`Sep 19 23:15:09 raspberrypi RASCSI[1827]: terminate called after throwing an instance of 'std::filesystem::__cxx11::filesystem_error'
Sep 19 23:15:09 raspberrypi RASCSI[1827]:   what():  filesystem error: cannot get file size: Value too large for defined data type [/home/pi/images/powermac.hda]
Sep 19 23:15:09 raspberrypi systemd[1]: rascsi.service: Main process exited, code=killed, status=6/ABRT
Sep 19 23:15:09 raspberrypi RASCSIWEB[1894]: ERROR:root:The response from RaSCSI did not contain a protobuf header.
Sep 19 23:15:09 raspberrypi systemd[1]: rascsi.service: Failed with result 'signal'.
Sep 19 23:15:09 raspberrypi systemd[1]: rascsi.service: Service RestartSec=100ms expired, scheduling restart.
Sep 19 23:15:09 raspberrypi systemd[1]: rascsi.service: Scheduled restart job, restart counter is at 13.
Sep 19 23:15:09 raspberrypi systemd[1]: Stopped RaSCSI service.
Sep 19 23:15:09 raspberrypi systemd[1]: Started RaSCSI service.
Sep 19 23:15:09 raspberrypi RASCSI[1911]: SCSI Target Emulator RaSCSI(*^..^*) version 21.99 --DEVELOPMENT BUILD-- (Sep 19 2021, 22:58:27)
Sep 19 23:15:09 raspberrypi RASCSI[1911]: Powered by XM6 TypeG Technology / Copyright (C) 2016-2020 GIMONS
Sep 19 23:15:09 raspberrypi RASCSI[1911]: Connect type : FULLSPEC
Sep 19 23:15:09 raspberrypi RASCSI[1911]: [2021-09-19 23:15:09.827] [info] No images currently attached.
Sep 19 23:15:09 raspberrypi RASCSI[1911]: No images currently attached.
`

File:
`-rw-r--r-- 1 root root 2621440000 Sep 18 20:29 powermac.hda`

It was created with the stable version.

uweseimet, 2021-09-20:
@Aapjedotinfo I cannot reproduce this issue withthe current develop branch (commit 9795c55):

```
>ls -al /home/us/test.hds
-rw-rw-r-- 1 us us 2621440000 Sep 20 10:21 /home/us/test.hds

sudo ./bin/fullspec/rascsi -id 0 ~/test.hds

SCSI Target Emulator RaSCSI(*^..^*) version 21.99 --DEVELOPMENT BUILD-- (Sep 20 2021, 10:20:57)
Powered by XM6 TypeG Technology / Copyright (C) 2016-2020 GIMONS
Connect type : FULLSPEC
[2021-09-20 10:24:24.274] [info] Validating: operation=ATTACH, device id=0, unit=0, type=UNDEFINED, device params='file=/home/us/test.hds', vendor='', product='', revision='', block size=0
[2021-09-20 10:24:24.276] [info] Executing: operation=ATTACH, device id=0, unit=0, type=UNDEFINED, device params='file=/home/us/test.hds', vendor='', product='', revision='', block size=0
[2021-09-20 10:24:24.278] [info] Attached SCHD device, ID 0, unit 0
[2021-09-20 10:24:24.279] [info] +----+----+------+-------------------------------------
[2021-09-20 10:24:24.280] [info] | ID | UN | TYPE | DEVICE STATUS
[2021-09-20 10:24:24.282] [info] +----+----+------+-------------------------------------
[2021-09-20 10:24:24.283] [info] |  0 |  0 | SCHD | /home/us/test.hds
[2021-09-20 10:24:24.284] [info] +----+----+------+-------------------------------------
+----+----+------+-------------------------------------
| ID | UN | TYPE | DEVICE STATUS
+----+----+------+-------------------------------------
|  0 |  0 | SCHD | /home/us/test.hds
+----+----+------+-------------------------------------
```

Please run rascsi from the command line (without the web interface). Also ensure that your Raspberry OS is up to date.

rdmark, 2021-09-20:
@uweseimet Can you try doing 'rasctl -s' with the large image present and see if that reproduces the issue?
When I was helping them with this crash yesterday, it seems to happen the moment the web ui sends a IMAGE_FILES_INFO command to rascsi to get the list of valid image files to show in the ui.  I believe 'rasctl -s' is the closest thing on the CLI.

uweseimet, 2021-09-20:
```
Attached devices:
  0:0  SCHD  RaSCSI:SCSI HD 2500 MB:2199  512 bytes per sector  2621440000 bytes capacity  /home/us/test.hda
```

Also no issues. The file size handling is not some custom code, but part of the GNU C++ library. It's unlikely that the library has a bug.

uweseimet, 2021-09-21:
@Aapjedotinfo Please provide details on your environment (see above), and how to reproduce this issue.

uweseimet, 2021-09-21:
@Aapjedotinfo Yes, the web interface fails, but the error is caused by something else. I'm afraid without being able to reproduce this issue I cannot do anything.
@rdmark Can you reproduce this? Please also use rasctl and not the web UI.

Aapjedotinfo, 2021-09-21:
I might be able to share the hda file.

rdmark, 2021-09-21:
I'll try to set up an environment to test this later today.

uweseimet, 2021-09-21:
@Aapjedotinfo Thank you, but most likely this will not help, because the issue appears not to be related to the file content. There appears to be an issue with getting the file size, not the file content.

uweseimet, 2021-09-21:
By the way, see https://unix.stackexchange.com/questions/355953/ubuntu-16-04-lts-value-too-large-for-defined-data-type for similar issues. This is why I asked for information about the OS and the compiler.

rdmark, 2021-09-21:
@uweseimet I could reproduce the crash in my setup too, with a 4096 MB image file, and a completely freshly installed Raspbian system & RaSCSI develop branch.
> Sep 21 15:47:35 rpi3b RASCSI[29385]:   what():  filesystem error: cannot get file size: Value too large for defined data type [/home/pi/images/4g.hda]

Let me try out your latest branch, and if it still happens I'll post more details.

rdmark, 2021-09-21:
@uweseimet The issue persists in issue_247 fa63022 -- and I didn't need a 4GB file, it happens also with a 2048 MB file created with develop web ui.  It seems that's the size that triggers it too. If you have a 2047 MB file, there's no crash (didn't test down to the byte.)

This is with a Pi 3b with a fresh Pi OS created by the official Pi imager tool, and all packages at the latest version (apt upgrade)

uweseimet, 2021-09-22:
@rdmark Well, I cannot reproduce this. I think the compiler version used has still not been mentioned here. This is an issue with the C++ library.

rdmark, 2021-09-22:
@uweseimet For starters...

> pi@rpi3b:~/RASCSI/src/raspberrypi $ g++ --version
> g++ (Raspbian 8.3.0-6+rpi1) 8.3.0
> Copyright (C) 2018 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
>

rdmark, 2021-09-22:
> pi@rpi3b:~/RASCSI/src/raspberrypi $ ldd bin/fullspec/rascsi
>         linux-vdso.so.1 (0x7eaa4000)
>         /usr/lib/arm-linux-gnueabihf/libarmmem-${PLATFORM}.so => /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so (0x76f8c000)
>         libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76f62000)
>         libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0x76f37000)
>         libpcap.so.0.8 => /lib/arm-linux-gnueabihf/libpcap.so.0.8 (0x76eef000)
>         libprotobuf.so.17 => /lib/arm-linux-gnueabihf/libprotobuf.so.17 (0x76c8b000)
>         libstdc++.so.6 => /lib/arm-linux-gnueabihf/libstdc++.so.6 (0x76b44000)
>         libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76ac2000)
>         libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x76a95000)
>         libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76947000)
>         /lib/ld-linux-armhf.so.3 (0x76fa1000)
>         libatomic.so.1 => /lib/arm-linux-gnueabihf/libatomic.so.1 (0x7692e000)

rdmark, 2021-09-22:
I'm going to bed here now, but if there's anything else that I can do to help troubleshoot I'll have a look in the morning.

uweseimet, 2021-09-22:
@rdmark Everything is the same as on my system. But I can reproduce this now. What I missed before was that the big image file I was testing with was not in my default folder but in a different location.
I hope I can find a solution that can make use of the GNU C++ library, which obviously has an issue here. ...

uweseimet, 2021-09-22:
:-( Looks like the C++ library was not built with the required flags for large file support: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=91947

uweseimet, 2021-09-22:
@Aapjedotinfo OK, looks as if I could fix this faster than expected. Fortunately we do not need code that also runs under Windows, because iterating files in a folder works different there.
Please try again with the latest issue_247 branch.

rdmark, 2021-09-22:
@uweseimet Yep, that did the trick! Both my 2GB and 4GB files can be iterated over and listed by rasctl / web.

rdmark, 2021-09-22:
@Aapjedotinfo Please pull the latest develop and try again!

Aapjedotinfo, 2021-09-22:
Tested! Works like a charm!

[image]

rdmark, 2021-09-22:
Thanks for reporting & helping out with troubleshooting @Aapjedotinfo !

closed 2021-09-22 by rdmark