PiSCSI/piscsi #247: Files with upper case endings are not detected as valid images

rdmark, 2021-09-20:
Latest develop 9795c55:  If a file has an upper case file ending of a normally detected valid image format, it won't be accepted by rascsi unless you specify the device type.

> pi@rpi3b:~/RASCSI/src/web $ rasctl -i 0 -c attach -f ~/images/zxcv.HDS
> Error: No device type provided for unknown file extension 'HDS'

uweseimet, 2021-09-20:
@rdmark Is it documented anywhere that extensions are case-insensive? The rascsi help message only lists lower case extensions, and I don't think it is common that software treats extensions as case insensitive.

rdmark, 2021-09-20:
@uweseimet The HDS example is perhaps moot, but I've found it a common occurrence when downloading CD-ROM images from various sources that is comes with an upper-case '.ISO' file ending. Cannot comment on what other software does, but I think it could be confusing for users when their .iso files work with RaSCSI out of the box but .ISO don't. 

Anyhow, it's just a minor point. You can still attach these images if you specify device type. If you don't think this is a good idea, I won't push it. :)

uweseimet, 2021-09-20:
@rdmark I simply did not have a good use case, but I see your point now. I am going update the code accordingly.

Regarding case-insensitivity, for rasctl the command line options are case-insensitive. Since we are already running out of letters for command line options I'd like to make command line options case-sensitive. Then we can replace some less common options by upper-case letters, and so we can get at least of some excotic options like -y. Would you agree?

rdmark, 2021-09-20:
@uweseimet I'm supportive of making rasctl options case sensitive. I thought about suggesting it myself back when you mentioned running out of letters in the alphabet. ;)

uweseimet, 2021-09-20:
@rdmark When starting to change this I noticed that it is not rasctl but rascsi where the options are case-insensitive. Anyway, I think they should be case-sensitive for both rascsi and rasctl. I will change this.

uweseimet, 2021-09-20:
@rdmark The issue_247 branch contains my changes. Some of the options rascsi and rasctl have in common can now share the same letter.

rdmark, 2021-09-20:
@uweseimet Seems like a good change just reading the code. I like how you maintained the backwards compatibility for uppercase -I for rascsi (since a lot of people will be using that from habit)

uweseimet, 2021-09-20:
@rdmark Yes, -ID and -HD are still supported.

closed 2021-09-22 by uweseimet