PiSCSI/piscsi #248: Warn when invalid images files exist in images dir

rdmark, 2021-09-20:
Latest develop 9795c55:  Since the web ui now relies on rascsi to tell it which image files are valid to list up in the ui, to the user's eye there is no obvious logic to which images gets listed and which not.

One case that came up recently while testing https://github.com/akuker/RASCSI/issues/228 is when an image file gets slightly corrupted (e.g. a few bytes larger or smaller than the original file) when transferred onto the Pi, it get quietly ignored by rascsi.  It becomes hard for the less technical user to troubleshoot such a situation.

Perhaps we want rascsi to throw warnings when it encounters an image-looking file (by file ending?) which turns out to be invalid by another heuristic (multiple of 512 / 2048)?  I haven't look at the code to see if this is feasible with the current implementation, but just spitballing here for ideas.

uweseimet, 2021-09-21:
@rdmark I suggest that you just add a note in the UI telling the user what the selection criteria are. We cannot compensate for any user error, e.g. copying files that are not image files to the image folder. The user might even explicitly want other files to be there. And we cannot fix issues caused by broken files. One has to be careful not to overengineer ...
Also, we should not rely more than necessary on file extensions. Otherwise there will constantly be tickets asking for more extensions to be added.

But there is something we can consider, a feature that some APIs offer: We can add a boolean option to the LIST_IMAGE_FILES command, telling rascsi not to filter at all. (The default would be to filter.) This enables the client to do the complete filtering if it wants to do so. Of course, the more options you have on the client side, the more complicated the client gets.

rdmark, 2021-09-22:
@uweseimet I do love the clean solution that we have now where rascsi tells the web ui which images that it can handle, and don't want to go back to adding back validation logic to the front end. I don't think I would personally opt use a non-filtered option.

The trace logging that you added pretty much resolves my immediate concern here though. If a user is confused why their image isn't showing up, we can point them towards the logs to help with troubleshooting.

As a bit of a tangent: The one area where the web ui is still trying to be 'smart' is some additional logic that tries to identify probable image type based on file ending (which is code that predates my tinkering with it.)  I've been thinking a lot about this, and I believe the reason file ending checks are still valuable in the front end, is because if you encounter an unknown file ending as a pure front end user, you have no facility to force attaching it as a particular device type. The CLI user can always to a '-t schd' or whatever, but not the front end user.

A recent example that I helped a user with yesterday was the '.dsk' file ending, which is what Basilisk II (Mac emulator) uses for hard drive images, and seems to be commonly used among pre-configured Mac drive images floating around on the net. Neither rascsi backend nor frontend was detecting it, and the user ended up hacking the front end code to get it to detect it as SCHD.

rdmark, 2021-09-22:
@uweseimet Actually, I just realized one web ui feature that we lost with the move to having rascsi list valid files -- zip archives. This is the one example of a file format that the web ui can handle (unzip) but rascsi doesn't. Perhaps I can ask for that boolean for unfiltered files after all?

uweseimet, 2021-09-22:
@rdmark What does the web UI do with ZIP files? Unpack and upload the result?

rdmark, 2021-09-22:
@uweseimet Yes, it takes a zip file that resides in ~/images and unzips it in place. That's it!

uweseimet, 2021-09-22:
@rdmark And the ZIP file itself is deleted afterwards?

rdmark, 2021-09-22:
@uweseimet Not currently, no. It might be good idea though, to save on limited SD card storage space.

uweseimet, 2021-09-22:
@rdmark This means the image folder is cluttered with files that are no image files, which does not look like a good approach to me. Can't the web UI use a separate folder for ZIP files and just copy the runpacked image files to the image folder?

rdmark, 2021-09-22:
@uweseimet Fair point. At this stage, I think it's worth revisiting where to store other non-image files, including the text files containing web ui configurations and drive properties. 

It would add complexity to the file upload / download features of the web ui, so it's not a quick fix! Let me think about it.

uweseimet, 2021-09-22:
@rdmark I was not aware that even more non-image files are located in the image folder ... I would really be great to have a better solution.

rdmark, 2021-09-22:
@uweseimet I can imagine your surprise as a non-web ui user. :)
Cut this ticket to figure out a better solution. https://github.com/akuker/RASCSI/issues/256

closed 2021-09-22 by rdmark