PiSCSI/piscsi #249: Facility for padding image files

rdmark, 2021-09-20:
develop 9795c55

**Problem space:**  There are plenty of images (both hard drive and CD-ROM) floating around out there that are of non-standard sizes, i.e. not a multiple of 512 (or 2048 etc.)  Other more lenient pieces of software such as Basilisk II will happily read such files, but RaSCSI won't.

**Solution:**  Just spitballing at this point, but provide some facility to pad such image files, either dynamically at load time, or as a separate tool/UI flow to add the padding permanently.

karrots, 2021-09-20:
Voting for a separate tool or just CLI option to do the padding. Seems like silently modifying images like that may have unintended consequences if someone assumes the file hasn't changed. Example. Maybe they intended to mount read-only then later a HASH doesn't match and they get confused.

uweseimet, 2021-09-21:
I agree that rascsi should not do anything like that. rascsi is not meant to be a tool that repairs files etc. In addition, if rascsi is too lenient with using files that are not usable it might cover filesystem issues or other severe issues, which should not be hidden from the user.

stinkerton18, 2021-09-23:
Attaching an image for a use case. This ISO boots and is installable in BasaliskII and, if burned to a CD, is bootable in drives that can read burned media.

[image]

rdmark, 2021-09-24:
@stinkerton18 Can you share the URL where you got the file from?

stinkerton18, 2021-09-24:
Certainly. I picked up the ISO I'm using from Winworld here: https://winworldpc.com/download/7724c394-e280-9362-c382-11c3a6e28094

rdmark, 2021-09-25:
@stinkerton18 Thanks for sharing!  Now, I was actually able to attach this image as-is with RaSCSI over here, so maybe the problem you had was with that upload form corruption bug?

Anyhow, I've found two test cases of my own:
https://macintoshgarden.org/apps/macintosh-os-71x (0074 in MacOS71_CD.ZIP)
https://macintoshgarden.org/apps/adam-the-inside-story

Was able to confirm that this method worked for padding:
`dd if=/dev/null of=TheInsideStory.iso bs=1 count=1 seek=456486912`

rdmark, 2021-09-25:
Implemented in rascsi-web here: https://github.com/akuker/RASCSI/pull/269