PiSCSI/piscsi #253: Handle MO .mos files in the web ui

rdmark, 2021-09-21:
MO (magneto-optical) images used primarily by the X68000 use a .mos file ending, as per the de-factor standard established by the XM6 emulator for X68000.
This is to support creating, listing and attaching .mos files in the web ui.

References:
https://mijet.eludevisibility.org/XM6%20Pro-68k/XM6%20Pro-68k.html
https://www.6502man.com/Bricolages/X68000/CD_MO_Scsi_Device/X68000_MO_howToUse_WintoX68KtoWin.html

Examples of supposedly valid .mos files:
https://nfggames.com/forum2/index.php?topic=4917.0
https://nfggames.com/X68000/

rdmark, 2021-09-21:
Out of the gate, rascsi on develop has no problem recognizing .mos images. However, the one and only verified .mos dump that I've found so far is just a few 100 bytes too large for rascsi to be able to attach it -- 228518912 when rascsi expects 228518400.

uweseimet, 2021-09-22:
Would anybody notice if the X68000 stuff was just discarded? I wonder whether this can be supported at all, as far as testing is concerned.

rdmark, 2021-09-22:
Last I heard @akuker wasn't ready to discard X68k code. Agreed that testing is a challenge on that platform.

BTW, wasn't SCSI MO drives standard on NeXT machines as well? I'm pretty sure I saw someone mentioning using RaSCSI with their NeXT system.

rdmark, 2021-09-27:
Current develop now recognizes .mos images on the back and frontend side: rascsi backend recognizes it as a SCMO device, while the frontend has it among the 'recognized file types'. 
At this point, I think creating mos images is out of scope, since noone on the dev team has an environment to validate the functionality further.

closed 2021-09-27 by rdmark

uweseimet, 2021-09-27:
@rdmark One more note on this: I tried different MO drives in the past with my Atari, and at least with the Atari no special handling is needed. You just need a driver that supports optical drives and (depending on the medium) 2048 bytes per sector. Then you can use MO drives like any other removable media drive.
Using any .mos image with the Atari works fine. The only restriction is that rascsi limits the choice of available images sizes to what it assumes to be the size of real-world MO media.