PiSCSI/piscsi #256: The web ui should store non-image files somewhere else

rdmark, 2021-09-22:
The web ui is currently storing configuration files (json/properties) as well as any uploaded file directly in the ~/images dir.
Ideally, ~/images should be reserved to actual image files, and non-image files stored somewhere else for a cleaner separation of data.

akuker, 2021-09-25:
Personally, I kind of like them co-located with the image files. Could we prepend a '.' so that they are somewhat hidden?

So, `Miniscribe_M8425.properties` would become `.Miniscribe_M8425.properties`.

rdmark, 2021-09-25:
@uweseimet Since you brought this up in the first place, do you have an opinion on akuker's suggestion? Would it be straight-forward for rascsi to ignore files starting with '.' when scanning the images dir?

uweseimet, 2021-09-25:
@rdmark One can ignore files starting with a dot, of course, but I would like to learn a compelling reason why non-image files should reside in a folder called "images". It would be better to have a common parent folder for everything needed by rascti, e.g. a folder called "rasci" with sub-folders "images " and "config".

akuker, 2021-09-26:
I have a slight preference to have everything in the same directory. (Don't care what its called) But its only a slight preference. 

If you change the structure, could you update easyinstall.sh to make the move a little easier? Something like...
```
if /home/pi/images exists
    ask user if they want to update to new structure
    move all of the metadata to the new correct folder
    move the images to the new correct folder
```

uweseimet, 2021-09-26:
By the way, isn't the $HOME/.config folder where most Linux applications nowadays place their configuration files (in a sub-folder in .config)? So that's probably the right place. Also see https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html.

rdmark, 2021-09-27:
How about this middle road:
We keep /home/pi/images as-is, but introduce /home/pi/.config/rascsi to store web ui config / properties files.
This way, we don't even need a migration path, since the csv files of current master aren't compatible with develop anyways.

uweseimet, 2021-09-27:
@rdmark Sounds like a good approach to me. Please note that the current image path is $HOME/images, not something hard-coded like /home/pi/images. The rascsi folder should be $HOME/.config/rascsi. For the pi user this is not relevant, but there is no reason to force the rascsi user to always be the pi user.

rdmark, 2021-10-04:
PR in https://github.com/akuker/RASCSI/pull/299

closed 2021-10-06 by rdmark