PiSCSI/piscsi #258: Expose in the protobuf interface the detected device type for an image

rdmark, 2021-09-23:
While working on removing the device type detection logic in the web frontend, I got the idea of having the backend tell the frontend what type of device that it thinks a particular image file should be attached as.
For instance, add a field to PbImageFile of PbDeviceType enum type.

The benefit would be twofold:
1. The web frontend can display in the ui which type devices the list of image files will be detected as 
2. When the backend does *not* detect which device type it is, the frontend can offer the user a list of device types to choose from

rdmark, 2021-09-23:
@uweseimet What do you think about this idea?

uweseimet, 2021-09-23:
@rdmark This sounds like a good idea. I Added this feature and am going to raise a PR.

rdmark, 2021-09-23:
@uweseimet Great!  I've leveraged this new field in the web ui now in this PR: https://github.com/akuker/RASCSI/pull/260
Seems to be working as expected in my testing. BTW, when the device type cannot be detected, I get "UNDEFINED" as the device type, so this is what the web ui is looking for when deciding when to show the device type selector drop down.

closed 2021-09-24 by rdmark