PiSCSI/piscsi #262: Issues using RaSCSI with DEC VAXstation

nsafran1217, 2021-09-24:
# Info

- Which version of Pi are you using: Zero
- Which github revision of software: develop commit 829356a
- Which board version: FULLSPEC
- Which computer is the RaSCSI connected to: Digital VAXStation 3100 m38

# Describe the issue

The RaSCSI doesn't seem to be working correctly with any combination of Operating Systems on the VAX. I will try to sperate the symptoms by OS. I'm not sure which would be the easiest to troubleshoot with. I'm willing to willing to do any troubleshooting requested. 

# OpenVMS v6.2
The computer came with 2 hard drives, an RZ24 and RZ23. The RZ24 had an install of OpenVMS 6.2 on it. I confirmed that I can boot from the original RZ24.
When attempting to boot the image I took of the disk (dd if=/dev/sdc of=rz24.img), It hangs at a mount verification.
[image]
The log repeats the following messages once every second for forever once the mount verification message comes up.
```
[debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $00
[debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing TestUnitReady ($00)
[debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $12
[debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Inquiry ($12)
[debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $1A
[debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing ModeSense6 ($1A)
[debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $1A
[debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing ModeSense6 ($1A) 
```
I have not attempted to mount a drive in VMS yet.
The install ISO I found for VMS does not boot, but I don't think its a RaSCSI issue.
# Ultrix v4.5
The Ultrix ISO I downloaded from archive.org boots successfully and is able to install the root file system onto a virtual drive (RZ57 created with the web gui)
After you reboot into the HDD, when it asks to install supported software, the install will fail and drop to a shell. Ultrix reports this error:
```
*** SUPPORTED SOFTWARE INSTALLATION ***
rzopen: RRD42 unit 2: mode select failed
ufs_mount: /dev/rz2c on /mnt: I/O error

The mount command failed to mount partition c of the distribution media, rz2. Thus, the installation software was unable to load the supported software subsets.
```
This error is logged by RaSCSI when this happens:
```
debug] virtual void Disk::Read6(SASIDEV*) READ(6) command record=$000070E0 blocks=16
[debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $00
[debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing TestUnitReady ($00)
[debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $15
[debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing ModeSelect6 ($15)
[debug] Error status: Sense Key $05, ASC $26, ASCQ $00
[debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $03
[debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing RequestSense ($03)
[debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $15
[debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing ModeSelect6 ($15)
[debug] Error status: Sense Key $05, ASC $26, ASCQ $00
```
Note that this error happens on the ISO, not the HDD. I can't confirm that the ISO is correct.

# NetBSD
The NetBSD ISO will not boot. I tried v9.2 and v8.0.
It begins reading data, then tries to read a block that is greater than the capacity.
```
[debug] virtual void Disk::Read6(SASIDEV*) READ(6) command record=$000000C8 blocks=4
[debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $08
[debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Read6 ($08)
[debug] Capacity of 167274 blocks exceeded: Trying to read block 660848, block count 4
[debug] Error status: Sense Key $05, ASC $21, ASCQ $00
[debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $03
[debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing RequestSense ($03)
[debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $03
[debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing RequestSense ($03)
```

nsafran1217, 2021-09-24:
When I attempt to mount a blank disk in VMS, it is shown as offline. RZ57 created with the web gui. I've updated to develop commit 9d66469 so the size should be correct.
[image]

And in the log:
```
[debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $00
[debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing TestUnitReady ($00)
[debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $12
[debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Inquiry ($12)
[debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $1A
[debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing ModeSense6 ($1A)
[debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $1A
[debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing ModeSense6 ($1A)
```

nsafran1217, 2021-09-25:
I found a tool in OpenVMS which runs SenseMode against the scsi device. (SYS$ETC:SCSI_INFO).
http://hoffmanlabs.org/vmsfaq/vmsfaq_025.html Section 14.25 has more info. It also has some info of what is supported in OpenVMS.
The real Hard drive gives short responses, but the emulated drive (RZ55 generated with web gui) gives very long responses and dumps lots of hex to the screen. The capture from the serial terminal is 24,000 lines! The real hard disk is only 500 lines.
I also ran it against the image of the RZ24 I have and it was much shorter than the RZ55 image.

The Block Descriptor Length seems to change  for each drive, but only for ModeSense(10).
RZ24 Real: 8
RZ24 Emulated : 16576
RZ55 Emulated : 59216

For ModeSense(6), Block Descriptor Length is always 8 on both real and emulated drives.
I don't know enough about SCSI to know if this is a problem.

I've attached all 3 outputs from the tool. You should see what I'm talking about
[rz55emulated.txt](https://github.com/akuker/RASCSI/files/7229035/rz55emulated.txt)
[rz24emulated.txt](https://github.com/akuker/RASCSI/files/7229036/rz24emulated.txt)
[realrz24.txt](https://github.com/akuker/RASCSI/files/7229037/realrz24.txt)

I'm attempting to understand how the block descriptor length is calculated in disk.cpp. I don't have much C++ experience, so I'll take any hints you can give.

rdmark, 2021-09-25:
@uweseimet I think you're the best person to comment on this.

uweseimet, 2021-09-25:
I'd suggest to consult the SCSI spec for block devices (https://www.t10.org/drafts.htm) for a descirption of short and long block descriptors. If the rascsi behavior is not compatible with the specification details about what is wrong should be added to this ticket.

nsafran1217, 2021-09-28:
I'm going to close this issue as I feel its too general. I made a new issue that is more focused on one problem.
https://github.com/akuker/RASCSI/issues/278

closed 2021-09-28 by nsafran1217