PiSCSI/piscsi #263: Log Level in Web UI is not Correct

akuker, 2021-09-25:
# Info
- Which version of Pi are you using: Any
- Which github revision of software: develop 1311317
- Which board version: All
- Which computer is the RaSCSI connected to: Any

# Describe the issue
In the Web GUI, the Server Log Level option menu should default to the log level. Right now, it looks like it just always defaults to "trace". 

I'm guessing this will take some updates in rasctl/rascsi to be able to retrieve the current log level.

rdmark, 2021-09-25:
@akuker Good idea. Easy fix: https://github.com/akuker/RASCSI/pull/268

BTW, with this change do we want to perhaps remove the "Server log level:" line in the footer to keep it simple?

rdmark, 2021-09-25:
Pushed a change with a simplified footer. Let me know what you think. I can revert the change if you liked the longer footer!

akuker, 2021-09-26:
Looks great! Thank you @rdmark  for the updates!!!

Change was merged into develop

closed 2021-09-26 by akuker