PiSCSI/piscsi #265: Add authentication capability to the Web UI

akuker, 2021-09-25:
Right now the web UI is completely open for anyone to connect and modify things.

A useful feature sometime in the future would be the ability to require a password. 

If its feasible, it would be nice if would allow any of the Linux users who are in a certain group to log in. Or, by default, it could just be the 'pi' user.

rdmark, 2021-11-11:
@akuker What would you like to see happen when you access the webapp without logging in? Shall you see only a login screen, or shall we have a 'read-only' view where you can inspect everything but not carry out any actions (similarly to how Jenkins does it).

On the other end of the spectrum, perhaps we only block access to endpoints that modify the file system, and allow anonymous users to f.e. attach or detach devices, look at logs and change log levels, etc.

Thoughts?

akuker, 2021-11-11:
I like the idea of going into read-only mode. But, we should also disable downloading files while in read-only mode. Basically if you're not logged in, you can see what is connected, but that's it.

rdmark, 2021-11-11:
@akuker Sounds like a plan that I can buy into. I was thinking of using Flask-Login https://flask-login.readthedocs.io/en/latest/ . Once we figured out a login flow, it should just be a matter of adding the @login_required decorators to the endpoints that we want to protect (pretty much all of them, except index, drive_list, send_pwa_files, show_logs, and maybe device_info.)

@erichelgeson For your visibility. Let us know if you have other recommended libraries!

erichelgeson, 2021-11-11:
Why not keep it simple and have it opt in? Maybe just ask the user if they want to secure it with a password and setup a simple basic auth: https://nginx.org/en/docs/http/ngx_http_auth_basic_module.html

akuker, 2021-11-12:
What about using PAM to authenticate users to make the setup a little easier? That would be my vote. We could still have a configuration option to enable or disable password requirements. 

There's the issue of sending un-encrypted passwords over the network though. But maybe that could be a feature for the next release - to add optional https support. (So that we can still use Netscape and other old browsers)

rdmark, 2021-11-25:
PR using simplepam and a webform in the header for authenticating with Linux users on the system. https://github.com/akuker/RASCSI/pull/483

uweseimet, 2021-12-12:
@rdmark @akuker I have a related question/proposal on authentication/security. IMO RaSCSI is not security-critical, but adding a decent level of authentication to the RaSCSI core might be useful. This is what I suggest: An additional optional parameter that denotes a file with a password. This file must be readable by root only and RaSCSI has to check this when it is started. All RaSCSI operations are only executed if a new "password" parameter is passed with each operation. It's good to have named parameters, so we can add this optional new parameter without losing backwards compatibility.

This approach has a loophole, though: The password has to be sent in cleartext via the remote interface. But as already mentioned, RaSCSI is not security critical, and any more elaborate solution is probably too complicated to be implemented at all.

uweseimet, 2021-12-14:
@rdmark @akuker I would appreciate some feedback.

rdmark, 2021-12-14:
@uweseimet Will the clients have to pass the password in their command payloads, or is this done internally in the rascsi process only?

uweseimet, 2021-12-14:
@rdmark The password would be an additional parameter to each interface operation. rascsi cannot do that, because this would defeat the purpose. The client has to authenticate against the service, i.e. rascsi, not the service against itself.
rasctl would most likely have an additional option, which triggers rascsl to wait for a password input. This would be the same as with the mysql command line client.

As already mentioned, this feature would be optional, i.e. the password parameter would only be required if authentication is configured, i.e. if a password file was specified when starting rascsi.

rdmark, 2021-12-14:
@uweseimet Understood, so this would be a replacement for the 'magic word' (RASCSI) that you introduced a few months ago. How would the client know whether or not to pass the password parameter? Or maybe I'm overthinking this, and both server and clients would just look for the existence of that password file on the file system, and if it exists take the auth code paths.

uweseimet, 2021-12-14:
@rdmark No, the magic word is just a technical feature on a low level, to ensure that you do not connect to and crash the service by accident.
The client does not know anything about the token file. It does not even know where the file is located, because that's specified when starting rascsi (see my initial comment above). It must definitely not use the file the server uses, because this would render everything useless. The client might not even run on the same machine as rascsi.

All in all this is just like with any other token-based approach. It's up to the client to know where to get the token from, but the client never uses the same source for the token as the server. For rasctl user input is required, the web ui might also require user input, or would be configured with a token or a token file, or woudl not use this feature at all. (The standard rascsi setup should IMO not require a token.) Other clients might store the token (entered by the user) in a database, e. g. an app.

In general, any authentication not enforced by rascsi but by other means is meaningless. If, for instance, there was some kind of authentication in the web ui only, it would not prevent any other client (e. g. rasctl) from having full access to rascsi.

uweseimet, 2021-12-16:
I created https://github.com/akuker/RASCSI/issues/528 for the authentication feature.

rdmark, 2021-12-22:
Client implementation of token authentication feature here https://github.com/akuker/RASCSI/pull/554

closed 2021-12-26 by rdmark