PiSCSI/piscsi #272: easyinstal overwrites custom systemd service files

karrots, 2021-09-26:
# Info

- Which version of Pi are you using: Zero W
- Which github revision of software: 3d0c200
- Which board version: Fullspec 2.3
- Which computer is the RaSCSI connected to: Various

# Describe the issue
When using easyinstall.sh to upgrade existing installs, the script will overwrite existing rascsi.service files. This causes default disk mounts to be lost. Additionally `systemctl daemon-reload` needs to be run when these files are updated.

karrots, 2021-09-26:
One of the two following operators could be used with copy to prevent overwriting.
`-n` never overwrite
`-u only copy if newer. Uses time stamps so may be problematic.

rdmark, 2021-09-26:
@karrots Thanks for reporting! The daemon-reload step is clear oversight. I'll get that added.

About rascsi.service, I want to try to consolidate two considerations: 1, to be respectful of user's data, and 2, make sure important upstream changes to the file gets applied.
Case in point for no.2, we've added the '-r' option in develop for controlling reserved scsi ids. It's important that this get applied for all users who upgrade.

The most straight-forward solution I can think of right now, would be to do the timestamp check as you suggest, and if it detects an older file, move it to something like 'rascsi.service.old' in the same dir, and then copy over the new rascsi.service. If the time stamps are identical, echo a message to that effect and move on.

Thoughts?

rdmark, 2021-09-26:
Actually, while the daemon-reload step is easy to add, everything that has to do with installing rascsi proper is done in the Makefile, including the rascsi.service.  I'm looking into what can be done in the Makefile for this scenario.

karrots, 2021-09-26:
`-u` will copy only if newer. You could always copy a backup all the time and use the `-u` option to only over write the active file if new.

rdmark, 2021-09-26:
@karrots BTW, are you by any chance a user of rascsi-web? In develop, rascsi-web has a feature of automatically loading the configuration stored in the 'default.json' file when it starts up. Once the next major version of RaSCSI is released, I'm going to promote this as the recommended way to manage default configurations, since it gives you a ton more flexibility and easy configuration management without having to jump through the hoops of systemd, while also completely removing the risk of overwriting user configurations when installing new versions of the software. I'd really appreciate if you tried this feature out and gave us constructive feedback. :)

rdmark, 2021-09-26:
This should address the immediate concern https://github.com/akuker/RASCSI/pull/273

@uweseimet Do you have any thoughts about doing a check in the Makefile for existing rascsi.service in systemd?

karrots, 2021-09-27:
I am a web user. That was one of my other thoughts. If config file for boot time mounting became the norm overwriting existing service files will be less of an issue.

If you look at what deb files do they rename the new file and give the user a choice of what to do. It may be worth automating to a deb file and utilize existing infrastructure.

uweseimet, 2021-09-27:
@rdmark A Makefile is meant to only build the binaries, and it is platform-independent. Anything platform-specific (systemd is platform-specific) would have to be done somewhere else.
@karrots  I agree with the deb file approach. IMHO the best installation method would be to have a Debian package that is managed by the apt package manager. Using apt to install and configure is what Raspberry OS/Debian/Ubuntu users are familiar with, and it usually provides the best (and a standardized) user experience.

uweseimet, 2021-09-28:
@rdmark I may be missing something, but did you find the reason why this was discarded?

rdmark, 2021-09-28:
@uweseimet I think @akuker is the best person to address this.

rdmark, 2021-09-28:
Further discussing moving to deb warrants a new ticket I think. Let me close this one now. Fixes to the reported issues have been merged to develop!

closed 2021-09-28 by rdmark