PiSCSI/piscsi #274: Bring back scsihd_apple functionality

rdmark, 2021-09-27:
With the removal of scsihd_apple code, we lost the code paths that allowed special handling for hda hard drive images on the Mac platform. What this means in real terms is that by default, a RaSCSI hard drive image won't be recognized by Apple HD SC Setup or Drive Setup for initialization / formatting. There may be other considerations that I am not aware of, so a more knowledgeable person can add requirements here. :)

Anyhow, as far as my testing shows, it is the INQUIRE vendor string that is the key part that tricks the setup tools. In this sense, the equivalent workaround in current develop is to attach a drive image with:
`-n " SEAGATE:          ST225N:0210"`

However, I argue that by removing the hda codepath, we're making it harder to use RaSCSI on a Mac, which I don't think is in this project's best interest.

At least, we should bring back SCSIHD_APPLE::Inquiry or an equivalent solution. I'm not sure if the logic in SCSIHD_APPLE::AddVendor is needed at this point.

uweseimet, 2021-09-27:
@rdmark We can simply set valid (from the Mac perspective) INQUIRY data for the .hda extension. There is no need for SCSIHD_APPLE to do that. Which INQUIRY data do you suggest? The SEAGATE one above?

uweseimet, 2021-09-27:
@rdmark As I had to touch the relevant files anyway I added the Seagate device identifier for .hda as part of https://github.com/akuker/RASCSI/pull/275. This is currently without a leading blank (see above). Can you please check whether your hard disk tool is fine with this?

rdmark, 2021-09-27:
@uweseimet I've tested this now, and bizarrely enough Apple HD SC Setup expects exactly those leading blanks in both the vendor and product strings. Drive Setup (the later tool) is ok with a product string without the 10 leading blanks, but it still expects exactly one blank before SEAGATE, otherwise it will treat the drive as 'unsupported'.

uweseimet, 2021-09-27:
@rdmark That's weird, but I have just added those blanks. Can you retry, please?

rdmark, 2021-09-27:
@uweseimet Can you please make it *10*  leading blanks in the product string, before ST255N specifically (yes, 10....)

uweseimet, 2021-09-27:
@rdmark Done, but very strange ...

rdmark, 2021-09-27:
@uweseimet Thanks for the update!  There's something that's tripping up the Apple tools here, unfortunately.  
On a code logic level, is there *anything* different in how INQUIRY is handled between attaching an 'hda' image without arguments, to doing something like:

`rasctl -i 3 -c attach -f /home/pi/images/qf3.hda -n " SEAGATE:          ST225N:0210"`
(with 10 blanks in the middle string)

I can't see any difference in the vendor strings through Apple System Profiler - reporting all three vendor strings with the correct number of blanks - but frustratingly enough neither of the drive setup tools recognizes the drive when not using the '-n' option...

Anyhow, I'll keep testing and trying a few things. There may be something obvious that I'm just not seeing.

rdmark, 2021-09-27:
@uweseimet Yes to my eyes they looked idential:

> pi@rpi3b:~/RASCSI/src/raspberrypi $ rasctl -i 3 -c attach -f /home/pi/images/qf3.hda
> pi@rpi3b:~/RASCSI/src/raspberrypi $ rasctl -i 3 -c show
>   3:0  SCHD   SEAGATE:          ST255N:0210  512 bytes per sector  545400320 bytes capacity  /home/pi/images/qf3.hda
> pi@rpi3b:~/RASCSI/src/raspberrypi $ rasctl -i 3 -c detach
> pi@rpi3b:~/RASCSI/src/raspberrypi $ rasctl -i 3 -c attach -f /home/pi/images/qf3.hda -n " SEAGATE:          ST225N:0210"
> pi@rpi3b:~/RASCSI/src/raspberrypi $ rasctl -i 3 -c show
>   3:0  SCHD   SEAGATE:          ST225N:0210  512 bytes per sector  545400320 bytes capacity  /home/pi/images/qf3.hda

uweseimet, 2021-09-27:
@rdmark Indeed, but there is no code difference I am aware of. This is what -n does:

```
		if (!pb_device.vendor().empty()) {
			device->SetVendor(pb_device.vendor());
		}
		if (!pb_device.product().empty()) {
			device->SetProduct(pb_device.product());
		}
		if (!pb_device.revision().empty()) {
			device->SetRevision(pb_device.revision());
		}
```

And this is the code for .hda:

```
					if (ext == "hda") {
						device->SetVendor(" SEAGATE");
						device->SetProduct("          ST255N");
						device->SetRevision("0210");
					}
```

closed 2021-09-28 by rdmark