PiSCSI/piscsi #277: Disable navigating away from web page while an upload is in progress

akuker, 2021-09-28:
Currently, when an upload is in progress, the user is still able to click on other buttons. When they do, the browser will load a new page and the upload is aborted.

If possible, the web interface should be updated so that if a file transfer is in process it will prompt the user to make sure they want to navigate away.

Potential approach:
https://stackoverflow.com/questions/1289234/alerts-when-navigating-away-from-a-web-page

If this is not usable on old browsers (Netscape 4, etc), maybe this could be conditionally embedded in the HTML?

rdmark, 2021-10-01:
I like the idea. Since we only want the prompt when a file transfer is in progress, we should somehow activate the beforeunload event handler only then. Some research on how the dropzone javascript library kicks off a file transfer will be required, so that we can latch on to that...

rdmark, 2021-11-28:
So researching this a bit, I think using the beforeunload event is the right way to go to capture the navigation. Some references:
- https://developer.mozilla.org/en-US/docs/Web/API/Window/beforeunload_event
- https://www.javascripttutorial.net/javascript-dom/javascript-beforeunload-event/

And as for Dropzone events, the documentation states that we can listen for the "addedfile" event:
- https://docs.dropzone.dev/configuration/events

However, I can't get the latter to work in the "declarative" style that we are using Dropzone with. I'm getting hard to debug errors, or no effect at all.

rdmark, 2022-08-25:
I made several attempt to implement such a hook for browser navigation based on signals coming from Dropzone but failed to get it working as intended. The event either did not take effect, or had other unintended effects.

In the end, I think any solution would involve adding a bunch of javascript to the page which isn't quite in line with keeping it lean for vintage browsers. So let me close this as wontfix for now.

closed 2022-08-25 by rdmark