PiSCSI/piscsi #278: Issues on OpenVMS with ModeSense(10)

nsafran1217, 2021-09-28:
# Info

- Which version of Pi are you using: Zero W
- Which github revision of software: Develop bfdaec6
- Which board version: FULLSPEC
- Which computer is the RaSCSI connected to: Digital VAXStation 3100 m38

# Describe the issue
This will focus on OpenVMS v7.2.

I am seeing crashes in the OpenVMS 7.2 operating system when using emulated hard drives with the RaSCSI. It seems to be related to the ModeSense(10) command.

I got rid of all repeatable crashes by disabling ModeSense(10) and always returning 0, but that's probably not a good solution for other computers.

In both `int Disk::ModeSense10(const DWORD *cdb, BYTE *buf)` and `int Disk::ModeSense6(const DWORD *cdb, BYTE *buf)` in disk.cpp, length is defined by the Allocation length parameter passed in the command. 
However, if I am reading the SCSI specification correctly, the amount of data that is transferred should be the length of the data, or the length requested by the device server, whichever is less. Section 4.2.5.6 Allocation length from Working Draft SCSI Primary Commands - 6 (SPC-6) page 48

I've changed the logic for the length of data returned by both ModeSense(6) and ModeSense(10) to match the specification as I am reading it. I am not an expert at programming or SCSI, so someone should check my work.
https://github.com/nsafran1217/RASCSI/commit/58397d7cc6796ed6507a073c953ed6efba608bdc

I can do a pull request, or someone else can implement this if this seems okay
[image]
.

nsafran1217, 2021-09-28:
I've changed the block descriptor data returned by ModeSense(10) to match the spec. The header being returned was incorrect.
I based this off Table 449 in SPC-6
https://github.com/nsafran1217/RASCSI/commit/8e3a2cf134a47923a66cf14cf67e026ec3c7949e

This has fixed all my issues in OpenVMS 7.2. I do not have any Mac computers to test this change with. I've confirmed that it does not break anything on a Sun SPARCStation 5 with Solaris 2.6.

[image]

uweseimet, 2021-09-29:
@nsafran1217 Sounds good, thank you for your contribution. I've just merged your changes.

nsafran1217, 2021-09-29:
Resolved in PR #280

closed 2021-09-29 by nsafran1217