PiSCSI/piscsi #279: Daynaport over wifi not working, Atari Falcon

mzryz, 2021-09-28:
# Info

- Which version of Pi are you using:
- Pi ZeroW
- Which github revision of software:
- develop c98dec8f44c15d9360514a959e899c81a110a1f5
- Which board version:
- FULL
- Which computer is the RaSCSI connected to:
- Atari Falcon

# Describe the issue
Using latest HD-Driver 11 & Mint 1.19-CUR. 
Drive image mounting is working perfect and reliably.
Attempting to use Daynaport over Wifi bridge doesn't work. I have put the scsilink.xif into the mint directory with all other XIF files, the driver is loaded and fully detected during boot on en0, so the driver DOES see the device. The Daynaport is also detected by HD-DRIVER and is on ID 6. I ran auto-configuration also.
Using E-NICK to configure the Mint network I set:
IP Address (static): 10.10.20.2
Router Address: 10.10.20.1
Subnet Mask: 255.255.255.0
DNS Server: 1.1.1.1
Rascsi Dayna mounted as such:
rascsi -ID 6 -t scdp wlan0:10.10.20.1/24

But when accessing anything on the internet, ie; ping www.google.com etc, the system completely locks for over 5 minutes, sometimes permanently. No results found. Also tried updating NTP date, after 5 minutes it returns it cannot access the network.

I configured everything through easysetup.sh.

@uweseimet I hope you can help! Thank you

uweseimet, 2021-09-28:
@mzryz I'm afraid I can't help with a MiNT setup.

mzryz, 2021-09-29:
@uweseimet I know it might be easy to assume it is freemint being the issue here, but is there some diagnostic commands I could run while ssh'd into the Rascsi to confirm the bridge is functioning properly? I have opened an issue log in the freemint repo, but what proof do I have that it's anything to do with their end?

uweseimet, 2021-09-29:
@mzryz  You can enable log level "trace" (by starting rascsi with "-L trace") to get some analytic output, in order to see which SCSI commands are sent by the Atari driver. Since the Daynaport emulation is working fine for Macs there might be an issue with the Atari drivers, or just with the driver setup. Roger Burrows should be able to help here, because he has implemented Daynaport drivers in the past. I did not succeed in getting his drivers working with my Atari (under MagiC) and a Riebel VME network card, though, probably because I could not find any up to date setup instructions.

mzryz, 2021-09-29:
@uweseimet could you please be more specific about enabling the trace for me please, I tried 'systemctl start rascsi -L trace' which is the only way I know, but this returned an error '-L unknown'. Also where will the trace be logged? Thanks

ps - I posted some new statistics in the other issue that you might find helpful: https://github.com/freemint/freemint/issues/229

uweseimet, 2021-09-29:
@mzryz Please ensure that you run the rascsi binary from the develop branch. Old versions do not support the -L option. The output is in the rascsi log.
I usually run rascsi from the command line, not by systemd, because especially for testing one has better control when starting rascsi manually. If you run rascsi from the command line the log output is displayed in the console window, and you can capture everything in a file with "rascsi -L trace >& /tmp/rascsi.log", for instance.

mzryz, 2021-10-01:
> @mzryz I'm afraid I can't help with a MiNT setup. I suggest to get help in the MiNT mailing list.

Roger Burrows says his driver works fine with a real Daynaport, so the issue is to do with the RaSCSI. I have closed the topic on the Atari side as they really had no interest in the RaSCSI and I was feeling that I was annoying them by even asking.

uweseimet, 2021-10-01:
@mzryz Well, we do not know on which side the issue is, do we? But I also was surprised by the lack of interest on the Atari side, because I'm quite sure that other users would also be interested in getting this issue resolved.

mzryz, 2021-10-01:
@uweseimet Thanks Uwe I understand, I will try my best to provide data people ask for. I will see if any network experts in the Discord can chip in. Thanks for doing what you can - by the way, here are the traces you asked for:

Upon first boot, device detect by HD-DRIVER, this is the trace:
`[2021-10-01 10:43:33.864] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=1 (with device)
[2021-10-01 10:43:33.865] [trace] void SCSIDEV::MsgOut() ID 1
[2021-10-01 10:43:33.865] [trace] Message Out Phase
[2021-10-01 10:43:33.865] [trace] void SCSIDEV::MsgOut() ID 1
[2021-10-01 10:43:33.865] [trace] virtual void SCSIDEV::Receive()
[2021-10-01 10:43:33.865] [trace] virtual void SCSIDEV::Receive() length was != 0
[2021-10-01 10:43:33.865] [trace] void SCSIDEV::MsgOut() ID 1
[2021-10-01 10:43:33.865] [trace] virtual void SCSIDEV::Receive()
[2021-10-01 10:43:33.865] [trace] virtual void SCSIDEV::Receive() ctrl.phase: 10 (msgout)
[2021-10-01 10:43:33.865] [trace] Message code IDENTIFY $80
[2021-10-01 10:43:33.866] [trace] virtual void SASIDEV::Command() Command Phase
[2021-10-01 10:43:33.866] [trace] virtual void SCSIDEV::Execute() Execution phase command $12
[2021-10-01 10:43:33.866] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $12
[2021-10-01 10:43:33.866] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Inquiry ($12)
[2021-10-01 10:43:33.866] [trace] void SASIDEV::DataIn() Going into Data-in Phase
[2021-10-01 10:43:33.866] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 32
[2021-10-01 10:43:33.867] [trace] virtual void SCSIDEV::Send() Move to next phase datain (6)
[2021-10-01 10:43:33.868] [trace] void SASIDEV::Status() Status phase
[2021-10-01 10:43:33.868] [trace] void SASIDEV::Status() Status Phase $00
[2021-10-01 10:43:33.868] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[2021-10-01 10:43:33.868] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
[2021-10-01 10:43:33.868] [trace] void SASIDEV::MsgIn() Starting Message in phase
[2021-10-01 10:43:33.868] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2021-10-01 10:43:33.868] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[2021-10-01 10:43:33.868] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2021-10-01 10:43:33.869] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
[2021-10-01 10:43:33.869] [trace] virtual void SCSIDEV::BusFree() Bus free phase
[2021-10-01 10:43:33.870] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=1 (with device)
[2021-10-01 10:43:33.870] [trace] void SCSIDEV::MsgOut() ID 1
[2021-10-01 10:43:33.871] [trace] Message Out Phase
[2021-10-01 10:43:33.871] [trace] void SCSIDEV::MsgOut() ID 1
[2021-10-01 10:43:33.871] [trace] virtual void SCSIDEV::Receive()
[2021-10-01 10:43:33.871] [trace] virtual void SCSIDEV::Receive() length was != 0
[2021-10-01 10:43:33.871] [trace] void SCSIDEV::MsgOut() ID 1
[2021-10-01 10:43:33.871] [trace] virtual void SCSIDEV::Receive()
[2021-10-01 10:43:33.871] [trace] virtual void SCSIDEV::Receive() ctrl.phase: 10 (msgout)
[2021-10-01 10:43:33.871] [trace] Message code IDENTIFY $80
[2021-10-01 10:43:33.871] [trace] virtual void SASIDEV::Command() Command Phase
[2021-10-01 10:43:33.872] [trace] virtual void SCSIDEV::Execute() Execution phase command $25
[2021-10-01 10:43:33.872] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $25
[2021-10-01 10:43:33.872] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing ReadCapacity10 ($25)
[2021-10-01 10:43:33.872] [trace] bool Disk::CheckReady() Disk is ready
[2021-10-01 10:43:33.872] [trace] void SASIDEV::DataIn() Going into Data-in Phase
[2021-10-01 10:43:33.872] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 8
[2021-10-01 10:43:33.873] [trace] virtual void SCSIDEV::Send() Move to next phase datain (6)
[2021-10-01 10:43:33.873] [trace] void SASIDEV::Status() Status phase
[2021-10-01 10:43:33.873] [trace] void SASIDEV::Status() Status Phase $00
[2021-10-01 10:43:33.873] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[2021-10-01 10:43:33.873] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
[2021-10-01 10:43:33.873] [trace] void SASIDEV::MsgIn() Starting Message in phase
[2021-10-01 10:43:33.874] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2021-10-01 10:43:33.874] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[2021-10-01 10:43:33.874] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2021-10-01 10:43:33.874] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
[2021-10-01 10:43:33.874] [trace] virtual void SCSIDEV::BusFree() Bus free phase
[2021-10-01 10:43:33.875] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=1 (with device)
[2021-10-01 10:43:33.875] [trace] void SCSIDEV::MsgOut() ID 1
[2021-10-01 10:43:33.875] [trace] Message Out Phase
[2021-10-01 10:43:33.875] [trace] void SCSIDEV::MsgOut() ID 1
[2021-10-01 10:43:33.875] [trace] virtual void SCSIDEV::Receive()
[2021-10-01 10:43:33.875] [trace] virtual void SCSIDEV::Receive() length was != 0
[2021-10-01 10:43:33.875] [trace] void SCSIDEV::MsgOut() ID 1
[2021-10-01 10:43:33.875] [trace] virtual void SCSIDEV::Receive()
[2021-10-01 10:43:33.875] [trace] virtual void SCSIDEV::Receive() ctrl.phase: 10 (msgout)
[... cut ...]
[2021-10-01 10:46:57.053] [trace] virtual void SCSIDEV::Receive() ctrl.phase: 10 (msgout)
[2021-10-01 10:46:57.053] [trace] Message code IDENTIFY $80
[2021-10-01 10:46:57.053] [trace] virtual void SASIDEV::Command() Command Phase
[2021-10-01 10:46:57.053] [trace] virtual void SCSIDEV::Execute() Execution phase command $08
[2021-10-01 10:46:57.054] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $08
[2021-10-01 10:46:57.054] [debug] virtual bool SCSIDaynaPort::Dispatch(SCSIDEV*) Executing Read6 ($08)`

mzryz, 2021-10-01:
And here they are zipped as text files:
[traces.zip](https://github.com/akuker/RASCSI/files/7266032/traces.zip)

uweseimet, 2021-10-01:
@mzryz Thank you. It's a pity that nobody who knows the MiNT drivers wants to have a closer look at this issue. Usually, when all sources are available it is not a big deal to find out why something goes wrong. I did the same with the RaSCSI mass storage isses I stumbled upon when using  HDDRIVER with RaSCSI, and since then a lot has been extended and improved, see  https://www.rascsi.de/en/rascsi.html for an overview.

uweseimet, 2021-10-01:
@mzryz One more thing: Maybe you can find somebody who would like to have a look at this issue by posting a request on one of the Atari internet forums.

mzryz, 2021-10-01:
@uweseimet good idea, hopefully it can attract some experts: https://atari-forum.com/viewtopic.php?f=27&t=41021

harbaum, 2021-10-01:
This endless loop looks like something is permanently polling the TAP device for packets. But there aren't any (at least at the moment of that trace). In none of these traces it looks like the device is trying to send anything. This isn't necessarily wrong. If the Atari/Mac side has a static setup and doesn't explicitely initiate any communication then such constant polling is actually all that I'd expect.

In case network communication is actually expected at that time then I'd run tcpdump or the like on the pi and check if e.g. data coming from the SCSI side can be seen incoming on the tap device.

mzryz, 2021-10-01:
Thanks @harbaum this has also been suggested to me by another person (Gizmo) so I'll get a TCP dump as soon as I can. Cheers.

mzryz, 2021-10-01:
@harbaum 

Here are some results for you to see:

Upon starting tcpdump I'm seeing nothing, but when submitting 'ping 10.10.20.1' on the host (atari) activity in the TCPdump is this:
`pi@rascsi:~ $ sudo tcpdump -nqi ras0 host 10.10.20.1
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on ras0, link-type EN10MB (Ethernet), capture size 262144 bytes
18:30:16.662244 ARP, Request who-has 10.10.20.1 (ff:ff:ff:ff:ff:ff) tell 10.10.20.2, length 46
18:30:16.662432 ARP, Reply 10.10.20.1 is-at 1e:ba:20:07:0f:bc, length 28
18:30:17.659324 ARP, Request who-has 10.10.20.1 (ff:ff:ff:ff:ff:ff) tell 10.10.20.2, length 46
18:30:17.659446 ARP, Reply 10.10.20.1 is-at 1e:ba:20:07:0f:bc, length 28
18:30:18.659386 ARP, Request who-has 10.10.20.1 (ff:ff:ff:ff:ff:ff) tell 10.10.20.2, length 46
18:30:18.659507 ARP, Reply 10.10.20.1 is-at 1e:ba:20:07:0f:bc, length 28
18:30:19.659439 ARP, Request who-has 10.10.20.1 (ff:ff:ff:ff:ff:ff) tell 10.10.20.2, length 46
18:30:19.659559 ARP, Reply 10.10.20.1 is-at 1e:ba:20:07:0f:bc, length 28
18:30:21.659967 ARP, Request who-has 10.10.20.1 (ff:ff:ff:ff:ff:ff) tell 10.10.20.2, length 46
18:30:21.660087 ARP, Reply 10.10.20.1 is-at 1e:ba:20:07:0f:bc, length 28
18:30:22.659541 ARP, Request who-has 10.10.20.1 (ff:ff:ff:ff:ff:ff) tell 10.10.20.2, length 46
18:30:22.659658 ARP, Reply 10.10.20.1 is-at 1e:ba:20:07:0f:bc, length 28
18:30:23.659607 ARP, Request who-has 10.10.20.1 (ff:ff:ff:ff:ff:ff) tell 10.10.20.2, length 46
18:30:23.659729 ARP, Reply 10.10.20.1 is-at 1e:ba:20:07:0f:bc, length 28
18:30:24.659623 ARP, Request who-has 10.10.20.1 (ff:ff:ff:ff:ff:ff) tell 10.10.20.2, length 46
18:30:24.659747 ARP, Reply 10.10.20.1 is-at 1e:ba:20:07:0f:bc, length 28`
* Does this forever, the host machine locks-up after a while.

Here is the output of ipaddress show, if important:
`pi@rascsi:~ $ ip address show
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:fe:3d:55 brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.51/24 brd 192.168.0.255 scope global noprefixroute wlan0
       valid_lft forever preferred_lft forever
    inet6 fd51:42f8:caae:d92e::ff/64 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::4579:c572:ae6c:fc26/64 scope link 
       valid_lft forever preferred_lft forever
3: ras0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast master rascsi_bridge state UNKNOWN group default qlen 1000
    link/ether 1e:ba:20:07:0f:bc brd ff:ff:ff:ff:ff:ff
    inet6 fe80::1cba:20ff:fe07:fbc/64 scope link 
       valid_lft forever preferred_lft forever
4: rascsi_bridge: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether 1e:ba:20:07:0f:bc brd ff:ff:ff:ff:ff:ff
    inet 10.10.20.1/24 brd 10.10.20.255 scope global rascsi_bridge
       valid_lft forever preferred_lft forever
    inet6 fe80::dcfe:99ff:fe1b:b337/64 scope link 
       valid_lft forever preferred_lft forever`

mzryz, 2021-10-01:
[tcpdump.zip](https://github.com/akuker/RASCSI/files/7269074/tcpdump.zip)

mzryz, 2021-10-01:
The guys on atari-forum are asking me for some low-level SCSI logs, not just the regular trace output of rascsi @uweseimet would you know how I can generate this? hopefully I can do it on the Pi side, as I think I might find it difficult to do so on the Atari side.

uweseimet, 2021-10-01:
@mzryz In case you need more logging output than currently available on trace level I suggest that you create a feature branch (e.g. on a forked repository) and (temporarily) add whatever is needed. The logging on trace level is already quite detailed, though, 
With the scsimon tool (part of rascsi) you can capture even more output, but I have never used it, i.e. I cannot comment on how useful this tool is in practice.

harbaum, 2021-10-08:
You explicitely asked for help and a bunch of people have spent some of their time to help you. I think it's somewhat impolite to then give up after a few days only.

jcs, 2021-11-08:
@mzryz can you run a `tcpdump -n -i rascsi_bridge` and see if it prints anything while pinging?

Also FYI when pasting a block of text, prefix it with three backticks on a line at the beginning and one at the end, that way GitHub formats it properly.

    ```
    Like this
    ```

mzryz, 2021-11-09:
Hi @jcs okay:
```
pi@rascsi:~ $ sudo tcpdump -n -i rascsi_bridge
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on rascsi_bridge, link-type EN10MB (Ethernet), capture size 262144 bytes
17:30:36.067614 ARP, Request who-has 10.10.20.1 (ff:ff:ff:ff:ff:ff) tell 10.10.20.2, length 46
17:30:36.067762 ARP, Reply 10.10.20.1 is-at a6:cd:ad:d6:d5:52, length 28
17:30:37.069393 ARP, Request who-has 10.10.20.1 (ff:ff:ff:ff:ff:ff) tell 10.10.20.2, length 46
17:30:37.069467 ARP, Reply 10.10.20.1 is-at a6:cd:ad:d6:d5:52, length 28
17:30:38.069472 ARP, Request who-has 10.10.20.1 (ff:ff:ff:ff:ff:ff) tell 10.10.20.2, length 46
17:30:38.069546 ARP, Reply 10.10.20.1 is-at a6:cd:ad:d6:d5:52, length 28
17:30:39.069472 ARP, Request who-has 10.10.20.1 (ff:ff:ff:ff:ff:ff) tell 10.10.20.2, length 46
17:30:39.069545 ARP, Reply 10.10.20.1 is-at a6:cd:ad:d6:d5:52, length 28
17:30:40.070189 ARP, Request who-has 10.10.20.1 (ff:ff:ff:ff:ff:ff) tell 10.10.20.2, length 46
```
While on the Atari, it hangs while the above messages scroll, when doing ctrl&c the Atari shows:
```
ping 192.168.0.77  56 bytes of data
25 packets transmitted, 0 packets received, 100% packet loss
```
@harbaum a bunch of people? Just one, Uwe Siemet of which I have thanks. I did everything asked of me but the last suggestion was about gathering more data than a trace provides. I do not know how to do this and no suggestions were made to me as how to progress. You do not know what is happening in my life so please reserve your judgement.

uweseimet, 2021-11-09:
@mzryz Please calm down ;-). Since you provided trace logs of what happens after the device driver is loaded there were changes related to logging SCSI commands, which may help with analyzing your issue. Can you please attach a new trace log (only the part after the device driver was loaded), based on the current (!!) develop branch?

uweseimet, 2021-11-09:
@mzryz Yet another question: Is it possible for you to provide a drive image that contains a valid network setup? This means a drive image that contains nothing but a fully working MiNT networking setup that is supposed to work with the Daynaport. Nothing else, i.e. no AUTO folder programs or ACCs that are not needed, just the bare essentials. This image should either work out of the box with a Falcon or TT.

mzryz, 2021-11-09:
@uweseimet Ok will try and get a trace tonight, just re-compiling and updating. I think I can make a drive image for you, just give me a few days, I'll see what I can do.

uweseimet, 2021-11-09:
@mzryz OK. When the image is ready please send me an email, so that I can provide a (non-public) upload location.

mzryz, 2021-11-09:
[trace091121.zip](https://github.com/akuker/RASCSI/files/7507787/trace091121.zip)

Okay, updated to latest dev branch, attached Daynalink to SCSI ID 6 with default settings, same as before. But now the Falcon cannot set it's default route  (router location) IP, as soon as it tries to, it locks the system up and needs a hard reset. Log is attached.

```pi@rascsi:~ $ sudo rascsi -L trace
SCSI Target Emulator RaSCSI(*^..^*) version 21.99 --DEVELOPMENT BUILD-- (Nov  9 2021, 19:25:26)
Powered by XM6 TypeG Technology / Copyright (C) 2016-2020 GIMONS
Copyright (C) 2020-2021 Contributors to the RaSCSI project
Connect type : FULLSPEC
[2021-11-09 20:28:37.861] [info] No images currently attached.
No images currently attached.
[2021-11-09 20:28:48.341] [trace] Received SERVER_INFO command
[2021-11-09 20:28:48.382] [trace] Received DEVICES_INFO command
[2021-11-09 20:28:48.391] [trace] Received DEVICE_TYPES_INFO command
[2021-11-09 20:28:48.414] [trace] Received DEFAULT_IMAGE_FILES_INFO command
[2021-11-09 20:28:48.427] [trace] Received SERVER_INFO command
[2021-11-09 20:28:48.752] [trace] Received NETWORK_INTERFACES_INFO command
[2021-11-09 20:29:18.020] [trace] Received DEVICES_INFO command
[2021-11-09 20:29:18.033] [info] Validating: operation=ATTACH, device id=6, unit=0, type=SCDP, device params='interfaces=wlan0:10.10.20.1/24', vendor='', product='', revision='', block size=0
[2021-11-09 20:29:18.035] [info] Executing: operation=ATTACH, device id=6, unit=0, type=SCDP, device params='interfaces=wlan0:10.10.20.1/24', vendor='', product='', revision='', block size=0
[2021-11-09 20:29:18.037] [trace] Opening Tap device
[2021-11-09 20:29:18.039] [trace] Opened tap device 7
[2021-11-09 20:29:18.041] [trace] Going to open ras0
[2021-11-09 20:29:18.055] [trace] return code from ioctl was 0
[2021-11-09 20:29:18.057] [info] rascsi_bridge is already available
[2021-11-09 20:29:18.060] [debug] ip link set ras0 up
[2021-11-09 20:29:18.064] [debug] brctl addif rascsi_bridge ras0
[2021-11-09 20:29:18.078] [trace] Getting the MAC address
[2021-11-09 20:29:18.081] [trace] Got the MAC
[2021-11-09 20:29:18.082] [info] Tap device ras0 created
[2021-11-09 20:29:18.083] [debug] Tap interface created
[2021-11-09 20:29:18.083] [trace] virtual bool SCSIDaynaPort::Init(const std::map<std::__cxx11::basic_string<char>, std::__cxx11::basic_string<char> >&) memset(m_mac_addr, 0x00, 6);
[2021-11-09 20:29:18.084] [trace] virtual bool SCSIDaynaPort::Init(const std::map<std::__cxx11::basic_string<char>, std::__cxx11::basic_string<char> >&) m_mac_addr[0]=0x00;
[2021-11-09 20:29:18.085] [info] Attached SCDP device, ID 6, unit 0
[2021-11-09 20:29:18.187] [trace] Received SERVER_INFO command
[2021-11-09 20:29:18.276] [trace] Received DEVICES_INFO command
[2021-11-09 20:29:18.305] [trace] Received DEVICE_TYPES_INFO command
[2021-11-09 20:29:18.339] [trace] Received DEFAULT_IMAGE_FILES_INFO command
[2021-11-09 20:29:18.358] [trace] Received SERVER_INFO command
[2021-11-09 20:29:18.849] [trace] Received NETWORK_INTERFACES_INFO command
[2021-11-09 20:30:26.885] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
[2021-11-09 20:30:26.885] [trace] void SCSIDEV::MsgOut() ID 6
[2021-11-09 20:30:26.885] [trace] Message Out Phase
[2021-11-09 20:30:26.885] [trace] void SCSIDEV::MsgOut() ID 6
[2021-11-09 20:30:26.885] [trace] virtual void SCSIDEV::Receive()
[2021-11-09 20:30:26.886] [trace] virtual void SCSIDEV::Receive() length was != 0
[2021-11-09 20:30:26.886] [trace] void SCSIDEV::MsgOut() ID 6
[2021-11-09 20:30:26.886] [trace] virtual void SCSIDEV::Receive()
[2021-11-09 20:30:26.886] [trace] virtual void SCSIDEV::Receive() ctrl.phase: 10 (msgout)
[2021-11-09 20:30:26.886] [trace] Message code IDENTIFY $80, LUN 0 selected
[2021-11-09 20:30:26.886] [trace] virtual void SASIDEV::Command() Command Phase
[2021-11-09 20:30:26.886] [trace] virtual void SASIDEV::Command() Command Byte 0: $12
[2021-11-09 20:30:26.886] [trace] virtual void SASIDEV::Command() Command Byte 1: $00
[2021-11-09 20:30:26.887] [trace] virtual void SASIDEV::Command() Command Byte 2: $00
[2021-11-09 20:30:26.887] [trace] virtual void SASIDEV::Command() Command Byte 3: $00
[2021-11-09 20:30:26.887] [trace] virtual void SASIDEV::Command() Command Byte 4: $20
[2021-11-09 20:30:26.887] [trace] virtual void SASIDEV::Command() Command Byte 5: $00
[2021-11-09 20:30:26.887] [trace] virtual void SCSIDEV::Execute() Execution phase command $12
[2021-11-09 20:30:26.887] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $12
[2021-11-09 20:30:26.887] [trace] virtual bool SCSIDaynaPort::Dispatch(SCSIDEV*) Calling base class for dispatching $12
[2021-11-09 20:30:26.887] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Inquiry ($12)
[2021-11-09 20:30:26.887] [trace] virtual int SCSIDaynaPort::Inquiry(const DWORD*, BYTE*) Inquiry, allocation length: 32
[2021-11-09 20:30:26.888] [trace] response size is 32
[2021-11-09 20:30:26.888] [trace] void SASIDEV::DataIn() Going into Data-in Phase
[2021-11-09 20:30:26.888] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 32
[2021-11-09 20:30:26.888] [trace] virtual int GPIOBUS::SendHandShake(BYTE*, int, int) DELAYING for 100us after 6 bytes
[2021-11-09 20:30:26.889] [trace] virtual void SCSIDEV::Send() Move to next phase datain (6)
[2021-11-09 20:30:26.889] [trace] void SASIDEV::Status() Status phase
[2021-11-09 20:30:26.889] [trace] void SASIDEV::Status() Status Phase $00
[... cut ...]
[2021-11-09 20:31:12.455] [trace] void SASIDEV::MsgIn() Starting Message in phase
[2021-11-09 20:31:12.455] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2021-11-09 20:31:12.455] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[2021-11-09 20:31:12.456] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2021-11-09 20:31:12.456] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
[2021-11-09 20:31:12.456] [trace] virtual void SCSIDEV::BusFree() Bus free phase

```

uweseimet, 2021-11-09:
@mzryz The log shows a lot of details, so that there is no need to prepare an image for me. I would not be able to gather more information than the logs already reveal. As already mentioned I am not familiar with the Daynaport, so currently I cannot provide more help. But hopefully somebody else can, based on the logs.
From my perspective there is nothing obviously wrong with what the log shows. Are you absolutely sure that your software setup is correct?

mzryz, 2021-11-11:
Yeah I'm sure my network configuration is correct, like here I'm running ifconfig. The Mint TCP/IP system is connected to the driver fine and is communicating with 'en0', trying to send and receive. The IP address also corresponds with the RaSCSI 10.10.20.1 and the Atari itself set to 10.10.20.2 as in the wiki suggests.

Here in the ping command and ifconfig, we can see that it *has* been sending and receiving packets, but there are many packet errors.

I am going to start playing with different Mint kernels and things next... because like you say, it is looking to be more likely on the Atari side.. So I will do some testing in that area next.

[image]

uweseimet, 2021-11-11:
@mzryz Just to avoid misunderstandings: I don't say that it is more likely that there is something wrong on the Atari side. I just noticed that the logs show that no READ command is sent.

mzryz, 2021-11-14:
@uweseimet I am now testing with STING (to take Mint out of the equation) and I am also testing on my TT as well as my Falcon. So far my TT is showing exactly the same results for Mint as the Falcon does. For Sting it is also not working, I'll post some photos.

By the way, in the SCSILINK driver information it says:
"Make sure that arbitration is enabled for the SCSIDRV driver, and
that the id of the SCSI/Link device is being controlled by SCSIDRV."

I have just run 'auto-configure' in HD DRIVER. Would that enable arbitration?

mzryz, 2021-11-14:
[image]
[image]
[image]
[image]
[image]
[image]

uweseimet, 2021-11-14:
@mzryz The information that bus arbitration is required is wrong.  What's needed is initiator identification, which means that the TT or Falcon must have a SCSI ID of their own. The SCSI ID is saved in the NVRAM, and you can set it with HDDRUTIL or BOOTCONF or any tool that can configure the NVRAM.

uweseimet, 2021-11-14:
@mzryz How confident are you about your STinG configuration being correct? Maybe you have tested the same (as far as possible) configuration with a different adapter, e.g. Netusbee or whatever else there is for the Atari?

Do you think you can create a STinG setup for me that would work with a TT and a Riebel VME ethernet card? I was not able to create such a setup myself. If I had a working setup for the Riebel card I might be able to draw conclusions when replacing the Riebel card by the emulated Daynaport.

mzryz, 2021-11-14:
@uweseimet 
Although it is the first time I've ever used STinG, it was quite straight forward to configure. firstly see this website:
http://hardware.atari.org/manuals/netsting.htm

They host a full STinG zip file, you simply extract it to C:\
This puts a STING.PRG in AUTO and provides a /STING/ folder where you extract your driver (.STX) 
You then edit the ROUTE.TAB file inside /STING/ to this:
```10.10.20.1	255.255.255.0	EtherNet	10.10.20.2```
And then configure the Sting port CPX as shown in the Netusbee configuration I linked above.

But I don't know if a Sting driver exists for Riebel? I did a good search and came up with nothing. Good idea for me to try a NetUSBee with Sting. I'll try that out next.

mzryz, 2021-11-14:
I have put out an invitation for more Atari users to get involved on both Atari-Forum and Exxos forum. Hopefully more people testing will help.

mikrosk, 2021-11-16:
I think I have already mentioned it, Frank Lukas has a working Daynaport (I sold it to him) + a lot of Atari experience, so you can surely do some tests for you.

mzryz, 2021-11-16:
@mikrosk I might not understand correctly, but how does having a real Daynaport help the situation, what kind of tests could he perform? (Honest question Miro sorry not trying to sound sarcastic etc!) I think what we really need is the author of the Sting/Mint drivers, Roger, to be sent a RaSCSI. Since he is familiar with how the driver works and how the Daynaport should behave, he really would be the key to this. @akuker has mentioned he would send Roger a RaSCSI for this reason. It would be great to have him on-board with the project. We already have one SCSI Atari legend here, Uwe, who is doing great work, having Roger here too would be the icing on the big SCSI cake ;)

mikrosk, 2021-11-16:
It's simple - he can verify that the original and current drivers work in his machine. Of that is confirmed (most likely it will be), then you can send Use the very same software setup and tell him that yes, this 100% works with real hardware and bit with RaSCSI.

Right now there's still a chance that something is misconfigured on your side.

uweseimet, 2021-11-17:
@mzryz According to https://www.atari-forum.com/viewtopic.php?f=15&t=41208 the Atari/Daynaport issue is resolved, i.e. the Daynaport emuation in RaSCSI 21.10 is working also for the Atari. Can you confirm, so that we can close this ticket? If you cannot confirm you should contact the user who mentioned in https://www.atari-forum.com/viewtopic.php?f=15&t=41208 that his setup is working, in order to find out what might be wrong with your setup.

rdmark, 2021-11-17:
I updated the RaSCSI wiki to point to the thread. https://github.com/akuker/RASCSI/wiki/Dayna-Port-SCSI-Link#-atari-stttfalcon-setup-instructions

uweseimet, 2021-11-17:
@rdmark Sounds good, let's see whether @mzryz can confirm that these instructions also work for him.

mzryz, 2021-11-17:
@uweseimet This user is not trying the same thing as me, I am trying over Wifi, he has just done Ethernet. I have asked if he can try Wifi for me. (I have not been doing ethernet because for instance, my RaSCSI is located inside my case and cannot easily connect a LAN port)

uweseimet, 2021-11-17:
@mzryz How the DaynaPort emulation handles SCSI commands does not depend on any networking details. Only the networking setup (with or without a bridge) on the Pi is different.

rdmark, 2021-11-17:
FWIW, the main difference between wired and wifi configurations from the perspective of your Atari system is DHCP vs. static IP assignment. It should not make a huge difference to the network stack.

mzryz, 2021-11-17:
@rdmark Is there perhaps a static-ip incompatibility with my home network perhaps? The default bridge IP is 10.10.20.1 and it says set the Atari to 10.10.20.2 etc, yet my home network is 192.168.0.xx - is this supposed to be OK.

uweseimet, 2021-11-17:
@rdmark will probably also comment, but what I can say is that my home network also is 192.168.0.xx, and with the bridge being 10.10.20.1 I had no issues with the DaynaPort emulation and my Mac.
I do have the issue now, though, that my Mac is not working anymore, most likely because the battery is empty ... But that's a different story, of course.

rdmark, 2021-11-17:
@mzryz Yes I think this is perfectly fine. If you follow the steps correctly, NAT should be configured on the Pi which translates network traffic between the isolated network on the bridge and the rest of your home network.

mikrosk, 2021-11-18:
So this has been a RPi setup issue all this time? For future debugging attempts, I would recommend to strip the setup to bare minimum before inviting half of the Atari community to look into it... I was under impression that the networking doesn't work at all, not that it is specific to wifi.

mzryz, 2021-11-18:
@mikrosk sorry Mikro I guess you can't read topics. I specifically wrote WiFi in this particular topic. Whole Atari community? Oh you mean Uwe Siemet and Rahuken, the only two Atari users bothered enough to help? Or do you mean the issue I opened on FreeMint GitHub? That was suggested by the guys here as they thought it was a driver issue. Thanks again for your invaluable input.

mzryz, 2021-11-18:
@rdmark Rahuken on Atari forum as reported back that it isn't working on WiFi for him either. Are there any specific tests I could perform to try and diagnose this? Cheers

uweseimet, 2021-11-18:
@mzryz According to https://www.atari-forum.com/viewtopic.php?f=15&t=41208&sid=39ad156b617f2581388ccd96ce8278c8 my understanding is that WiFI was working, but it was very slow: "The Atari TT is becoming massively slow when trying to reach an ftp site. It works but it is very very slow." Slow is not "isn't working". This is something technically completely different.
I think it's time now that you open the case of your PI, in order to test your setup with a LAN connection in order to verify whether this makes a difference for you.

mzryz, 2021-11-18:
@uweseimet I have 2x RaSCSI's and I have 4x Pi's. 2x Pi Zero1's, 2x Pizero2'z & 4x fresh installations.

I do not see this issue log being irrelevant? This is still a RaSCSI issue, I am dependant on the advice given by the experts, I am simply a tester. If I'm provided with logical diagnostic steps to run on the Pi, to diagnose the bridge, I will run them and provide results for analysis.

uweseimet, 2021-11-18:
@mzryz Nobody said that this issue is not relevant. But if it turns out that a WiFi bridge is slow with a Pi Zero than that's the way it is, and in this case it's not a RaSCSI issue because bridiging is a feature of Linux, not of RaSCSI.

I failed with testing on my TT because the STinG CPX crashes when I try to configure the network. The CPX (one of them) also crashes with the Riebel driver, which actually exists. But if, like mentioned on https://www.atari-forum.com/viewtopic.php?f=15&t=41208&sid=39ad156b617f2581388ccd96ce8278c8, WiFi networking works but is just slow, then maybe a Pi 3 or 4 is faster.

mzryz, 2021-11-18:
@uweseimet I understand, but last week I upgraded to a Pi Zero 2 which is 4x faster than a Pi Zero 1. I have asked Rahuken to be a bit more specific and perhaps comment on here if he has a Github account.

uweseimet, 2021-11-18:
@mzryz I'm afraid you are not correct regarding bridging. Bridging is done by Linux, it's just set up by RaSCSI, but can be set up the same way on the command line. In the past (before RaSCSi 21.10) it was indeed required to do this on the command line. In order to simplify the bridging setup, what needed to be done on the command line is now done programmatically, that's all.

It's best if you closely communicate with Rarkhaan (I don't know who Rahukan is, you probably mean Rarkhaan) on the Atari forum, in order to find out whether there is a difference in your setup. 

I do use COPX by the way, and I never said that I was using XControl.

mzryz, 2021-11-18:
I have asked Rarkhaan if he can share his SD card image with me so I can test with his configuration.

uweseimet, 2021-11-18:
@mzryz Sounds good!

uweseimet, 2021-11-18:
@mzryz Please ensure that both you and Rarkhaan use the "fix_duplicate_logging" branch for your testing. It contains some improvements for the DaynaPort emulation.

uweseimet, 2021-11-18:
I found that I don't need all of the CPXes, especially not the crashing one (STING.CPX, which just seems to display version information). I also don't think I needed MASQUE.STX, so I disabled it.
Now I could configure the Atari network and used 10.10.20.2 as static IP, 255.255.255.0 as netmask. When booting (DaynaPort is SCSI ID 6) I soon end up in an endless loop, just like @mzryz at the beginning of this thread.
It looks to me as if the STING driver is constantly polling the Daynaport for data, without any interruption. This essentially freezes my TT (running under MagiC) after loading STING. The TT appears to be busy just with polling the DaynaPort every 50 ms, which also imposes a high load even on a Pi 4. I would not be surprised if a slower Pi was just freezing due to the constant polling. I remember that the Mac driver also polls, but which a much lower frequency.
See the details in my log below, which shows what happens when my TT boots and STING is loaded. I'm afraid I cannot provide more insight than my conclusions (which are just a guess based on the logs) above. Rarkhaan's logs may be important. The logs below were taken with the fix_duplicate_logging branch.

```
SCSI Target Emulator RaSCSI(*^..^*) version 21.99 --DEVELOPMENT BUILD-- (Nov 17 2021, 21:31:36)
Powered by XM6 TypeG Technology / Copyright (C) 2016-2020 GIMONS
Copyright (C) 2020-2021 Contributors to the RaSCSI project
Connect type : FULLSPEC
[2021-11-18 16:17:40.275] [info] Validating: operation=ATTACH, device id=6, unit=0, type=SCDP, vendor='', product='', revision='', block size=0
[2021-11-18 16:17:40.276] [info] Executing: operation=ATTACH, device id=6, unit=0, type=SCDP, vendor='', product='', revision='', block size=0
[2021-11-18 16:17:40.276] [trace] Opening Tap device
[2021-11-18 16:17:40.289] [trace] Opened tap device 5
[2021-11-18 16:17:40.289] [trace] Going to open ras0
[2021-11-18 16:17:40.290] [trace] return code from ioctl was 0
[2021-11-18 16:17:40.290] [info] rascsi_bridge is not yet available
[2021-11-18 16:17:40.290] [trace] Checking which interface is available for creating the bridge
[2021-11-18 16:17:40.290] [trace] Interface eth0 is not available or is not up
[2021-11-18 16:17:40.290] [trace] Interface wlan0 is up
[2021-11-18 16:17:40.290] [info] Creating rascsi_bridge for interface wlan0
[2021-11-18 16:17:40.290] [debug] ip address add 10.10.20.1/24 dev rascsi_bridge
[2021-11-18 16:17:40.290] [debug] brctl addbr rascsi_bridge
[2021-11-18 16:17:40.318] [debug] ip address add 10.10.20.1/24 dev rascsi_bridge
[2021-11-18 16:17:40.318] [debug] ip link set dev rascsi_bridge up
[2021-11-18 16:17:40.318] [debug] ip link set ras0 up
[2021-11-18 16:17:40.318] [debug] brctl addif rascsi_bridge ras0
[2021-11-18 16:17:40.319] [trace] Getting the MAC address
[2021-11-18 16:17:40.319] [trace] Got the MAC
[2021-11-18 16:17:40.319] [info] Tap device ras0 created
[2021-11-18 16:17:40.319] [debug] Tap interface created
[[2021-11-18 16:17:40.319] [info] Attached SCDP device, ID 6, unit 0
[2021-11-18 16:17:40.320] [info] +----+-----+------+-------------------------------------
[2021-11-18 16:17:40.321] [info] | ID | LUN | TYPE | IMAGE FILE
[2021-11-18 16:17:40.321] [info] +----+-----+------+-------------------------------------
[2021-11-18 16:17:40.321] [info] |  6 |   0 | SCDP | DaynaPort SCSI/Link
[2021-11-18 16:17:40.321] [info] +----+-----+------+-------------------------------------
+----+-----+------+-------------------------------------
| ID | LUN | TYPE | IMAGE FILE
+----+-----+------+-------------------------------------
|  6 |   0 | SCDP | DaynaPort SCSI/Link
+----+-----+------+-------------------------------------
[2021-11-18 16:18:07.213] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
[2021-11-18 16:18:07.213] [trace] void SCSIDEV::MsgOut() ID 6
[2021-11-18 16:18:07.213] [trace] Message Out Phase
[2021-11-18 16:18:07.213] [trace] void SCSIDEV::MsgOut() ID 6
[2021-11-18 16:18:07.213] [trace] virtual void SCSIDEV::Receive()
[2021-11-18 16:18:07.213] [trace] virtual void SCSIDEV::Receive() length was != 0
[2021-11-18 16:18:07.213] [trace] void SCSIDEV::MsgOut() ID 6
[2021-11-18 16:18:07.213] [trace] virtual void SCSIDEV::Receive()
[2021-11-18 16:18:07.213] [trace] virtual void SCSIDEV::Receive() ctrl.phase: 10 (msgout)
[2021-11-18 16:18:07.213] [trace] Message code IDENTIFY $80, LUN 0 selected
[2021-11-18 16:18:07.213] [trace] virtual void SASIDEV::Command() Command Phase
[2021-11-18 16:18:07.213] [trace] virtual void SASIDEV::Command() CDB[0]=$12
[2021-11-18 16:18:07.213] [trace] virtual void SASIDEV::Command() CDB[1]=$00
[2021-11-18 16:18:07.213] [trace] virtual void SASIDEV::Command() CDB[2]=$00
[2021-11-18 16:18:07.213] [trace] virtual void SASIDEV::Command() CDB[3]=$00
[2021-11-18 16:18:07.214] [trace] virtual void SASIDEV::Command() CDB[4]=$20
[2021-11-18 16:18:07.214] [trace] virtual void SASIDEV::Command() CDB[5]=$00
[2021-11-18 16:18:07.214] [trace] virtual void SCSIDEV::Execute() Execution phase command $12
[2021-11-18 16:18:07.214] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $12
[2021-11-18 16:18:07.214] [trace] virtual bool SCSIDaynaPort::Dispatch(SCSIDEV*) Calling base class for dispatching $12
[2021-11-18 16:18:07.214] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Inquiry ($12)
[2021-11-18 16:18:07.214] [trace] virtual int SCSIDaynaPort::Inquiry(const DWORD*, BYTE*) Inquiry, allocation length: 32
[... cut ...]
[2021-11-18 16:18:13.610] [trace] void SASIDEV::DataIn() Going into Data-in Phase
[2021-11-18 16:18:13.610] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 6
[2021-11-18 16:18:13.611] [trace] virtual void SCSIDEV::Send() Move to next phase datain (6)
[2021-11-18 16:18:13.611] [trace] void SASIDEV::Status() Status phase
[2021-11-18 16:18:13.611] [trace] void SASIDEV::Status() Status Phase $00
[2021-11-18 16:18:13.611] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
```

mzryz, 2021-11-18:
Ok, thanks Uwe, this is the first bit of data that correlates with what I've been experiencing all along. Maybe I won't be submitted to the mad-house at this rate. In the mean time I have received images from Rarkhaan and will test soon as I can to see if there is a difference in my previous results. I assume if the results are consistent we may be able to ask Roger Burrows if he could adjust the polling rate based on this.

[6 more replies]

closed 2021-12-10 by mzryz