PiSCSI/piscsi #308: Block size validation does not take custom block size into consideration

rdmark, 2021-10-08:
develop f7cdc91

Block size validation of an image being attached seems to be carried out against the default block size for the device type, rather than the block size specified when attaching.
For instance consider: 

>  $ rasctl -i 1 -c attach -t cd -b 512 -f /home/pi/images/71A.iso
> Error: Tried to open an invalid file '/home/pi/images/71A.iso': ISO CD-ROM file size must be a multiple of 2048 bytes

Where 71A.iso has a file size of 80075776 bytes which is a multiple of 512, but not 2048.

uweseimet, 2021-10-08:
@rdmark I just checked this, it was not related to the previous issue. There was still a hard-coded block size check for CDs in the code. I removed it and am going to create a PR.

rdmark, 2021-10-08:
@uweseimet Thanks! Do you have any more code changes planned, or shall I run a few tests and get this merged?

uweseimet, 2021-10-08:
@rdmark No more changes are planned. Thank you for testing.

rdmark, 2021-10-08:
@uweseimet Testing done, and merged! Now CD images that were previously failing the 2048 multiple check can be used with RaSCSI as-is. I'll ping the original reporter to let them test it out on their systems.

closed 2021-10-08 by rdmark