PiSCSI/piscsi #321: Cache data is not being automatically written back to SD card file

Pacjunk, 2021-10-13:
# Info

- Which version of Pi are you using: ZeroW
- Which github revision of software: develop
- Which board version: FULLSPEC
- Which computer is the RaSCSI connected to: Alphaserver 1000

# Describe the issue

If I write a small amount of data (<1.5MB) to an attached disk, this data is cached and not written to the SD card. I can mount and re-mount the disk on the host and everything is normal, but if sometime later I restart the rascsi service or reboot the raspberry pi, then reconnect the disk to the host all the changes are gone. If I write larger files, then the file is partially written to the SD card (the timestamps change), but not completely so the file ends up corrupt.

Now ,importantly, if I detach the drive using the cli or gui, then the data is correctly flushed to the file and everything is fine. If I restart rascsi without detaching the drives, the cache is not flushed and corruption results.

Example:
  - Create 100MB drive
  - Attach drive to ID 1:0
  - Format drive from host
  - wait 10 minutes or an hour, doesnt matter - the cache never flushes.
  - restart the rascsi service or reboot the pi
  - reattach drive to ID 1:0 if not done automatically
  - mount the drive on the host --> The drive is unformatted

I know the cache can't be flushed too often otherwise performance would suffer, but I feel it should flush automatically after a certain idle time. Having to "detach all" every time I shut down the Pi to make sure the files are valid should not be required.

uweseimet, 2021-10-14:
In the issue_319 branch I added support for detaching all devices to rasctl (rasctl -D).

Pacjunk, 2021-10-14:
Tested OK. I stopped and restarted using rasctl, rebooted the pi and even killed off the processes. In all cases it seems to have flushed the cache and the files were consistent upon restart. The web restart functions were OK as well (I realise you didn't do those).

Now there only remains the issue of possibly flushing the cache at some interval or idle time. I think this is still important as the last few changes are not flushed until shutdown time. This could be hours or days depending on usage. If I absent-mindedly disconnect the power, then the volume is corrupted. I realise that this is the harder thing to do!  I imagine the detach routine must at some stage call another routine to write out the cache. Or is the fault with the fat32/exfat driver caching stuff until file close?