PiSCSI/piscsi #350: Non-verbose server status protobuf commands

rdmark, 2021-10-20:
The server status command DEVICES_INFO causes rascsi to log the currently attached devices as an info level log in syslog.
This becomes a problem when running the rascsi_oled_monitor script which repeatedly (every 250ms) inquires the server about attached device in order to display real-time status on the OLED display. This causes syslog to be entirely flooded by rascsi logs while the monitor script is running.

rdmark, 2021-10-20:
@uweseimet What are your thoughts here? What's the best way to address this in your opinion?

rdmark, 2021-10-20:
> 
> 
> @rdmark By the way, I don't think that a process should regularly poll for the device list or anything else. The monitor should register with rascsi, and rascsi should send update information whenever there is a change.

Yes, if rascsi would send some kind of message upon change of the device configuration, it would be straight-forward to have the python script driving the monitor to handle that event. This is preferable to constant polling. Such a feature doesn't already exist on the rascsi side, does it?

uweseimet, 2021-10-20:
@rdmark No, it doesn't. If you want to add it, my suggesion is to add new REGISTER and UNREGISTER operations to rascsi. REGISTER registers a message subscriber based on the subscriber's host and port, which would be passed as message parameters.
rascsi should support a list of subscribers, so that there can be multiple clients of this kind. In case of a device list change rascsi would send a message (defined in a *new* protobuf interface) to all subscribers. If a subscriber does not respond, rascsi may (or even should) unregister this dead subscriber automatically (immediately or after a message could not be delivered to this subscriber several times).

I don't think there is any alternative to this or a similar approach. Constantly polling for something that hardly ever happens (there may not be a device list change for hours) would keep the Pi busy all the time (CPU and network load) without doing something useful.

uweseimet, 2021-10-20:
@rdmark Nevertheless I would like to remove the logging of the device list. Maybe you can convince @akuker that not logging the list is a good idea. Sometimes backwards compatibility is not that useful. In particular Mac users should be used to not having backwards compatibility ;-).

rdmark, 2021-10-20:
I think this PR should do it https://github.com/akuker/RASCSI/pull/355

akuker, 2021-10-20:
FYI: I threw the start of some release notes here. Its a combination of @uweseimet 's release notes and the auto generated ones. 
https://github.com/akuker/RASCSI/wiki/DRAFT-21.10_Release_Notes

rdmark, 2021-10-21:
I created a ticket for creating release notes at https://github.com/akuker/RASCSI/issues/361 so perhaps the conversation can shift over there. :)

@uweseimet I'm a big fan of UT and automation testing in general. It would be an excellent project for the next release to get that into place, and perhaps even webapp testing using Selenium Python bindings with Robot framework, for instance.

closed 2021-10-21 by rdmark