PiSCSI/piscsi #407: Improved error reporting for daynaport attach failures

akuker, 2021-10-30:
# Info

- Which version of Pi are you using: all
- Which github revision of software: v20.10
- Which board version: 2.3
- Which computer is the RaSCSI connected to: N/A

# Describe the issue

When a user tries to attach a daynaport scsilink, but they haven't configured the network interface with easy install, the web interface just gives a generic failure message. It would be useful for the web interface to figure out if the easyinstall network setup has been done, and if not, tell the user what to do.

rdmark, 2021-11-08:
@akuker I don't think we even throw any kind of error message when attaching daynaport before configuring ipv4 forwarding (for wired) or NAT (for wireless). Even worse, we've gotten reports that attaching daynaport before running the configuration causes the network interface to become unreachable. I assume this is due to how rascsi dynamically brings up rascsi_bridge when you attach daynaport (or the host bridge).

rdmark, 2021-11-08:
@akuker This is a kind of crude approach, introspecting the configuration files themselves. A true Linux networking guru can probably do better, but in the meantime it's better than nothing I think:  https://github.com/akuker/RASCSI/pull/428

closed 2021-11-09 by rdmark