PiSCSI/piscsi #408: Manage device reservations as SCSI devices

akuker, 2021-10-30:
Currently, there is special case logic implemented in RaSCSI to handled reserved IDs.

A potential alternative solution to doing this would be to create a new SASI/SCSI controller type that is just a "dummy" object. Then the reservation of the device could be treated in the same way that drives are. We could also store additional information about the reservation and why it was created.

For example, if I wanted to reserve ID 6 for my zip drive, it would be useful to add the reservation through the Web UI. Also having a string to explain why it is reserved would be helpful as well.

If this change is made, the web UI should be updated so that users can add and remove reserved devices, along with provide information about them.

akuker, 2021-10-30:
@uweseimet  - Do you have any thoughts on this?

uweseimet, 2021-10-30:
@akuker A controller is not the same as a drive, so I don't think one can treat them similarly. The fact that parts of the code do not properly make a difference between drives and controllers is still an issue anyway.
IMHO I don't think comments for reserved IDs are needed, because in practice the only reserved ID is usually the initiator ID, and there is only one reason why you reserve it, namely because duplicate IDs are not permitted.

rdmark, 2021-10-30:
@uweseimet FWIW, I've found ID reservation highly useful not just for protecting the initiator ID, but also to keep track of physical SCSI devices that are permanently attached to the system using the RaSCSI (i.e. boot hard drives, CD-ROM drives, Zip drives.) I think the "additional information" that @akuker refers to are conceptually similar to the product information of, say, the Host Bridge device.

Anyhow, this is certainly something that can be handled entirely on the client side by tracking special metadata for ID reservations and store it as part of the configuration. It will just the device configuration logic a bit complex.

uweseimet, 2021-10-30:
@rdmark Not 100% sure what you mean with "product information". If it's the INQUIRY data, these data the backend has to know. But I guess you mean the name that is displayed as filename by rasctl, like "DaynaPort SCSI/Link" for SCDP. If you are referring to this name, I tend to say that this is something the backend also should not know, because it's just something a client presents to the user, but it's currently used in the logs in the "IMAGE FILE" column. Even though it's not an image file ;-).

There should be a clear separation between technical data (belonging to the backend) and data only relevant for humans (belonging into a UI). Instead of "DaynaPort SCSI/Link" a different name "DaynaPort SCSI/Link network adapter", for instance, might be preferred by some users. A UI is the right place to handle something like this, with the backend being agnostic of if. It's a bit like with I18N: The backend can provide abstract information (like SCBR, SCDP), but the frontend has to take care of I18N.

Using the reserved IDs to blacklist IDs for physical devices (this is what you mean with permanently attached, don't you) that's indeed a good use case.

rdmark, 2021-10-31:
> Using the reserved IDs to blacklist IDs of physical devices (this is what you mean with permanently attached, don't you) that's indeed a good use case.

Yep, this is what I meant.

rdmark, 2021-11-04:
Web UI code for managing ID reservation and saving/loading reservations into the same config file as device configurations. https://github.com/akuker/RASCSI/pull/416

rdmark, 2021-11-05:
@akuker Implemented support for reservation memos too! The code should be ready to review.

closed 2021-11-08 by rdmark