PiSCSI/piscsi #411: SN74LS641-1 Substitutes?

ggr8, 2021-11-01:
With the great chip shortage of 2021 now affecting logic chips, is there a “best” substitute for the SN74LS641-1 transceiver?

From the list here:

https://github.com/akuker/RASCSI/wiki/Transceiver-Comparison

SN74ALS245A-1 or SN74F245DW are options. 

Has anyone tried either?

marciot, 2021-11-03:
This is very tangential to the question at hand, but I am also rather curious whether a bidirectional MOSFET based level shifter such as the ones described in this page could work for SCSI:

https://learn.sparkfun.com/tutorials/bi-directional-logic-level-converter-hookup-guide/all

Obviously this would require an entirely new board design, and many, many little SMD components, but it would be amazing to have a sort of "universal" Raspberry Pi adapter that could interface with all sorts of retro things, not just SCSI.

akuker, 2022-01-06:
@marciot - That could be an interesting experiment. With the low cost PCB assembly available now, it isn't very expensive to use a TON of small transistors instead of a pricey IC. Definitely something worth looking into.

In the interest of cleaning things up, I'm going to merge this ticket into #591. They both have the same end goal.

closed 2022-01-06 by akuker