PiSCSI/piscsi #412: Missing header file during easyinstall.sh compilation

marciot, 2021-11-02:
# Info

- Which version of Pi are you using: Raspberry Pi Zero W
- Which github revision of software: v21.10.01
- Which board version: 68kmla Version 2.4a
- Which computer is the RaSCSI connected to: Mac Plus (well, eventually, I have not soldered in the headers to my RaspPi yet)

# Describe the issue

After following the Setup Instructions and running `./easyinstall.sh`, I get the following build error:

```

-- Creating directory obj/fullspec
mkdir -p obj/fullspec
arm-linux-gnueabihf-g++ -O3 -Wall -Werror -DNDEBUG -std=c++17 -Wno-psabi -iquote                                    . -D_FILE_OFFSET_BITS=64 -MD -MP  -DCONNECT_TYPE_FULLSPEC -c rascsi.cpp -o obj/                                   fullspec/rascsi.o
In file included from devices/device_factory.h:18:0,
                 from rascsi.cpp:19:
./rascsi_interface.pb.h:9:42: fatal error: google/protobuf/stubs/common.h: No su                                   ch file or directory
 #include <google/protobuf/stubs/common.h>
                                          ^
compilation terminated.
Makefile:153: recipe for target 'obj/fullspec/rascsi.o' failed
make: *** [obj/fullspec/rascsi.o] Error 1
```

uweseimet, 2021-11-03:
@marciot Looks as if easyinstall does not (anymore) install all required protobuf messages. Can you please run "sudo apt install libprotobuf-dev" and then run easyinstall again? My assumption is that the missing header files are part of the libprotobuf-dev package.

marciot, 2021-11-03:
This seems related to "https://github.com/gabime/spdlog/issues/1088". I tried ```sudo apt-get install libspdlog-dev``` and I get the message ```libspdlog-dev is already the newest version (1:0.11.0-2)```

As is mentioned in that ticket, a newer version of the library is needed, but it does not seem to exist for the RaspPi

uweseimet, 2021-11-03:
@marciot More packages appear to be missing in easyinstall. I would have guessed that libspdlog-dev fixes your issue, but looks as if did not help.

@rdmark @akuker Looks as if fresh installations do not work. I created a release branch names release_v21.10, based on v21.10.1. Any bugfixes for the 21.10 release should be committed and tagged on this branch, and this branch should only be used for bug fixes but nothing else. Changes most likely need to be cherry-picked into the develop branch.
The release_v21.10 branch already contains the fix for the missing libprotobuf-dev dependency, but it appears as if other dependencies are also missing. I suggest that you set up a fresh Pi OS installation for testing easyinstall.

marciot, 2021-11-03:
Okay, it looks like the crux of the problem is that I am running Debian "stretch" on my pi, while the current version is "buster". This may well explain my difficulties building the RaSCSI software. I think I'll need to set aside some time to upgrade or rebuild my Pi from scratch :grin:

uweseimet, 2021-11-03:
@akuker @rdmark @marciot Looks as if this was an OS version issue after all. Still, we should retain the release_v21.10 for potential release fixes.
@akuker The new issue template does not ask for the OS version, maybe this information should be requested.

marciot, 2021-11-03:
@uweseimet: Okay. I'll give the upgrade steps a try and will report on my progress once it is done.

rdmark, 2021-11-04:
@uweseimet Thanks for following up on this issue. Well noted on the bugfix branch.

@marciot Let us know if you run into any other problems. You may want to join us in the Discord channel for realtime troubleshooting help. Follow the link in https://github.com/akuker/RASCSI/wiki#want-to-chat

uweseimet, 2021-11-04:
@rdmark This issue was closed without a comment?

rdmark, 2021-11-04:
@uweseimet that's odd, you don't see the comment I left yesterday?

uweseimet, 2021-11-04:
@rdmark I don't see any comment that @marciot's issue is resolved. Am I missing somthing?

rdmark, 2021-11-04:
@uweseimet Oh, I just read that the original issue of missing libraries was due to using an old OS version, so I considered it resolved as 'Won't Fix'.

@marciot Don't hesitate to keep commenting on this ticket, or reopen it if necessary!

uweseimet, 2021-11-04:
@rdmark I see. I was asking why the ticket was closed because I was expecting feedback saying that updating the OS resolved thie issue, or whether something else popped up I could follow up on.

marciot, 2021-11-04:
It might take me a while to report back. I tried and in-place upgrade, and it was mostly progressing okay, but then I ran out of space on the card and ended up with an unbootable system. I'll need to rebuild the RaspPi from scratch, but first I'll need to save some of my configuration files from my previous build.

uweseimet, 2021-11-04:
@marciot Thank you for updating the ticket status. Just take your time.

One general question regarding your upgrade approach: I usually run Ubuntu and not Raspberry Pi (which is a quite outdated version of Debian) on my Pi. Ubuntu is essentially also Debian-based, and when recently upgrading my Pi to the latest Ubuntu I could do so without saving or rebuilding anything. I would expect that you can also upgrade Raspberry Pi OS without installing it from scratch and losing data, but at least in your case this does not seem to be possible?

marciot, 2021-11-04:
@uweseimet: I am running the Raspberry Pi version, alas. I did find instructions on how to do an upgrade, and they seemed to have been running okay, but unfortunately I only had a 8GB card in my Pi, and apparently it filled up. I managed to break into a recovery shell and clean up a bit and it seems like the upgrade is proceeding now. I'll probably let it finish, and if it does, test out RaSCSI. Then I'll probably swap out the card for a 32GB one and redo everything, because I know I'll run into problems again in the future with how little space I have left.

Anyhow, I kind of wish someone would port RaSCSI to something like an ESP32 so we didn't have to deal with running a full-scale Linux and all the administration chores that come with that. Maybe I'll try wiring a ESP32 to my RaSCSI board with some jumper wires and see if I can do a proof of concept with that! :grin:

uweseimet, 2021-11-04:
@marciot I know the SD card space issue quite well ;-). Some time ago I replaced my 16 GB cards by 32 GB cards, in particular because one of my images has a size of 9 GB. Now I use two 16 GB cards for Raspberry Pi OS (32 and 64 bit) and the 32 GB cards for 32 and 64 bit Ubuntu.
Regarding administration usually there is not much more to do than running apt update/upgrade now and then. From my perspective the advantage of running Linux is that one can also do other things with the same Pi than running rascsi.

marciot, 2021-11-05:
So after some fiddling, I was able to successfully complete the upgrade from RaspOS "stretch" to "buster" on my 8GB card.

I ran "easyinstall.sh" and it completed without a hitch. I haven't tested RaSCSI yet, because I still have to do some soldering, but since this ticket pertained to "easyinstall.sh", and it now runs, I am closing this ticket.

To recap, this is what I have confirmed:

- "easyinstall.sh" does not work with "stretch" due to a missing "stdout_color_sinks.h" 
- Upgrading from "stretch" to "buster" **does** resolve the problem.

I more or less used this guide https://pimylifeup.com/upgrade-raspbian-stretch-to-raspbian-buster for the upgrade, but had to resolve several issues due to low disk space and/or incompatible packages. None of this really has any bearing on RaSCSI and are just quirks of my own setup.

closed 2021-11-05 by marciot