PiSCSI/piscsi #417: updated web frontend breaks akai sampler cdrom support

ewanuno, 2021-11-04:
# Info

RaSCSI version: 21.10.1 cf9a578

# Describe the issue
in the updated web frontend it is no longer possible to load an .iso as a .hda,  since it now automaticaly detects the image type.
for akai samplers this is a problem, loading an iso as a cdrom device does not work.

in the old version i could load any file as any device type, so i simply loaded my akai cdrom .iso file as .hda

i have found a  workarround: just rename the .iso files to .iso.hda

rdmark, 2021-11-04:
@ewanuno May I ask which version of the RaSCSI software you were using prior to this? Even the previous stable version, 21.05, had device type auto detection based on file ending, so I'm just wondering how far back I should go to see what the old workflow was like.

Anyhow, the Akai scenario is a very unusual one, and we have deliberately decided to not enable any image to be attached as any device type in the Web UI, since it would open up for user error that could easily lead to data loss. Would it introduce a lot of inconvenience for you to use the renaming workaround moving forward?

BTW, just for the record, your workaround is already documented on the [Samplers wiki page](https://github.com/akuker/RaSCSI/wiki/RaSCSI-for-Hardware-Samplers#compatibility) in the Akai section.

rdmark, 2021-11-04:
@ewanuno One thing that you can try, if you'd like to help us understand the needs of the Akai platform, is to try mounting the iso as a CD-ROM drive with 512 bytes block size. By default, iso images are mounted with a 2048 bytes block size (with is the standard for CD-ROM). By my understanding, one of the quirks of the Akai is that it expects either 1 or 512 bytes block size, which is non-standard. Do this on the command line:

rasctl -i 0 -c attach -b 512 -f /path/to/iso

The above assumes SCSI ID 0. Change the -i parameter if needed.

If the above allows you to use the iso on the Akai, I can look into a solution in the Web UI.

rdmark, 2022-08-19:
Closing, since workaround exists, and changing the default behavior would make the UX of the web UI less convenient for most users and open up for incorrect operation.

There is always the option to create a Web UI specifically for Samplers, which is something that has been discussed in the past.

hayesey, 2022-08-24:
This is a shame, I ran into the same issue. I think a lot of people are using these with Akai samplers. I would have thought having a dropdown box in the GUI with some options for block size of ISOs that defaults to 2048 would be acceptable. Appreciate it is a corner case.

I'd be happy to do a pull request but if you think it's not a good idea I'll leave it alone.

rdmark, 2022-08-24:
@hayesey So I've gotten conflicting messages on the requirements for Akai. When discussed on Discord last year, someone told me that it's not enough to mount a CD-ROM device with 512 block size and attach the iso image, but that the iso image actually has to be mounted as a hard drive for the host to recognize it. Is this not in line with your experience with Akai devices?

hayesey, 2022-08-24:
@rdmark this command you posted earlier on:

rasctl -i 0 -c attach -b 512 -f /path/to/iso

Works, I ran it with 6 different Akai Sample CD ISOs earlier on with my Akai S3000xl. Refreshing the web UI shows them as CDROMs. 

Could well be different on the older samplers such as 950 or 1000 (in fact I'm not sure those even have SCSI interfaces though).

rdmark, 2022-08-24:
@rdmark Ok this is good to know.

Just to throw an alternative idea out there: How about we introduce a new "bespoke" CD-ROM profile, with an arbitrary vendor/product name and the block size set to 512?

```
{
    "device_type": "SCCD",
    "vendor": "RaSCSI",
    "product": "SCSI CD-ROM",
    "revision": "",
    "block_size": 512,
    "size": null,
    "name": "Generic CD-ROM 512 block size",
    "file_type": null,
    "description": "Used to mount iso images to hosts that expect block size 512, such as Akai samplers.",
    "url": ""
},
```

Then you'd go though the bespoke device UI to generate a properties file for each of the iso images that you use with Akai. Thoughts?

rdmark, 2022-08-25:
@hayesey Would you be able to test this PR? https://github.com/akuker/RASCSI/pull/808

hayesey, 2022-08-25:
@rdmark I can confirm that this change works. Creating a Named Drive from an existing uploaded ISO using this template creates a CD-ROM with 512byte block size and it successfully loads on my Akai S3000XL.

rdmark, 2022-08-25:
Thanks for testing! I'll get this added to the next release.

rdmark, 2022-08-26:
The change is in develop

closed 2022-08-26 by rdmark