PiSCSI/piscsi #420: [Troubleshooting] My canary is starving! :sob:

marciot, 2021-11-06:
# Info

- Which version of Pi are you using: Raspberry Pi Zero W
- Which github revision of software:   October 2021 Release
- Which board version: 2.4a
- Which computer is the RaSCSI connected to: Mac Plus

# Describe the issue

Hey guys, thanks for the help getting me sorted out with the `./easyinstall.sh`. I am now making some progress, but RaSCSI seems somewhat unreliable.

First off, a question to make sure I didn't mess this part up:

 -- How do I properly configure termination? There were two switches on the board, but I can't find info on whether to toggle one or both. Since I am daisy chaining RaSCSI after another SCSI device, I guessed that both switches needed to be on.
 
Anyhow, I was able to get my Mac to recognize a drive, but eventually it froze during a copy. This is what I get in the logs:

```
Nov 05 17:55:41 raspberrypi RASCSI[367]: [2021-11-05 17:55:41.921] [info] Validating: operation=ATTACH, device id=6, unit=0, type=SCHD, device params='file=System_7_Boot.hds', vendor='', product='', revision='', block size=0
Nov 05 17:55:41 raspberrypi RASCSI[367]: [2021-11-05 17:55:41.924] [info] Executing: operation=ATTACH, device id=6, unit=0, type=SCHD, device params='file=System_7_Boot.hds', vendor='', product='', revision='', block size=0
Nov 05 17:55:41 raspberrypi RASCSI[367]: [2021-11-05 17:55:41.925] [info] Attached SCHD device, ID 6, unit 0
Nov 05 18:06:58 raspberrypi kernel: sched: RT throttling activated
Nov 05 18:06:58 raspberrypi rtkit-daemon[721]: The canary thread is apparently starving. Taking action.
Nov 05 18:06:58 raspberrypi RASCSI[367]: [2021-11-05 18:06:52.757] [error] virtual void SCSIDEV::Receive() Not able to receive all data. Going to error
Nov 05 18:06:58 raspberrypi rtkit-daemon[721]: Demoting known real-time threads.
Nov 05 18:06:58 raspberrypi rtkit-daemon[721]: Successfully demoted thread 720 of process 720 (n/a).
Nov 05 18:06:58 raspberrypi rtkit-daemon[721]: Demoted 1 threads.
Nov 05 18:10:03 raspberrypi systemd[1]: Starting Cleanup of Temporary Directories...
Nov 05 18:10:04 raspberrypi systemd[1]: systemd-tmpfiles-clean.service: Succeeded.
Nov 05 18:10:04 raspberrypi systemd[1]: Started Cleanup of Temporary Directories.
```

I suspect "**RT throttling activated**", "**The canary thread is apparently starving.** Taking action." and "**Not able to receive all data. Going to error**" are all bad things :cry:

Anyhow, just for fun, here is a picture of my first test using RaSCSI. This is gonna rock once all the kinks are worked out! :grin:

[image]

marciot, 2021-11-06:
I have been disabling a bunch of services and uninstalling things, and while I am getting further in the copy than before, it still eventually fails with this error. Is the Raspberry Pi Zero W sufficiently powerful for RaSCSI?

Is there anyway to format a disc so that the Mac Plus won't write to it as fast? For some odd reason, my go-to formatting tool "Special Apple HD Setup 7.3.5" was not recognizing the RaSCSI disk, so I ended up using something called "Blue Disk Manager". I wonder if that has something to do with the problem. What is the recommended tool for 68k Macs?

Another weird symptom: If I try booting my Mac Plus with a RaSCSI disk attached, the Mac Plus wigs out, makes a strange sound, and the screen goes all garbled. Detaching the RaSCSI disk allows it to start.

rdmark, 2021-11-06:
> 
> 
> I have been disabling a bunch of services and uninstalling things, and while I am getting further in the copy than before, it still eventually fails with this error. Is the Raspberry Pi Zero W sufficiently powerful for RaSCSI?
> 
@marciot Sorry to hear that you're having some challenges here. Let us help you work through them! I have two RaSCSI setups with original Zero Ws and it works perfectly with my SE and other later Macs, so this should not be the blocker at all.

> Is there anyway to format a disc so that the Mac Plus won't write to it as fast? For some odd reason, my go-to formatting tool "Special Apple HD Setup 7.3.5" was not recognizing the RaSCSI disk, so [...]

The Mac Plus has an early non-standards compliant SCSI implementation, so it is known to be picky about hard disk drivers in particular (not limited to RaSCSI.) The one driver to avoid with the Plus is Lido, but I don't think Blue Disk Manager is a known compatible one either.

The driver that I personally like to use is Silverlining, but to be safe you can use patched versions of Apple's tools to use the official drivers that are a bit slower but perhaps more compatible (perhaps.)

See the wiki for more info: https://github.com/akuker/RASCSI/wiki/Drive-Setup#Initializing_a_new_disk_on_Classic_Mac_OS

> Another weird symptom: If I try booting my Mac Plus with a RaSCSI disk attached, the Mac Plus wigs out, makes a strange sound, and the screen goes all garbled. Detaching the RaSCSI disk allows it to start.

This sounds suspiciously like how the Plus behaves when you attempt to boot off a drive using the Lido drivers, as mentioned above. See https://github.com/akuker/RASCSI/issues/40 for details. Did you by any chance use the easyinstall.sh to create the drive image you're trying to boot off?

marciot, 2021-11-06:
Okay, this is weird:

1. Using "Lido 7" to format the disk rather than "Blue Disk Manager" allows me to copy files to RaSCSI without triggering the RT Throttling.
2. Turning off my external HD (making it so the RaSCSI is the only powered on device in the SCSI chain), allows my Mac to boot properly from RaSCSI.
3. Once booted, I can power on my APS HD, use SCSIProbe 3.5 to mount it, and copy files back and forth with no issues.

So, even though it seems to involve some weird workarounds, I now have a way to boot my Mac Plus from RaSCSI and copy files to and from my external HD. So I consider it a success! :grin:

BTW, this is they type of screen I get when I try booting my Mac with RaSCSI and my external HD powered on. It's like a fractal pattern, kind of pretty:

[image]

marciot, 2021-11-06:
@rdmark: I posted that before I read your message. It's funny that so far I have made more progress with Lido than I did with Blue Disk Manager. I'll have to try Silverlining.

I believe I have the patched version of Apple HD Setup, what I have is called "Special Apple HD Setup 7.3.5", but it refuses to see the RaSCSI drive at all :shrug:

marciot, 2021-11-06:
So I am having difficulties coming up with a strategy for copying Silverlining to my Mac Plus. It doesn't look like RaSCSI disk images can be mounted in Basillisk II. Is there any other way to mount RaSCSI disk images so that you can copy files directly into it?

In the past I have used a Zip drive on a PowerBook to transfer files to my Mac Plus, but lately it has been suffering from clicks-of-death, so I would rather avoid using it as much as possible.

marciot, 2021-11-06:
Nevermind, I got Basillisk II to see the drive.

marciot, 2021-11-06:
@rdmark: I was able to format my RaSCSI partitions with Silver Lining. My Mac Plus still gets a garbled screen when I try booting with a RaSCSI device attached and my external drive powered on. I can boot fine when my external drive is powered off, and I can always use SCSI Probe to mount my external drive after the fact. It's pretty much the same result I was getting with Lido 7.

rdmark, 2021-11-06:
@marciot The question that you always have to ask:  Have you made sure the SCSI bus is terminated correctly? What is the order of the devices in the chain, and how is it terminated?

marciot, 2021-11-06:
@rdmark: I have the Mac Plus, connected to one SCSI connector on my external hard drive (termination switch off); the other SCSI connector on the external hard drive is connected to the RaSCSI (termination SW on).

marciot, 2021-11-06:
To wrap up this ticket, I have confirmed that the "RT throttling activated" and "The canary thread is apparently starving" problem only happens when using "Blue Disk Manager". Something in particular about that disk driver must put RaSCSI into a state where the real time task fails or takes too long to execute. While this may indicate a latent bug in the RaSCSI code, I imagine it isn't a high priority as Lido and Silver Lining seem to work just fine.

The boot/termination issue is unrelated to this ticket and is something I am still working on. The developer's can close this ticket as "Won't fix," if they do not wish to follow up on the problem with Blue Disk Manager.

rdmark, 2021-11-06:
@marciot Thanks for reporting back and providing all this context! Others can learn from this and not choose Blue Disk Manager with their Plus. :)

The two things I can think of regarding the problem when using RaSCSI with your external drive are:  Poor contacts in the SCSI cable and ports. You can try to get some contact cleaner solution in there and work it in. I've had weird SCSI instability issues with my old Macs that were simply due to oxidation on contacts leading to signals not getting through properly.

And secondly, the power supply on your external hard drive may be failing and providing out of spec voltages on the SCSI bus. You may want to open it up and test the voltage rails under load (while being careful with mains voltages, of course. ;) )

Anyhow, while I have your attention, may I ask you a favor? Did you by any chance test the DaynaPort ethernet adapter emulation with your Plus? We've had reports that it's not working properly, but I don't have access to a Plus myself. If you're up for it, I can walk you through how to get started. And besides, it's super neat to get your old Mac online, if just for the novelty factor.

marciot, 2021-11-06:
@rdmark: Yes, I hope to eventually work on getting DaynaPort running, but I need a break :) I already had a super busy day fiddling with retro hardware. Found out my Iomega Zip 100 Plus (SCSI + Parallel) had succumbed to the dreaded click of death, so I ended up taking the drive mechanism out of a Zip 100 drive (parallel-only) and transplanting it into the Zip 100 Plus. I then went down a rabbit-hole trying to get the Zip testing utility [Trouble in Paradise](https://www.grc.com/tip/clickdeath.htm) to test my refurbished drive. I installed SoftWindows 95 on a PowerBook 3400c and hooked up the Zip drive, and even though I could run TIP, it failed to see the SCSI Zip 100 Plus drive, even though SoftWindows 95 is *supposed* to allow access to SCSI devices. So now I am pretty much stuck trying to find a PC with a parallel port so I can test my refurbished drive using TIP. Oh wey!

rdmark, 2021-11-06:
@marciot Take your time! Once you get around to it, I'd love to hear back from you about your success or failure. As I mentioned in the other ticket, you can join the Discord channel for real-time conversations with other RaSCSI users.

closed 2021-11-06 by rdmark

marciot, 2021-11-08:
@rdmark: I began fiddling with the DynaPort drivers. See my comment on the thread #419