PiSCSI/piscsi #422: Lack of validation of duplicate use of removable media with INSERT protobuf command

rdmark, 2021-11-07:
With the latest develop code:

When using the INSERT protobuf command, rascsi allows you to insert the same removable image file into multiple devices.
This is not reproducible when using rasctl, but can be triggered by using the webapp (currently easily done with the code in https://github.com/akuker/RASCSI/pull/421 )

> $ rasctl -l
> +----+-----+------+-------------------------------------
> | ID | LUN | TYPE | IMAGE FILE
> +----+-----+------+-------------------------------------
> |  3 |   0 | SCMO | /home/pi/images/foobar1.mos
> |  4 |   0 | SCMO | /home/pi/images/foobar1.mos
> +----+-----+------+-------------------------------------

rdmark, 2021-11-07:
@uweseimet What are you thoughts here? I haven't looked at the code yet, but rascsi does effective validation when you use the ATTACH command, but seemingly not with the INSERT command.

uweseimet, 2021-11-07:
@rdmark Like always, it would be good to have trace logs that show in more detail what's happening.

rdmark, 2021-11-07:
@uweseimet Here you go. Note that the repeated DEVICES_INFO command is due to the OLED monitor script polling rascsi.

```
Nov 07 07:10:03 rascsi3b RASCSI[552]: [2021-11-07 07:10:03.986] [trace] Received DEVICES_INFO command
Nov 07 07:10:04 rascsi3b RASCSI[552]: [2021-11-07 07:10:04.734] [trace] Received DEVICES_INFO command
Nov 07 07:10:04 rascsi3b RASCSI[552]: [2021-11-07 07:10:04.740] [info] Validating: operation=INSERT, device id=6, unit=0, type=SCCD, device params='file=Adobe_Photoshop_4.iso', vendor='', product='', revision='', block size=0
Nov 07 07:10:04 rascsi3b RASCSI[552]: [2021-11-07 07:10:04.740] [info] Executing: operation=INSERT, device id=6, unit=0, type=SCCD, device params='file=Adobe_Photoshop_4.iso', vendor='', product='', revision='', block size=0
Nov 07 07:10:04 rascsi3b RASCSI[552]: [2021-11-07 07:10:04.740] [info] Insert file 'Adobe_Photoshop_4.iso' requested into SCCD ID 6, unit 0
Nov 07 07:10:04 rascsi3b RASCSI[552]: [2021-11-07 07:10:04.769] [trace] Received SERVER_INFO command
Nov 07 07:10:04 rascsi3b RASCSI[552]: [2021-11-07 07:10:04.812] [trace] Received DEVICES_INFO command
Nov 07 07:10:04 rascsi3b RASCSI[552]: [2021-11-07 07:10:04.816] [trace] Received DEVICE_TYPES_INFO command
Nov 07 07:10:04 rascsi3b RASCSI[552]: [2021-11-07 07:10:04.821] [trace] Received DEFAULT_IMAGE_FILES_INFO command
Nov 07 07:10:04 rascsi3b RASCSI[552]: [2021-11-07 07:10:04.836] [trace] Received SERVER_INFO command
Nov 07 07:10:04 rascsi3b RASCSI[552]: [2021-11-07 07:10:04.940] [trace] Received NETWORK_INTERFACES_INFO command
Nov 07 07:10:05 rascsi3b RASCSI[552]: [2021-11-07 07:10:05.092] [trace] Received DEVICES_INFO command
Nov 07 07:10:05 rascsi3b RASCSI[552]: [2021-11-07 07:10:05.095] [trace] Received DEVICES_INFO command
Nov 07 07:10:06 rascsi3b RASCSI[552]: [2021-11-07 07:10:06.173] [trace] Received DEVICES_INFO command
Nov 07 07:10:07 rascsi3b RASCSI[552]: [2021-11-07 07:10:07.289] [trace] Received DEVICES_INFO command
Nov 07 07:10:07 rascsi3b RASCSI[552]: [2021-11-07 07:10:07.479] [trace] Received DEVICES_INFO command
Nov 07 07:10:07 rascsi3b RASCSI[552]: [2021-11-07 07:10:07.484] [info] Validating: operation=INSERT, device id=7, unit=0, type=SCCD, device params='file=Adobe_Photoshop_4.iso', vendor='', product='', revision='', block size=0
Nov 07 07:10:07 rascsi3b RASCSI[552]: [2021-11-07 07:10:07.484] [info] Executing: operation=INSERT, device id=7, unit=0, type=SCCD, device params='file=Adobe_Photoshop_4.iso', vendor='', product='', revision='', block size=0
Nov 07 07:10:07 rascsi3b RASCSI[552]: [2021-11-07 07:10:07.484] [info] Insert file 'Adobe_Photoshop_4.iso' requested into SCCD ID 7, unit 0
Nov 07 07:10:07 rascsi3b RASCSI[552]: [2021-11-07 07:10:07.506] [trace] Received SERVER_INFO command
Nov 07 07:10:07 rascsi3b RASCSI[552]: [2021-11-07 07:10:07.543] [trace] Received DEVICES_INFO command
Nov 07 07:10:07 rascsi3b RASCSI[552]: [2021-11-07 07:10:07.549] [trace] Received DEVICE_TYPES_INFO command
Nov 07 07:10:07 rascsi3b RASCSI[552]: [2021-11-07 07:10:07.557] [trace] Received DEFAULT_IMAGE_FILES_INFO command
Nov 07 07:10:07 rascsi3b RASCSI[552]: [2021-11-07 07:10:07.583] [trace] Received SERVER_INFO command
Nov 07 07:10:07 rascsi3b RASCSI[552]: [2021-11-07 07:10:07.697] [trace] Received NETWORK_INTERFACES_INFO command
Nov 07 07:10:08 rascsi3b RASCSI[552]: [2021-11-07 07:10:08.409] [trace] Received DEVICES_INFO command
Nov 07 07:10:08 rascsi3b RASCSI[552]: [2021-11-07 07:10:08.415] [trace] Received DEVICES_INFO command
Nov 07 07:10:09 rascsi3b RASCSI[552]: [2021-11-07 07:10:09.540] [trace] Received DEVICES_INFO command

```

uweseimet, 2021-11-07:
@rdmark I could reproduce this issue with rasctl after providing an explicit default folder when starting rascsi (-F option) and then running these commands without getting an error:

```
>rasctl -i 6 -u 0 -t sccd -c a
>rasctl -i 7 -u 0 -t sccd -c a
>rasctl -i 6 -u 0 -c i -f test.hds
>rasctl -i 7 -u 0 -c i -f test.hds
```

The issue should be fixed with the fix_issue_422 branch and was related to checking the filename before it was expanded to the full name, including the default folder path. Can you please verify?

rdmark, 2021-11-07:
@uweseimet Thanks for looking into this! I can confirm that rascsi now returns an error when you do the same, but oddly enough it still allows the image to be attached, at least when I use the web app. I will try to reproduce with rasctl as well, but in the meantime:

[image]

```
 $ rasctl -l
+----+-----+------+-------------------------------------
| ID | LUN | TYPE | IMAGE FILE
+----+-----+------+-------------------------------------
|  5 |   0 | SCCD | /home/pi/images/Adobe_Photoshop_4.iso (READ-ONLY)
|  6 |   0 | SCCD | /home/pi/images/Adobe_Photoshop_4.iso (READ-ONLY)
+----+-----+------+-------------------------------------
```

rdmark, 2021-11-07:
@uweseimet Reproduces also with rasctl. Could you have a look?

```
pi@rascsi3b:~/RASCSI $ sudo rascsi -F /home/pi/images/ &
[1] 1843
pi@rascsi3b:~/RASCSI $ SCSI Target Emulator RaSCSI(*^..^*) version 21.99 --DEVELOPMENT BUILD-- (Nov  7 2021, 09:00:52)
Powered by XM6 TypeG Technology / Copyright (C) 2016-2020 GIMONS
Copyright (C) 2020-2021 Contributors to the RaSCSI project
Connect type : FULLSPEC
[2021-11-07 09:39:18.467] [info] Default image folder set to '/home/pi/images/'
[2021-11-07 09:39:18.468] [info] No images currently attached.
No images currently attached.

pi@rascsi3b:~/RASCSI $ rasctl -i 6 -u 0 -t sccd -c a
[2021-11-07 09:39:22.024] [info] Validating: operation=ATTACH, device id=6, unit=0, type=SCCD, vendor='', product='', revision='', block size=0
[2021-11-07 09:39:22.025] [info] Executing: operation=ATTACH, device id=6, unit=0, type=SCCD, vendor='', product='', revision='', block size=0
[2021-11-07 09:39:22.025] [info] Attached read-only SCCD device, ID 6, unit 0
pi@rascsi3b:~/RASCSI $ rasctl -i 7 -u 0 -t sccd -c a
[2021-11-07 09:39:27.329] [info] Validating: operation=ATTACH, device id=7, unit=0, type=SCCD, vendor='', product='', revision='', block size=0
[2021-11-07 09:39:27.329] [info] Executing: operation=ATTACH, device id=7, unit=0, type=SCCD, vendor='', product='', revision='', block size=0
[2021-11-07 09:39:27.329] [info] Attached read-only SCCD device, ID 7, unit 0
pi@rascsi3b:~/RASCSI $ rasctl -i 6 -u 0 -c i -f Adobe_Photoshop_4.iso
[2021-11-07 09:39:50.867] [info] Validating: operation=INSERT, device id=6, unit=0, type=UNDEFINED, device params='file=Adobe_Photoshop_4.iso', 'interfaces=Adobe_Photoshop_4.iso', vendor='', product='', revision='', block size=0
[2021-11-07 09:39:50.867] [info] Executing: operation=INSERT, device id=6, unit=0, type=UNDEFINED, device params='file=Adobe_Photoshop_4.iso', 'interfaces=Adobe_Photoshop_4.iso', vendor='', product='', revision='', block size=0
[2021-11-07 09:39:50.867] [info] Insert file 'Adobe_Photoshop_4.iso' requested into SCCD ID 6, unit 0
pi@rascsi3b:~/RASCSI $ rasctl -i 7 -u 0 -c i -f Adobe_Photoshop_4.iso
[2021-11-07 09:40:10.862] [info] Validating: operation=INSERT, device id=7, unit=0, type=UNDEFINED, device params='file=Adobe_Photoshop_4.iso', 'interfaces=Adobe_Photoshop_4.iso', vendor='', product='', revision='', block size=0
[2021-11-07 09:40:10.862] [info] Executing: operation=INSERT, device id=7, unit=0, type=UNDEFINED, device params='file=Adobe_Photoshop_4.iso', 'interfaces=Adobe_Photoshop_4.iso', vendor='', product='', revision='', block size=0
[2021-11-07 09:40:10.862] [info] Insert file 'Adobe_Photoshop_4.iso' requested into SCCD ID 7, unit 0
[2021-11-07 09:40:10.862] [error] Image file 'Adobe_Photoshop_4.iso' is already used by ID 6, unit 0
Error: Image file 'Adobe_Photoshop_4.iso' is already used by ID 6, unit 0
pi@rascsi3b:~/RASCSI $ rasctl -l
+----+-----+------+-------------------------------------
| ID | LUN | TYPE | IMAGE FILE
+----+-----+------+-------------------------------------
|  6 |   0 | SCCD | /home/pi/images//Adobe_Photoshop_4.iso (READ-ONLY)
|  7 |   0 | SCCD | /home/pi/images//Adobe_Photoshop_4.iso (READ-ONLY)
+----+-----+------+-------------------------------------

```

uweseimet, 2021-11-07:
@rdmark Can you try again with the latest code in fix_issue_422? Looks as if I cannot prevent some code duplication ...

rdmark, 2021-11-07:
@uweseimet This did the trick, thanks! Let me merge that code and close this ticket out.

uweseimet, 2021-11-07:
@rdmark PR is available. Maybe you can also have a look at my other PR (for the develop branch).

uweseimet, 2021-11-07:
@rdmark I re-created the PR, because I created the bugfix branch against develop instead of the bugfix branch. After merging everything into develop I will cherry-pick the actual bugfix into the bugfix branch.

rdmark, 2021-11-07:
PR reviewed & merged. Closing this ticket.

uweseimet, 2021-11-08:
@rdmark This ticket is still open?

rdmark, 2021-11-08:
@uweseimet Closing it for real now ;)

closed 2021-11-08 by rdmark