PiSCSI/piscsi #427: Can unzip provide some indication that it is doing something?

Pacjunk, 2021-11-07:
# Info

- Which version of Pi are you using: ZeroW
- Which github revision of software: develop
- Which board version: FULLSPEC
- Which computer is the RaSCSI connected to: Doesn't matter. Web GUI

# Describe the issue

Unzipping large files or ISOs can take some time. When I click unzip, nothing appears from the interface - the only indication I get is the spinning marker on the browser that it is waiting. Is it possible to add some text somewhere to say something like "UnZIPping, please wait" or something like that? or even a prompt after you click the button? or an hourglass wait? At the moment unless you notice the browser waiting, you wouldn't know whether the button press even registered. Some feedback might prevent users from clicking the button multiple times (which the interface allows).

Thanks,

rdmark, 2021-11-08:
@Pacjunk This is a good improvement suggestion. As an experiment, I tried to simply add a 'flash' message early in the method that kicks off the process, but the problem is that it starts upzipping that locks the Python process, and the message doesn't have time to be displayed until after unzipping is finished. I think the solution is to make the unzipping process itself asynchronous so that it doesn't lock the Python process, but this would require some non-trivial coding. It can be done for sure, it just wasn't as easy as I hoped...

Pacjunk, 2021-11-08:
I don't mind it pausing the python process, just didn't know whether it was possible to flag that it was busy somehow. In the windows world, the cursor is changed to an hourglass. Something like this would be good. At the moment you click on the unzip button which starts the process, but there is nothing stopping you from pressing it several times - then you get an error message later saying it can't overwrite existing file.
Anyway, just thought I would flag it - not worth a lot of effort!

rdmark, 2021-11-27:
@Pacjunk Potential solution in _rdmark_webapp_async_ branch.  To achieve something close to what you were asking for, this is quite hackishly modifying the innerHTML of the flash message using Javascript.  A cleaner solution would probably involve making the webapp a truly threaded application, which seems overkill at this moment...

Note that this also introduces an async operation for unzipping, so even if you navigate away from the page, the unzip process will continue in the background.

rdmark, 2021-11-27:
Code is in develop now.

@Pacjunk Please test if you get the chance. We'll be cutting a new release within 24h or so.

Pacjunk, 2021-11-27:
@rdmark Seems to be OK, although you still have to scroll up to the top to see the message, so I'm not getting that feedback. I was thinking of something a bit more "in your face" like a popup. e.g. "Unzipping xxx - please wait" etc. You get pop-ups for deleting images, maybe just something similar for unzip.

rdmark, 2021-11-28:
@Pacjunk How about the solution in https://github.com/akuker/RASCSI/pull/493 -- the viewport is shifted to the top of the page for long-running processes, to put attention to the flash message.

Pacjunk, 2021-11-28:
It seems to work OK. Has shifting the viewport always been there for delete/attach etc? or have I only just noticed it?

Bit of a toss-up for the best method. Extra click for the popup versus scrolling back down to unzip another one (although I suppose that is less likely to happen).

Doesn't really matter either way, at least you can see something is happening. I'm sure people will comment if they don't like it (or are we the only ones interested?).  Thanks,

rdmark, 2021-11-28:
@Pacjunk Yes, the shifting of the viewport to the top after each action is how this webapp always was supposed to behave. The UI is designed in a way that it takes one action at a time, and once the action is completed the page refreshes and a flash message is displayed in the header. This concept is complicated by long-running actions, hence this ticket.

closed 2021-11-28 by rdmark