PiSCSI/piscsi #429: [Feature Request] Mounting Floppy Images?

marciot, 2021-11-08:
I know SCSI disks and floppies are totally different things (AFAIK the SCSI drive image has a boot record, a driver and partition table), but I wonder if you could start with a blank SCSI disk image of the appropriate size and have RaSCSI copy the data from the floppy into the SCSI disk image at just the right place so it over-writes the empty partition and presents a valid SCSI drive. There is a bit of complexity involved in dealing with the various types of disk images, such as header-less disk images or DiskCopy 4.2 disk images, but I think those issues can be solved (I've written code to translate from one type of disk image to another before).

marciot, 2021-11-08:
PS: I know it's kind of rude to ask for niche features, as if the developers don't already have enough on their plates, maybe I should re-phase this as "what could I do to help come up with something like this?"

I think I could probably try writing some C or Python code that converts a floppy disk image into a SCSI disk image, especially if I can find the appropriate documentation of the SCSI disk layout. If I can come up with some code that does it, could it be incorporated into RaSCSI easily?

I notice the feature matrix suggests the SCSI2SD is the only device that currently does this. Does anyone know how they do it or whether their code is open source?

rdmark, 2021-11-08:
@marciot This is certainly an interesting feature idea. What are the use cases that you have in mind? Booting software that can only run off a floppy boot disk?

Of interest to you may be the research done on the [Iomega Floptical](https://github.com/akuker/RASCSI/wiki/Iomega-Floptical) SCSI drive that also was compatible with standard magnetic floppy disks. I played around with it briefly a few weeks ago, and was able to get the Iomega drivers to load and recognize the drive, but that was as far as I got. I didn't try very hard though, to be honest.

You can play with it by attaching a removable drive with the correct INQUIRY strings, f.e.:

rasctl -i 2 -c attach -t scrm -n "IOMEGA:Io20S         *F:PP33" -f /path/to/image/file

Pacjunk, 2021-11-08:
This reminds me, I should add the DEC SCSI floppy drives to the drive properties table...  (not that they're terribly usable, but just for completeness)

marciot, 2021-11-10:
> @marciot This is certainly an interesting feature idea. What are the use cases that you have in mind? Booting software that can only run off a floppy boot disk?

Well, I'm fairly certain that if software can't run off a floppy boot disk, such as something like Dark Castle, then it won't boot of a SCSI device because of the SCSI driver. Rather, my use case is that I spend a lot of time using disk images for emulators, such as vMac or PCE (I am the curator of the [RetroWeb Vintage Computer Museum](http://retroweb.maclab.org/)). Having to convert all my pre-existing image files is rather time consuming, so being able to load them into RaSCSI immediately would be amazing.

> Of interest to you may be the research done on the Iomega Floptical SCSI drive that also was compatible with standard magnetic floppy disks. 

I'll play around with it, but if it requires extra drivers then it would not exactly be compatible with pre-made disk images.

Is there are reason this device isn't available through the web UI yet?

rdmark, 2021-11-10:
> Is there are reason this device isn't available through the web UI yet?

No particular reason. I simply found out about it after putting together the initial list. If you create a PR against develop I'd be happy to accept it.

BTW, that is a very neat museum website you have there! Great work!

marciot, 2021-11-10:
The disk image situation is pretty much a mess when it comes to emulators, which makes inter-operability very difficult. When it comes to floppy disk images, there are "raw" headerless images and images with DiskCopy 4.2 headers. Most generally supported is 800k and 1.4MB images, but Mini vMac in particular throws a wrench into the equation by patching the ROM at runtime to load arbitrarily sized images as a floppy device. Other emulators, such as Basillisk II or PCE only support arbitrarily sized images as SCSI disks with a partition table, which is what RaSCSI does, but Mini vMac does not emulate SCSI at all. So all in all it leads to a situation when making one disk image that runs across the board is pretty much impossible.

rdmark, 2021-11-12:
Made a PR myself so that I doesn't get forgotten. https://github.com/akuker/RASCSI/pull/453

And yes, I looked into the Mini vMac disk image format a while ago, since it is such a popular format for image files floating around on the Net. Supporting it out of the box in RaSCSI would be hugely convenient. But it's not really documented (AFAICT) and non-standard, so it doesn't seem feasible at a glance, unfortunately.

rdmark, 2021-11-27:
For the record, I was able to use the Iomega Floptical driver for Mac (Mac OS 8.6) to mount an existing 1.44MB floppy image as a SCRM type RaSCSI drive and read data from it. 

It isn't able to format a new floppy disk image, but reading the specs for the original device, this is not something that I think it supported in the first place.

Also, it doesn't seem to recognize Mac OS's eject command, which again is expected since the original device had a manual eject mechanism. This means you cannot use multi-disk installers this way, as the OS won't recognize that you've injected a new disk image after forcing an eject from the RaSCSI side.

Anyhow, this is a very limited usecase for Macs using this particular Iomega driver and device profile, but it does mean that Mac users have a path to easily extract data from single floppy disk images.

marciot, 2021-11-27:
@rdmark: That is very cool. Of course, all that makes me wonder if we can somehow locate documentation on how to write a device driver without those limitations. At least after my adventures with my Trouble in Paradise port, I have a good understanding how to write SCSI code on the Mac.

rdmark, 2021-11-29:
> This reminds me, I should add the DEC SCSI floppy drives to the drive properties table... (not that they're terribly usable, but just for completeness)

@Pacjunk Have you tested that an emulated DEC SCSI floppy drive actually works on your systems, and you can read floppy disk images this way? We're preparing release notes and I'd like to know what kinds of claims we can make. :)

Pacjunk, 2021-11-29:
Yes, I tested it - only under VMS though. The DEC systems just treats a floppy as a small hard drive, except they are removable. You still have to manually mount and dismount them in the operating system, but you have to do this with real drives too. I don't really recall using floppy drives for any serious purpose on those machines. Software always came on tape, or CD-ROM. Floppies were used occasionally on the alpha systems for firmware/ECU updates, but they may have been in FAT format as Alphas can switch to an ARC console if required. I didn't test this functionality.

closed 2022-02-23 by rdmark