PiSCSI/piscsi #437: Support for RPi OS Bullseye

rdmark, 2021-11-09:
With the new Raspberry Pi OS spdlog causes the same issues as the ones observed on x86 platforms. Additional compiler flags are needed to address this.

Code snippet in https://github.com/akuker/RASCSI/pull/431

We might consider to address this in easyinstall, by checking the OS version and setting the EXTRA_FLAGS accordingly. Or even better checking the gcc version, because spdlog causes issues based on the compiler being used.
gcc 10 and newer require EXTRA_FLAGS="-DSPDLOG_FMT_EXTERNAL -DFMT_HEADER_ONLY" in order to work with spdlog.

rdmark, 2021-11-14:
PR ready at https://github.com/akuker/RASCSI/pull/469

rdmark, 2021-12-01:
@uweseimet We got reports in the #rascsi-support Discord channel that compilation fails on 64bit Raspbian Bullseye (on various Pi models) when using the DSPDLOG_FMT_EXTERNAL flag. 

When using only the DFMT_HEADER_ONLY flag it compiles well for that user. I could also confirm that rascsi compiles well on my 32bit Bullseye with only DFMT_HEADER_ONLY.

Do you think it'll be safe to exclude the DSPDLOG_FMT_EXTERNAL flag at this point, or is it still needed for some platforms (x86)?

rdmark, 2021-12-01:
Tentatively put up a PR at https://github.com/akuker/RASCSI/pull/501

uweseimet, 2021-12-01:
@rdmark I currently cannot test on a 64 bit Bullseye, but if this change is fine for you and others it's also fine for me. I am going to merge it. If required for x86 one can still use the EXTRA_FLAGS environment variable to pass additional flags if required.

closed 2021-12-06 by rdmark