PiSCSI/piscsi #438: Advanced User install options in easyinstall.sh

rdmark, 2021-11-09:
To introduce options for installing rascsi without configuring the systemd services, for advanced users who prefer to run rascsi / rascsi-web manually.

rdmark, 2021-11-15:
PR in https://github.com/akuker/RASCSI/pull/472

rdmark, 2021-11-17:
@uweseimet Please see the PR above. Do you think this approach would be helpful to you, and others who don't want to run rascsi/rascsi-web as services?

uweseimet, 2021-11-17:
@rdmark IMO there should be an option that does nothing except installing the required dependencies and compiling the source. It should not even create an images folder, like option 10) currently does.
I'd suggest to add a comment that tells the user why a password is required. Currently there is the "Password:" prompt without any explanation. If you start some script and this script asks you for a password, do you just enter it? You shouldn't :).

erichelgeson, 2021-11-17:
> does nothing except installing the required dependencies and compiling the source

I'd argue if an advanced user knows enough to know they need that it seems they could just do it themselves or write an alias/script. Whatever options we choose here it seems an advanced user may have different opinions on what they want done.

An option may be to make all the individual methods here `source`able into a shell/script so advanced users could compose their own needs.

rdmark, 2021-11-17:
@uweseimet I added an early check for sudoers privileges for the install options that touch system configuration, with a descriptive prompt message.

uweseimet, 2021-11-17:
@rdmark Great, thank you!

closed 2021-11-21 by rdmark