PiSCSI/piscsi #440: OLED screen saver

rdmark, 2021-11-09:
Screen saver functionality for when the OLED screen is idle for a certain period of time.
Ostensibly to prevent screen burn-in, although it is not known if the OLED screens that are used for RaSCSI are prone to burn-in.

akuker, 2021-11-14:
My understanding is that ALL OLEDs are susceptible to burn-in. From Adafruit's discussion of OLED displays:

> Display is showing burn in on some pixels.
> The display can have image burn in for any pixels left on over a long period of time - many days. Try to avoid having the display on constantly for that length of time.

The easiest way to address this is to ensure that the scrolling feature is always running, even if the text fits on the display. Just by scrolling the text periodically should help limit the burnin. 

TBH - I'm not too worried about burnin with a $5 board after a few years :-)

akuker, 2021-11-14:
FYI: https://youtu.be/GWOFF5tMv_A?t=439  shows an example of the burnin.

rdmark, 2021-11-14:
@akuker This video is a good illustration of the level of risk involved, which is to say relatively low, unless you keep your RaSCSI running 24/7 for years without changing the configuration (in which case you can argue that you probably want to disable the monitor as it doesn't have a purpose except ornamental?)

> The easiest way to address this is to ensure that the scrolling feature is always running, even if the text fits on the display. Just by scrolling the text periodically should help limit the burnin.

Agreed that this is the easiest way, although it makes it harder to read what's on the screen. Maybe this could be another install-time option? And throw in an option to change the update frequency as well?

> TBH - I'm not too worried about burnin with a $5 board after a few years :-)

The counter-argument here is for folks who buy a pre-assembled unit and don't have the means or know-how to replace the part...

rdmark, 2021-12-22:
For the record: After running one RaSCSI with a 32x128 OLED permanently attached for about two months straight every day, but shut down during night, I have started to see clearly visible burn-in of the 'no images attached' and IP address strings.

rdmark, 2022-02-23:
Closing this one as WONTFIX since we have the control board as an option now, which comes with screensaver support out of the box.

closed 2022-02-23 by rdmark