PiSCSI/piscsi #451: not a issue , a system to add to compatibility

macphreak4evr, 2021-11-11:
# Info please Add the Commodore Amiga 3000 to the system compatibility 

- Which version of Pi are you using: zero
- Which github revision of software: latest
- Which board version:2.3b
- Which computer is the RaSCSI connected to: Commodore Amiga 3000

# Describe the issue
no issue, works on internal 50 pin, external db25, bootable on any id that is not in use by Amiga native. shares Amiga SCSI bus. can hot add drives on the fly, removals require reboot.

rdmark, 2021-11-11:
@macphreak4evr Thanks for letting us know that it works on your machine!  May I ask you to update the compatibility wiki page with this information? https://github.com/akuker/RASCSI/wiki/Compatibility

rdmark, 2021-11-12:
I added the system to the list myself, while I was doing other edits to the page. Please don't hesitate to add to it.

closed 2021-11-12 by rdmark

macphreak4evr, 2021-11-12:
Awesome thanks!

Chris Edwards 
Network Engineer