PiSCSI/piscsi #452: Compiling STANDARD hangs when using multiple cores

rdmark, 2021-11-12:
On develop branch:

Compiling with CONNECT_TYPE=STANDARD and multiple cores (tested with 4) causes the process to hang at the fifth object, and error out after a very long time of idling.
This is on a Pi3B, but has been reported on a Pi2 by Levo_man as well.

```
 $ make -j 4 all CONNECT_TYPE=STANDARD
echo "-- Generating protobuf-based source files"
-- Generating protobuf-based source files
protoc --cpp_out=. rascsi_interface.proto
arm-linux-gnueabihf-g++ -O3 -Wall -Werror -DNDEBUG -std=c++17 -Wno-psabi -iquote . -D_FILE_OFFSET_BITS=64 -MD -MP  -DCONNECT_TYPE_STANDARD -c rascsi.cpp -o obj/standard/rascsi.o
arm-linux-gnueabihf-g++ -O3 -Wall -Werror -DNDEBUG -std=c++17 -Wno-psabi -iquote . -D_FILE_OFFSET_BITS=64 -MD -MP  -DCONNECT_TYPE_STANDARD -c rascsi_image.cpp -o obj/standard/rascsi_image.o
arm-linux-gnueabihf-g++ -O3 -Wall -Werror -DNDEBUG -std=c++17 -Wno-psabi -iquote . -D_FILE_OFFSET_BITS=64 -MD -MP  -DCONNECT_TYPE_STANDARD -c rascsi_response.cpp -o obj/standard/rascsi_response.o
mv rascsi_interface.pb.cc rascsi_interface.pb.cpp
arm-linux-gnueabihf-g++ -O3 -Wall -Werror -DNDEBUG -std=c++17 -Wno-psabi -iquote . -D_FILE_OFFSET_BITS=64 -MD -MP  -DCONNECT_TYPE_STANDARD -c rasutil.cpp -o obj/standard/rasutil.o
arm-linux-gnueabihf-g++ -O3 -Wall -Werror -DNDEBUG -std=c++17 -Wno-psabi -iquote . -D_FILE_OFFSET_BITS=64 -MD -MP  -DCONNECT_TYPE_STANDARD -c protobuf_util.cpp -o obj/standard/protobuf_util.o
```

$ g++ --version
g++ (Raspbian 8.3.0-6+rpi1) 8.3.0

rdmark, 2021-11-12:
The threshold seems to be 4 cores, at least on my 3B. If I use 2 or 3 cores, compilation works fine.

rdmark, 2021-11-12:
This may actually happen with FULLSPEC as well, it just happens a little longer into the process. Either that, or my 3B is overheating now! To test further tomorrow...

uweseimet, 2021-11-12:
@rdmark If we cannot resolve this (even though it indicates a problem with the Pi) we can simply revert the related change, because it does not affect the functionality of rascsi as such.

rdmark, 2021-11-12:
This seems to be a good starting point for troubleshooting: https://unix.stackexchange.com/questions/42837/what-could-be-causing-make-to-hang-when-compiling-on-multiple-cores

rdmark, 2021-11-12:
> 
> 
> @rdmark If we cannot resolve this (even though it indicates a problem with the Pi) we can simply revert the related change, because it does not affect the functionality of rascsi as such. How much RAM do the affected Pis have? Maybe this is a RAM shortage.

I believe both Pi 2 and 3 have 1GB of RAM.

uweseimet, 2021-11-12:
@rdmark 1 GB may not be enough for compiling with 4 cores, at least not with extensive swapping. Running 'top' while compiling provides details on RAM usage.

rdmark, 2021-11-12:
@uweseimet You may be onto something here. With four cores used, there are four cc1plus processes each using 20%-25% of system memory, topping out and staying at 20MB free system memory while make is running. With three cores used, there are three cc1plus processes each using 28%-30% of system memory, topping out at about 40MB free system memory during peaks. So cc1plus seems to get starved with four processes.

uweseimet, 2021-11-12:
@rdmark In this case I suggest to revert this optimization. This will mean longer compile times with easyinstall again, of course.
In case you don't know yet: You can configure make with the MAKEFLAGS environment variable, e.g. in your .profile:

```
export MAKEFLAGS="-j4"
```

This way you can also increase the concurrency, independent of easyinstall. But manual intervention is involved in this case, i.e. setting the environment variable.

rdmark, 2021-11-14:
Put up a PR where easyinstall.sh takes a newly introduced -j | --cores parameter, and then passes that value to the 'make' command. Default fallback is '1'. 

https://github.com/akuker/RASCSI/pull/465

closed 2021-11-14 by uweseimet