PiSCSI/piscsi #454: Not an issue: act as SCSI host

ilventu, 2021-11-12:
Is it feasible or planned to use RaSCSI as a SCSI host to connect legacy devices, such as scanners or disks?

Thanks in advance
Andrea

rdmark, 2021-11-12:
@ilventu I believe what you're asking for is Initiator mode. There is code in RaSCSI for doing this, but it's not documented and probably broken. See https://github.com/akuker/RASCSI/issues/131

ilventu, 2021-11-13:
I need to use an old scanner. I suppose I should fix "Initiator mode" in RaSCSI and probably write a DKMS to emulate the controller. I don't think that's an easy way to go. I was hoping there would be something out of the box. Thanks again.

akuker, 2021-11-14:
Initiator mode (probably) will never work perfectly without a hardware change to RaSCSI. 

As part of the arbitration, the SCSI device needs to assert the data pin associated with its SCSI ID (ex - SCSI ID 2 will assert data line 2). At the same time, it also needs to read the other ID lines to make sure there is no conflict with another device.

With the current RaSCSI hardware, ALL of the data lines are switched between input/output at the same time. So, its not possible to assert one data line, while reading the others.

That being said.... it would be possible if you're directly connecting the SCSI signals to your Raspberry Pi (and not using the transceiver chips). But you would be doing that at your own risk.

I believe the way that rasdump and sasidump work, is that they assume there are only two devices on the scsi chain. If there's only ever two devices (one initiator and one target), you don't need to worry about collisions.

uweseimet, 2022-08-26:
Looks as if this ticket can be closed, because the requested feature cannot be implemented?

akuker, 2022-08-26:
It could be implemented with a single device on the SCSI chain. But I'm OK with closing this. This would be a huge undertaking.

closed 2022-08-26 by akuker