PiSCSI/piscsi #458: The SCSI INQUIRY command is returning too much data!

Pacjunk, 2021-11-13:
# Info

- Which version of Pi are you using: ZeroW
- Which github revision of software: develop
- Which board version: FULLSPEC
- Which computer is the RaSCSI connected to: VAXstation

# Describe the issue

The spec states at least 36 bytes are returned, but typically only 36 bytes are returned.

This is the return from a real HDD:

```
$!   SCSI Inquiry Data (36 bytes) : 
$!
$!      000:  00 00 01 01 1F 00 00 18 44 45 43 20 20 20 20 20 
$!      010:  52 5A 32 34 20 20 20 20 20 28 43 29 20 44 45 43 
$!      020:  31 44 31 38 
$!
$!	Peripheral Device Type : Direct Access
$!
$!	Peripheral Device Qualifier :
$!
$!		Peripheral device type supported & connected to target
$!
$!	Vendor Identification : DEC     
$!
$!	Product Identification: RZ24     (C) DEC
$!
$!	Product Revision Level: 1D18
$!
$!	ANSI Information:
$!
$!		Complies to ANSI X3.131-1986 (SCSI-1)
$!
$!	ISO Information:
$!
$!		0 - does not claim ISO SCSI (ISO IS 9316) compliance
$!
$!	ECMA Information:
$!
$!		0 - does not claim ECMA SCSI (ECMA-111) compliance
$!
$!	Removable Media: No
$!
$!	AEN Capability: No
$!
$!	Command Queuing: No
$!
$!	16-Bit Wide Bus Data Transfers: No
$!
$!	32-Bit Wide Bus Data Transfers: No
$!
$!	Linked Commands: Yes
$!
$!	Soft Reset Alternative: No
$!
$!	Relative Addressing: No
$!
$!	Synchronous Data Transfers: Yes
```
This ends with the Product Revision Level.

This is the data returned by RASCSI:

```
$!   SCSI Inquiry Data (130 bytes) : 
$!
$!      000:  00 00 02 02 7D 00 00 00 44 45 43 20 20 20 20 20 
$!      010:  52 5A 32 34 20 20 20 20 20 28 43 29 20 44 45 43 
$!      020:  31 44 31 38 00 02 00 00 00 00 00 00 00 00 00 00 
$!      030:  00 00 00 00 80 00 00 00 00 FE 02 00 04 00 01 00 
$!      040:  00 FA 04 00 04 00 00 00 00 00 00 00 07 00 00 00 
$!      050:  53 45 43 55 52 49 54 59 2E 53 59 53 3B 31 20 20 
$!      060:  20 20 20 20 03 00 20 F7 49 6A FC DC 96 00 00 40 
$!      070:  8D 8D BD 6D 97 00 00 00 00 00 00 00 00 00 00 00 
$!      080:  00 00 
$!
$!	Peripheral Device Type : Direct Access
$!
$!	Peripheral Device Qualifier :
$!
$!		Peripheral device type supported & connected to target
$!
$!	Vendor Identification : DEC     
$!
$!	Product Identification: RZ24     (C) DEC
$!
$!	Product Revision Level: 1D18
$!
$!	ANSI Information:
$!
$!		Complies to SCSI-2
$!
$!	ISO Information:
$!
$!		0 - does not claim ISO SCSI (ISO IS 9316) compliance
$!
$!	ECMA Information:
$!
$!		0 - does not claim ECMA SCSI (ECMA-111) compliance
$!
$!	Removable Media: No
$!
$!	AEN Capability: No
$!
$!	Command Queuing: No
$!
$!	16-Bit Wide Bus Data Transfers: No
$!
$!	32-Bit Wide Bus Data Transfers: No
$!
$!	Linked Commands: No
$!
$!	Soft Reset Alternative: No
$!
$!	Relative Addressing: No
$!
$!	Synchronous Data Transfers: No

```
130 bytes and stuff that shouldn't be there. Notice that there is text around the 0x50 mark. I decoded this and it is a filename from the FS in the disk image, so obviously a buffer overrun. I suspect that the buffer in is actually 36 bytes, but the length has been set incorrectly.

Saying all that, it actually works OK. I assume the driver is just ignoring the extra data. The buffer overrun is not good though!

uweseimet, 2021-11-13:
@Pacjunk Can you prepare a branch with a fix and a PR, please?

Pacjunk, 2021-11-14:
Done & submitted

closed 2021-11-14 by uweseimet