PiSCSI/piscsi #459: Modesense(10) commands are returning incorrect data length

Pacjunk, 2021-11-13:
# Info

- Which version of Pi are you using: ZeroW
- Which github revision of software: develop
- Which board version: FULLSPEC
- Which computer is the RaSCSI connected to: VAXstation

# Describe the issue

Need to check other 10 byte commands as well. Following is a dump:

```
$!   SCSI 10-Byte Mode Sense command returned 28 bytes of data:
$!
$!      000:  00 1B 00 00 00 00 00 08 00 06 40 C0 00 00 02 00 
$!      010:  01 0A 00 00 00 00 00 00 00 00 00 00 
$!
$!      Mode Sense Data Length ...... 27
$!      Medium Type ................. 00h
$!      Device-Specific Parameter ... 00h
$!      Block Descriptor Length ..... 8
$!      Block Descriptor Data ....... 00 06 40 C0 00 00 02 00 
$!
$!        Page Code ................. 01h
$!        Page Name ................. Read-Write Error Recovery
$!        Saveable .................. No
$!        Size ...................... 10
$!        Hex Data .................. 00 00 00 00 00 00 00 00 
$!                                    00 00 

```
The command returns 28 bytes of data, which is correct. The first 2 bytes give the data length, and there are 26 bytes remaining.
However the data incorrectly indicates the length as 27 bytes (0x1B). This should be 26 (0x1A). I suspect the code has been copied from the Modesense(6) command, where the length is only 1 byte and thus only has a 1 byte correction.

uweseimet, 2021-11-13:
@Pacjunk Can you prepare a branch with a fix and a PR, please?

Pacjunk, 2021-11-14:
Done & submitted

closed 2021-11-14 by uweseimet