PiSCSI/piscsi #460: [Question] Can RaSCSI capture SCSI traffic between two other devices on SCSI chain? (so I can reverse engineer a Zip/Jaz disk diagnostics tool)

marciot, 2021-11-13:
### Short Question:

If I were to put RaSCSI on a SCSI chain with a Windows PC and an Iomega Zip drive, could I used it to capture SCSI traffic in sufficient detail to possibly write a Mac program to do the same thing?

### TL;DR:

So I just had a week long ordeal trying to resurrect an old Windows 95/98 tool called TIP (Trouble in Paradise):

[image]
[Web Site](https://www.grc.com/tip/clickdeath.htm)

This came about as a lucky find at a surplus store of an old PC with a SCSI card for $10. Since I use Zip drives for my Mac Plus and had encountered the "click of death" before, I jumped at the opportunity to be able to scan my media and drives for defects. I *thought* it would be easy, just slap Windows on the PC and I'm off to the races, right?

Boy, was I wrong. I'll spare y'all the details, but getting all this old software and hardware to run was hell. But in the end, it was worth it, as TIP is an amazing tool and I learned *sooooo much* about the state of my Zip media and drives :grin:

I feel that anyone using a Zip drive on their Mac could benefit tremendously from this tool, yet the effort and rarity of the hardware require to get it to run in 2021 makes it unlikely anyone will. Personally, I wish I didn't have to deal with a jerry-rigged, barely functional Windows 98 PC to run it. It would take far less time, and help far more people, to simply re-write this tool for a classic Mac.

Which brings me to my question, can RaSCSI capture the traffic between TIP and the Iomega drive? TIP seems to be fairly simple. As far as I can tell, it does the following:

1. It reads and saves a sector from the drive
2. It overwrites the sector with a random pattern of bytes
3. It reads the sector and verifies that random pattern
4. It then re-writes the original sector to the drive
5. As it performs steps 1 through 4, it keeps a count of the following types of errors:
  - Soft errors: Normal, expected, recoverable errors
  - Firm errors: Bad blocks which are read after several retries but remapped by the drive into spare sectors
  - Hard errors: Unrecoverable sectors which are lost and cannot be recovered

The tool also reports the number of spare sectors on each side of the Zip disc. The general idea is that if a drive has a lot of firm or hard errors, it is on the way towards the "click of death".  When used on a good drive, this tool also recovers cartridges damaged by misaligned Zip drives.

While it's unclear how many people still use Zip drives for their Macs, and whether it is enough that it would justifies rewriting this tool for the Mac, I think it is something worth exploring. I still have some old Macintosh code I wrote in the 2000s for accessing a SCSI tape drive, so if there is enough demand, I could probably modify that to do something similar to what TIP is doing, provided I can get a capture of the exact SCSI commands at play.

uweseimet, 2021-11-13:
@marciot Does this tool do anything that is specific to the ZIP? From your description it sounds as if it just reads/writes/compares data like you can do it with any kind of drive.

marciot, 2021-11-13:
@uweseimet: It appears to do some things specific to the Zip drive. I can't find exactly where I read it, but the author of the tool says that it does not work on certain internal Zip drives because those drives do not expose the necessary SCSI API. My guess is that the reading and writing part is fairly generic, but if the tool only did that you would have no indication the drive or cartridge was actually damaged. The drive would simply remap sectors until the cartridge failed completely. I think the key thing TIP does is it uses special SCSI commands to query from the drive the exact nature of the error and the counts of remapped sectors.

marciot, 2021-11-13:
@uweseimet: If you want to dig further, you can check out the [GRC website](https://www.grc.com/tip/clickdeath.htm). He has *so* much information about the "click of death". Fascinating stuff. So far I have not found documentation on the specific SCSI commands, though.

uweseimet, 2021-11-13:
@marciot I cannot answer your intial question, but I assume @akuker can because he surely is more familiar with what you can do with the RaSCSI hardware.
An alternative approach to listening on the SCSI bus might be to have the tool run against a virtual SCSI drive, in order to see (based on the logs) which commands are sent. You might have to implement missing commands, of course.

Regarding ZIP/JAZ specific commands the only ones I know are those for read/write protecting media with a password, and for configuring the time until auto-sleep and auto-park. Besides those my JAZ (which is not functioning anymore) was not different from any other removable media drive. Well, one difference is that ZIP and JAZ tend to break easier than, for instance, SyQuest drives ;-). The latter are still working for me, even though they are older than the JAZ.

marciot, 2021-11-13:
@uweseimet: It would seem like having to reconstruct a virtual device would be *far harder* than simply listening to the traffic, so I am hoping for the latter 😀

On a whim, I reached out to the original author of the software. He still seems to be active on social media as of about a year ago. Maybe I'll luck out and he will share some info.

uweseimet, 2021-11-13:
@marciot Sounds good. Maybe he would even be willing to contribute to the RaSCSI project.

marciot, 2021-11-13:
> I would not be surprised, though, if RaSCSi is too slow to capture all data on the bus. The transfer rate for read operations is about 70% of the rate I get with physical drives (when connect to my Atari, I guess the Mac's SCSI bus is faster). I guess that if RaSCSI was just passively listening on the bus it would miss data.

Well, this just got the wheels turning in my brain. Your comment made me realize that all that needs to happen to capture SCSI bus traffic is to sample bits fast enough. Possibly an Raspberry Pi could do this, with the right code, but I imagine Linux has some overhead. A backhanded approach might be to remove the RaspPi and wire up sufficiently fast dedicated microcontroller to the RaSCSI board and write some code to capture bits directly to an SD card. Then the challenge would be to convert the bit stream into something I could read... oh man, so many projects, so little time!

uweseimet, 2021-11-13:
@marciot Aren't the rasdump or the scsimon tool supposed to do sampling? I've never used those tools, but I think one of them can do sampling.
Instead of sampling and probably just dumping the bytes as fast as possible, a tool that samples and then displays the result as decoded SCSI commands including their parameters and the results would be great, indeed.

marciot, 2021-11-13:
@uweseimet: I don't know. That is why I posted the question, lol. As of last week, I knew nothing about RaSCSI, so it's been a steep learning curve. I'll get there :grin:

Anyhow, I think on first blush you would want to capture then raw bits. Then you can probably run an offline tool to do the decoding. Doing it all at once might be a stretch. I suspect that the capture tool could simply read the SCSI pins continually and write out the data to a file only when it detected a change. This would make the code very simple and it probably could run very fast. Then, you would have to run some other tool that would try to decode the commands. It certainly seems doable, and I wouldn't be surprised if some of that work had already been done.

uweseimet, 2021-11-13:
@marciot Yes, you are right, this should be a two-step process, which a post-processing step doing the work of decoding to a human-readable format.

marciot, 2021-11-13:
@uweseimet: Hey, thanks for the hint about "scsimon". That was the magic keywords to search for. Here it is!

https://github.com/akuker/RASCSI/wiki/SCSI-Bus-Capture

marciot, 2021-11-13:
This ticket can be closed, but maybe I'll wait on @akuker to see it any add any information that might be missing in this thread.

I'll experiment with capturing a VCD file while TIP is running and see whether it can keep up. I bet I will end up with a ridiculously large file really quick, but I probably only need a second or two to capture all the good stuff because TIP basically does the same thing over and over.

marciot, 2021-11-13:
@uweseimet: The author, Steve Gibson, got back to me *super* quick and provided the entire assembly source code for me to look at. I'll be digging in to see what I can find! He mentioned these functions, in particular:

- EnumerateIomegaDevices
- GetModePage
- SetModePage
- GetNonSenseData
- UnlockAllMedia
- LockCurrentDrive
- SpinUpIomegaCartridge
- GetSpareSectorCounts
- ReadSectors
- WriteSectors
- ReadSectorLong
- WriteSectorLong
- FillBufferThroughASPI
- EjectIomegaCartridge
- PerformRegionTransfer

akuker, 2021-11-14:
@marciot - That is exciting!! SCSIMON should be able to do what you're looking for.

While its running, it just sits in a very tight loop reading the GPIO pins and saving any changes (along with their timestamps) to RAM. Once it fills up its RAM buffer, it will dump that data to the drive. (This way, you're not fighting drive latency).

If you run scsimon on a barebones system, and give it high prioriy, it seems to work reasonably well to capture data. Its not perfrect by any means. But you can usually figure out what's going on. Especially if you're using a Pi 4.

The main weakness of the tool is LONG data captures are a pain to work with. There is definitely some better post-processing of the SCSI data that could make it much easier to understand/parse. The GTKWave tool works, but with the current workflow, its not ideal.

As a Zip drive lover, this project excites me!! :-) Please keep us up to date with how its going! (Also- thank you for the sponsorship! I appreciate it!)

marciot, 2021-11-15:
Well, making a bit of progress. Today I was able to write a small program on my PowerBook 3400c that allows me to select a Zip drive on the SCSI bus and send a SCSI cartridge eject command to it. Next I'll have to tackle reading the spare sector count... and then everything else 😝

akuker, 2021-11-15:
Sounds like you're pretty much done! ;-)

hsiboy, 2021-11-19:
Hey @marciot 👋 

Fascinating. I've just spend the last 20 days down a deep rabbit hole. I have been looking at something very similar!
My difference is that i'm looking at ancient synths not Macs, and in this case, i'm looking at IDE/ATAPI.

So far as the Iomega drives go, i've learnt that the IDE, ATAPI drives and the USB drives, all use the SCSI commands. 
The only difference between the IDE and the ATAPI drives, is a firmware revision. With the IDE drive, it will often appear as a floppy drive and not a HDD with removable storage. The ATAPI version always appears as a HDD, but, i can be configured to appear as a floppy drive 🙄 
The ATAPI ZIP drive has a jumper (called the ARMD-jumper, or, more commonly, the A: jumper) that will hide the first 32 sectors of the disk. In other words: with the jumper we have a big floppy and no partition table; without jumper we have a small disk with DOS-type partition table. Yeah, helpful Iomega.

I've had to learn about zip disk geometry and how the zip drive appears, depending on how it is formatted (PC/MAC).
ZIP disks prepared by IOMEGA utilities have a partition table (usually having a single entry, the fourth, spanning the whole disk, so that access to such disks with partition table goes via mounting /dev/sda4. However, for Mac formatted discs, they have 6 partitions, and again the fourth can be mounted, with filesystem type "hfs".

Non-present disks have a capacity of 0 bytes. 100MB disks have a capacity of 100663296 bytes, i.e., 196608 sectors
> Note Iomega measures its disk capacities in megabytes where 1 megabyte = one million bytes. This is less than the conventional definition where 1 megabyte = 1,048,576 bytes.

Using Linux and the `sg_raw` command, here's what i've learned:

```bash

# Issue non-sense command for page 0
sg_raw --request=254 --outfile=page0.bin </dev/zipdisk>  06 00 00 00 FE 00

# Issue non-sense command for page 1
sg_raw --request=254 --outfile=page1.bin </dev/zipdisk> 06 00 01 00 FE 00

# Issue non-sense command for page 2
sg_raw --request=254 --outfile=page2.bin </dev/zipdisk> 06 00 02 00 FE 00

# Extract the disk ID.
dd if=page2.bin bs=1 skip=22 count=18

# Extract the (ASCII) disk serial number data to a file
dd if=page2.bin of=serial.txt bs=1 skip=22 count=40
```

The problem i have, is that i have not been able to replace the Zip drive in the synth with an IDE to CF or SD adaptor, because the Synth doesn't detect a zip drive. CF cards are electrically compatible with IDE and they speak ATAPI. What i need to do, is find out what handshake happens when a user hits the eject button. So, like you, i'm trying to sniff the IDE bus, something that the Raspi is not fast enough to do.

Would you share Mr Gibsons source?

uweseimet, 2021-11-19:
@hsiboy Just some related notes:

- It is normal that ATAPI drives (not just the IOMEGA ones, but also all others like the LS-120) use SCSI commands, just like modern MMC (multimedia drives) like DVD-RAM, BD, ... That's how everything is specified. The SCSI command set is still state of the art (not the hardware, but the commands, where the specification is regularly updated). The NVMe or a similar software protocol is a replacement candidate for the future.
- Jumpers that deal with configuring the ZIP drive geometry are only needed for very old (mainly DOS) drivers that do not use logical block addressing. With logical block addressing the geometry is irrelevant, because the drivers just tell the device with sector number to access.
- Whether a drive/medium has a partition table or not (i. e. looks like a floppy) depends on the software the filesystems on the drive was created with. There is software that allows you to choose which layout you would like to have.

marciot, 2021-11-19:
@hsiboy: Interesting! I can't say I followed all that, but it sounds like what you are working on is quite interesting! I feel like we need to start an Iomega Zip hacker's group and maybe create a Wiki with all our finding!

> Would you share Mr Gibsons source?

Let me just make sure it is okay with him, I don't want to ruffle any feathers by sharing what isn't mine. I'm sure it won't be a problem! I'll let you know as soon as I get a reply!

uweseimet, 2021-11-19:
@hsiboy I don't think your statement about CF cards speaking ATAPI is correct. They speak IDE, which is a quite primivate language, whereas ATAPI typically uses package writing, which is essentially like SCSI as far as the command set is concerned.
I am familiar with writing mass storage devices drivers for IDE, ATAPI and SCSI, and from a software perspective there is no difference between IDE hard disk drives and CF or SD cards from a driver perspective. The driver does not even know whether an IDE hard disk drive or a memory card is connected. But the language is IDE, not ATAPI.

hsiboy, 2021-11-19:
hey @uweseimet sure, IDE isnt a BUS but AT is, and IDE and AT and i'm so old i use interchangeably.

I pointed out the SCSI command comment because there is a lot of BS around Zip drives, especially with regard to the IDE vs ATAPI version of the Zip 100.

Yes 

LBA.

When the Zip drive is configured in the Drive A: mode (jumper in place):

- The first 32 LBS'a of the disk are not accessible (i.e., the drive adds 32 to all LBA's specified in all commands).
- Disk capacity is reduced by 32 LBA's.
- The FORMAT UNIT PACKET Command (04h) writes a DOS partition on LBA-32.
- ATA IDENTIFY PACKET DEVICE Command (A1h) reports "Floppy" at word offset 34.

When the host writes data to LBA 0 that the drive's firmware evaluates as a DOS boot record, the drive then checks the data at offset 28; if offset 28 contains 00h, the drive changes the value to 80h. 
The drive also checks the data at offset 36; if offset 36 contains 00h, the drive changes it to 20h. The drive makes these changes prior to writing the data to LBA 0 on the disk.

When the host reads data from LBA 0 and the drive's firmware evaluates the retrieved data as a DOS boot record, the drive then checks the data at offset 28; if offset 28 contains 80h, the drive changes it to 00h. The drive also checks the data at offset 36; if offset 36 contains 20h, the drive changes it to 00h. The drive makes these changes before sending the data on to the host.

It also seems possible to set the A: mode via commands (some BIOS can do this), but i cant find these... yet

uweseimet, 2021-11-19:
@hsiboy I am glad that I never used DOS :-). All those work-arounds required to get DOS (and sometimes also WINDOWS) somehow working with some peripherals are really ugly.

hsiboy, 2021-11-19:
You know, there was a time, when i might have argued for DOS over Unix. DOS is real mode, you can pas it machine code and it passes it straight to the hardware. Just 3 files (io.sys, msdos.sys and command.com) to boot to a rudimentary OS, in about 5 seconds. But as soon as you wanted to access a drive bigger than 32MB, yeah, we had to use LBA, luckily my first HDD (it was RLL not IDE) was just 20MB, i don't think i ever filled it!

marciot, 2021-11-20:
> Would you share Mr Gibsons source?

@hsiboy: I've gotten permission from Steve to reshare thecode, so I have uploaded it to GitHub.

https://github.com/marciot/grc-tip

 All the SCSI stuff is in the file "ASPI.ASM", while "TIP.ASM" is mostly the user interface. There are also a lot of constants in "ASPI.INC"

marciot, 2021-11-25:
Hey, check this out!

Happy Thanksgiving everyone!

[image]

marciot, 2021-11-27:
I am closing this issue since the tool has already been written and is now under active beta testing. To make a long story short, sniffing the SCSI bus was not necessary once Steve Gibson provided the original source code to me. Still cool to know RaSCSI supports bus sniffing.

closed 2021-11-27 by marciot

akuker, 2021-11-28:
Thank you for the update, @marciot !! Congratulations on the awesome tool!