PiSCSI/piscsi #473: Kobold 'Cannot write FAT tables' error (Atari)

mzryz, 2021-11-17:
# Info

- Which version of Pi are you using: Zero2 W
- Which github revision of software: develop-111920e
- Which board version: Full Spec
- Which computer is the RaSCSI connected to: Atari TT

# Describe the issue

Only when using Kobold, file transfers fail with an error when 'updating fat tables' when trying to write to RaSCSI drive (IDE to SCSI). I tested TOS transfer by copying 2800 files from IDE to SCSI (Thunder IDE interface) There is no problem when using normal TOS. Perhaps there is something Kobold does that is not supported yet?

I wonder if @uweseimet might know.

uweseimet, 2021-11-17:
@mzryz I am using Kobold (version 3.5.1) all the time and so far have not observe any issues. There is nothing Kobold could do that would not be supported. All in all Kobold does not do something unusual.

mzryz, 2021-11-17:
@mzryz just to confirm: I have the TT motherboard resistors installed & termination enabled for RaSCSI. Is this correct? (only SCSI device on the machine)

uweseimet, 2021-11-17:
@mzryz Both ends of the physical SCSI chain have to be terminated, so your setup sounds correct to me. Bad cables can also cause issues, of course.

mzryz, 2021-11-17:
@uweseimet Ok yes I think this must be a cable issue of some kind. I just tested on my Falcon and Kobold worked fine on RaSCSI. I'll close this issue. Thank you for your advice.

closed 2021-11-17 by mzryz