PiSCSI/piscsi #479: Add support for READ LONG(10) and WRITE LONG(10) ?

Pacjunk, 2021-11-23:
# Info

- Which version of Pi are you using: ZeroW
- Which github revision of software: develop
- Which board version: FULLSPEC
- Which computer is the RaSCSI connected to: OpenVMS

# Describe the issue

The OpenVMS SCSI driver "tests" the capability of the drive for the volume shadowing feature by sending hundreds of READ LONG(10) commands when mounting a volume. It seems to ignore the unsupported error message that this creates and keeps trying.

Is it possible to implement these commands - 1. To stop the above issue and 2. It might be fun to support volume shadowing.

I had a look at the code as I thought it might be easy since the CDB layout seems to be the same as the READ(10) and WRITE(10) commands, the only difference being that the length is in bytes, rather than blocks. I'm not a very good C++ programmer, so I find it confusing the way the code is spread amongst several files, there are multiple definitions of functions (which one to use?) and even functions calling themselves. I could start hacking it, but I fear the code will soon become a mess.

Can someone who is confident with the code have a look at the viability of this?

Thanks,

uweseimet, 2021-11-23:
@Pacjunk Can you please point out which function is calling itself? I don't think this happens anywhere. Of course, code for similar functionailty like READ(6/10/16) is now reused.

In order to add READ/WRITE LONG you might try to copy Read10 and Write10 (see disk.cpp) first, then modify the copied code as needed (maybe no modification is needed at all in your case), and then add the respective AddCommand() calls to the Disk class constructor. Note that the difference is NOT that the length is in bytes. This is from SBC-3:

```
The READ LONG (10) command (see table 74) requests that the device server transfer data from a single
logical block or physical block to the Data-In Buffer. The data transferred during the READ LONG (10)
command is vendor specific, but shall include the following items recorded on the medium:
a)if a logical block is being transferred, then:
A) user data or transformed user data for the logical block;
B) protection information or transformed protection information, if any, for the logical block; and
C) any additional information (e.g., ECC bytes) for all the physical blocks in the logical block;
or
b)if a physical block is being transferred, then:
A) user data or transformed user data for all the logical blocks in the physical block;
B) protection information or transformed protection information, if any, for all the logical blocks in the
physical block; and
C) any additional information (e.g., ECC bytes).
If the additional information contain an ECC, then any other additional bytes that are correctable by ECC
should be included (e.g., a data synchronization mark within the area covered by ECC). It is not required for
the ECC bytes to be at the end of the logical block data. The ECC bytes should be in the same order as they
are on the medium.
If a cache contains a more recent version of the specified logical block or physical block, then the device
server shall write the logical block or physical block to the medium before reading it. The values in the
Read-Write Error Recovery mode page (see 6.5.8) do not apply to this command. The device server may
perform retries while processing this command.
This command uses the SERVICE ACTION IN (16) CDB format (see A.2).
NOTE 14 - Migration from the READ LONG (10) command to the READ LONG (16) command is
recommended for all implementations.

The OPERATION CODE field is defined in SPC-4 and shall be set to the value shown in table 74 for the READ
LONG (10) command.
If there is more than one logical block per physical block (i.e., the LOGICAL BLOCKS PER PHYSICAL BLOCK
EXPONENT field in the READ CAPACITY (16) parameter data (see 5.16.2) is set to a non-zero value), then:
a) the device server shall support the physical block (PBLOCK) bit;
b) a PBLOCK bit set to one specifies that the device server shall return the entire physical block containing
the specified logical block; and
c) a PBLOCK bit set to zero specifies that the device server shall return bytes representing only the
specified logical block.
If there are one or more physical blocks per logical block (i.e., the LOGICAL BLOCKS PER PHYSICAL BLOCK
EXPONENT field in the READ CAPACITY (16) parameter data (see 5.16.2) is set to zero), and the PBLOCK bit is
set to one, then the device server shall terminate the command with CHECK CONDITION status with the
sense key set to ILLEGAL REQUEST and the additional sense code set to INVALID FIELD IN CDB.
A correct (CORRCT) bit set to zero specifies that a logical block be read without any correction made by the
device server. A CORRCT bit set to zero should result in the device server completing the command with
GOOD status unless data is not transferred for some reason other than that the data is non-correctable. In this
case the device server shall complete or terminate the command with the appropriate status and sense data.
A CORRCT bit set to one specifies that the data be corrected by ECC before being transferred to the Data-In
Buffer.
The LOGICAL BLOCK ADDRESS field specifies an LBA (see 4.5). If the specified LBA exceeds the capacity of the
medium (see 4.5), then the device server shall terminate the command with CHECK CONDITION status with
the sense key set to ILLEGAL REQUEST and the additional sense code set to LOGICAL BLOCK ADDRESS
OUT OF RANGE.
The BYTE TRANSFER LENGTH field specifies the number of bytes of data that shall be read from the specified
logical block or physical block and transferred to the Data-In Buffer. If the BYTE TRANSFER LENGTH field is not
set to zero and does not match the available data length, then the device server shall terminate the command
with CHECK CONDITION status with the sense key set to ILLEGAL REQUEST and the additional sense code
set to INVALID FIELD IN CDB. In the sense data (see 4.18 and SPC-4), the VALID bit shall be set to one and
the ILI bit shall each be set to one, and the INFORMATION field shall be set to the difference (i.e., residue) of the
requested byte transfer length minus the actual available data length in bytes. Negative values shall be
indicated by two's complement notation.
A BYTE TRANSFER LENGTH field set to zero specifies that no bytes shall be read. This condition shall not be
considered an error.
The CONTROL byte is defined in SAM-5.

```

By the way, READ LONG is (but not WRITE LONG) is legacy and not supported anymore since SBC-4 from 2020.

Pacjunk, 2021-11-23:
I know it's legacy, but we're generally dealing with legacy machines here!

From your text above: 

> The BYTE TRANSFER LENGTH field specifies the number of bytes of data that shall be read from the specified
> logical block or physical block and transferred to the Data-In Buffer. 

I can have a go at it, but it's going to be a mess...

```
void Disk::Read(SASIDEV *controller, uint64_t record)
{
	ctrl->length = Read(ctrl->cmd, ctrl->buffer, record);
```

I find this stuff confusing. I can see that there are two functions called "Read", this must be referring to the second one. As I say, my C++ skills are poor, and this sort of thing doesn't help.

Thanks,

uweseimet, 2021-11-23:
@Pacjunk When I mentioned that these commands are legacy I was not my intention to say they should not be implemented. This was just meant to be a general note, but I should have been clearer on that.

Yes, there are two Reads(). From their signature one can see which is the right one.

Unfortunately, the RaSCSI project appears to lack C++ contributors, even though almost everything is coded in C++ ...

uweseimet, 2021-11-23:
@Pacjunk I just had a quick look at the existing code. Currently the code can only deal with full blocks and not just a part of a block sequence (limited by the LONG transfer length). Changing this in order to support READ/WRITE LONG looks like a major change to me. Testing this would be difficult because you may be one of the few or even the only one with the required hardware. Unit tests would help, but there are none, and the automated testing issue (https://github.com/akuker/RASCSI/issues/17) has not been updated for more than a year.

Nevertheless, adding a log (see my last comment) to this ticket would definitely be useful, in order to see the CDB parameters used.

Pacjunk, 2021-11-24:
Here is the log extract...

It is not reading anything meaningful, just testing the capability. It requests LBA0, length 0, then length 1, then length $0200, then seems to increment by 1 all the way up to $0400. So lots of commands. Don't know why it ignores the returned error.

```
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.663] [trace] virtual void SASIDEV::Command() Command Phase
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.663] [trace] virtual void SASIDEV::Command() Command Byte 0: $3E
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.663] [trace] virtual void SASIDEV::Command() Command Byte 1: $00
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.663] [trace] virtual void SASIDEV::Command() Command Byte 2: $00
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.663] [trace] virtual void SASIDEV::Command() Command Byte 3: $00
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.663] [trace] virtual void SASIDEV::Command() Command Byte 4: $00
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.664] [trace] virtual void SASIDEV::Command() Command Byte 5: $00
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.664] [trace] virtual void SASIDEV::Command() Command Byte 6: $00
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.664] [trace] virtual void SASIDEV::Command() Command Byte 7: $00
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.664] [trace] virtual void SASIDEV::Command() Command Byte 8: $00
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.664] [trace] virtual void SASIDEV::Command() Command Byte 9: $00
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.664] [trace] virtual void SCSIDEV::Execute() Execution phase command $3E
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.664] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $3E
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.664] [trace] ID 0 LUN 0 received unsupported command: $3E
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.664] [debug] Error status: Sense Key $05, ASC $20, ASCQ $00
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.664] [trace] virtual void SCSIDEV::Error(ERROR_CODES::sense_key, ERROR_CODES::asc) Error (to status phase)
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.664] [trace] void SASIDEV::Status() Status phase
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.664] [trace] void SASIDEV::Status() Status Phase $02
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.664] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.665] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.665] [trace] void SASIDEV::MsgIn() Starting Message in phase
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.665] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.665] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.665] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.665] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.665] [trace] virtual void SCSIDEV::BusFree() Bus free phase
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.666] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=0 (with device)
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.666] [trace] void SCSIDEV::MsgOut() ID 0
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.666] [trace] Message Out Phase
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.666] [trace] void SCSIDEV::MsgOut() ID 0
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.666] [trace] virtual void SCSIDEV::Receive()
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.666] [trace] virtual void SCSIDEV::Receive() length was != 0
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.666] [trace] void SCSIDEV::MsgOut() ID 0
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.666] [trace] virtual void SCSIDEV::Receive()
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.666] [trace] virtual void SCSIDEV::Receive() ctrl.phase: 10 (msgout)
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.666] [trace] Message code IDENTIFY $C0, LUN 0 selected
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.666] [trace] virtual void SASIDEV::Command() Command Phase
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.667] [trace] virtual void SASIDEV::Command() Command Byte 0: $3E
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.667] [trace] virtual void SASIDEV::Command() Command Byte 1: $00
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.667] [trace] virtual void SASIDEV::Command() Command Byte 2: $00
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:58.667] [trace] virtual void SASIDEV::Command() Command Byte 3: $00
[... cut ...]
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:59.309] [trace] void SCSIDEV::MsgOut() ID 0
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:59.309] [trace] virtual void SCSIDEV::Receive()
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:59.309] [trace] virtual void SCSIDEV::Receive() ctrl.phase: 10 (msgout)
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:59.309] [trace] Message code IDENTIFY $C0, LUN 0 selected
Nov 24 15:25:59 raspberrypi RASCSI[431]: [2021-11-24 15:25:59.310] [trace] virtual void SASIDEV::Command() Command Phase

```

uweseimet, 2021-11-24:
@Pacjunk Did you notice that $0204 is omitted during these incremental reads? Most likely this means that a length of 516 bytes is something special for OpenVMS. Any idea? I dimly remember that there were operating systems using/supporting unusual sector sizes. Maybe the additional 4 bytes are supposed to contain a checksum for the respective sector.
https://forums.gentoo.org/viewtopic-p-3502086.html?sid=5a023a282ae59b47c4591923e546e80e mentions drives formatted with a block sizes of 516 bytes, by the way. I have a FUJITSU drive which can be formatted with different sector sizes, but I never had any use for sizes other than 512, 1024 and 2048.

uweseimet, 2021-11-24:
@Pacjunk I re-read the spec and this may be important:

```
A BYTE TRANSFER LENGTH field set to zero specifies that no bytes shall be read. This condition shall not be
considered an error.
```
This is not what RaSCSI currently does. I wonder if I should give it a try and implement a partial READ LONG, which in this special case does not return an error, but does in all other cases.

uweseimet, 2021-11-24:
@Pacjunk Not it's your turn: Feature branch is 'feature_read_long'. The implementations for READ/WRITE LONG 10 accept a transfer length of 0 and in this case do not return an error. Any other length results in INVALID_FIELD_IN_CDB, which is IMO compliant with the standard.
Regardless of the testing outcome, please attach a log on trace level.

Pacjunk, 2021-11-24:
Yes, I was thinking of this as a first step. Just return a success for the 0 length request and see what happens. I just had a look at the dump I did from bluescsi, and the $0204 length is in there so I don't think there is anything tricky going on. Maybe too much logging and it skipped one - it does take a long time to do things when trace logging is turned on.

The READ (10) command also specifies that 0 blocks is not an error!

I will checkout your change, but don't think I'll be able to test it tonight (10pm now). Will get to it in the morning.

uweseimet, 2021-11-24:
@Pacjunk I don't think that READ(10) currently reports an error if 0 blocks is specified, i.e. the implementation of READ(10) should be fine.

uweseimet, 2021-11-24:
Log excerpts from testing with an Atari, trying to transfer 0, 1 and 512 bytes:

```
[2021-11-24 16:29:06.110] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
[2021-11-24 16:29:06.110] [trace] void SCSIDEV::MsgOut() ID 6
[2021-11-24 16:29:06.110] [trace] Message Out Phase
[2021-11-24 16:29:06.111] [trace] void SCSIDEV::MsgOut() ID 6
[2021-11-24 16:29:06.111] [trace] virtual void SCSIDEV::Receive()
[2021-11-24 16:29:06.111] [trace] virtual void SCSIDEV::Receive() length was != 0
[2021-11-24 16:29:06.111] [trace] void SCSIDEV::MsgOut() ID 6
[2021-11-24 16:29:06.111] [trace] virtual void SCSIDEV::Receive()
[2021-11-24 16:29:06.111] [trace] virtual void SCSIDEV::Receive() ctrl.phase: 10 (msgout)
[2021-11-24 16:29:06.111] [trace] Message code IDENTIFY $80, LUN 0 selected
[2021-11-24 16:29:06.111] [trace] virtual void SASIDEV::Command() Command Phase
[2021-11-24 16:29:06.111] [trace] virtual void SASIDEV::Command() CDB[0]=$3E
[2021-11-24 16:29:06.111] [trace] virtual void SASIDEV::Command() CDB[1]=$00
[2021-11-24 16:29:06.111] [trace] virtual void SASIDEV::Command() CDB[2]=$00
[2021-11-24 16:29:06.111] [trace] virtual void SASIDEV::Command() CDB[3]=$00
[2021-11-24 16:29:06.111] [trace] virtual void SASIDEV::Command() CDB[4]=$00
[2021-11-24 16:29:06.111] [trace] virtual void SASIDEV::Command() CDB[5]=$00
[2021-11-24 16:29:06.111] [trace] virtual void SASIDEV::Command() CDB[6]=$00
[2021-11-24 16:29:06.112] [trace] virtual void SASIDEV::Command() CDB[7]=$00
[2021-11-24 16:29:06.112] [trace] virtual void SASIDEV::Command() CDB[8]=$00
[2021-11-24 16:29:06.112] [trace] virtual void SASIDEV::Command() CDB[9]=$00
[2021-11-24 16:29:06.112] [trace] virtual void SCSIDEV::Execute() Execution phase command $3E
[2021-11-24 16:29:06.112] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $3E
[2021-11-24 16:29:06.112] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing ReadLong10 ($3E)
[2021-11-24 16:29:06.112] [trace] void SASIDEV::Status() Status phase
[2021-11-24 16:29:06.112] [trace] void SASIDEV::Status() Status Phase $00
[2021-11-24 16:29:06.112] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[2021-11-24 16:29:06.112] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
[2021-11-24 16:29:06.112] [trace] void SASIDEV::MsgIn() Starting Message in phase
[2021-11-24 16:29:06.112] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2021-11-24 16:29:06.112] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[2021-11-24 16:29:06.112] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2021-11-24 16:29:06.112] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
[2021-11-24 16:29:06.112] [trace] virtual void SCSIDEV::BusFree() Bus free phase
[2021-11-24 16:29:06.115] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
[2021-11-24 16:29:06.115] [trace] void SCSIDEV::MsgOut() ID 6
[2021-11-24 16:29:06.115] [trace] Message Out Phase
[2021-11-24 16:29:06.115] [trace] void SCSIDEV::MsgOut() ID 6
[2021-11-24 16:29:06.115] [trace] virtual void SCSIDEV::Receive()
[2021-11-24 16:29:06.115] [trace] virtual void SCSIDEV::Receive() length was != 0
[2021-11-24 16:29:06.115] [trace] void SCSIDEV::MsgOut() ID 6
[2021-11-24 16:29:06.115] [trace] virtual void SCSIDEV::Receive()
[2021-11-24 16:29:06.115] [trace] virtual void SCSIDEV::Receive() ctrl.phase: 10 (msgout)
[2021-11-24 16:29:06.115] [trace] Message code IDENTIFY $80, LUN 0 selected
[2021-11-24 16:29:06.115] [trace] virtual void SASIDEV::Command() Command Phase
[2021-11-24 16:29:06.116] [trace] virtual void SASIDEV::Command() CDB[0]=$3E
[2021-11-24 16:29:06.116] [trace] virtual void SASIDEV::Command() CDB[1]=$00
[2021-11-24 16:29:06.116] [trace] virtual void SASIDEV::Command() CDB[2]=$00
[2021-11-24 16:29:06.116] [trace] virtual void SASIDEV::Command() CDB[3]=$00
[2021-11-24 16:29:06.116] [trace] virtual void SASIDEV::Command() CDB[4]=$00
[2021-11-24 16:29:06.116] [trace] virtual void SASIDEV::Command() CDB[5]=$00
[2021-11-24 16:29:06.116] [trace] virtual void SASIDEV::Command() CDB[6]=$00
[2021-11-24 16:29:06.116] [trace] virtual void SASIDEV::Command() CDB[7]=$00
[2021-11-24 16:29:06.116] [trace] virtual void SASIDEV::Command() CDB[8]=$01
[2021-11-24 16:29:06.116] [trace] virtual void SASIDEV::Command() CDB[9]=$00
[2021-11-24 16:29:06.116] [trace] virtual void SCSIDEV::Execute() Execution phase command $3E
[2021-11-24 16:29:06.116] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $3E
[2021-11-24 16:29:06.116] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing ReadLong10 ($3E)
[2021-11-24 16:29:06.116] [debug] Error status: Sense Key $05, ASC $24, ASCQ $00
[2021-11-24 16:29:06.116] [trace] virtual void SCSIDEV::Error(ERROR_CODES::sense_key, ERROR_CODES::asc) Error (to status phase)
[2021-11-24 16:29:06.116] [trace] void SASIDEV::Status() Status phase
[2021-11-24 16:29:06.116] [trace] void SASIDEV::Status() Status Phase $02
[2021-11-24 16:29:06.116] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[2021-11-24 16:29:06.116] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
[2021-11-24 16:29:06.116] [trace] void SASIDEV::MsgIn() Starting Message in phase
[2021-11-24 16:29:06.116] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2021-11-24 16:29:06.117] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[... cut ...]
[2021-11-24 16:29:06.124] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[2021-11-24 16:29:06.124] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[2021-11-24 16:29:06.125] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
[2021-11-24 16:29:06.125] [trace] virtual void SCSIDEV::BusFree() Bus free phase

```

I ran the same tests against a real IBM drive. The drive was fine with a transfer length of 0, but refused to transfer 1 or 512 bytes. For these transfer lengths that's the same as with RaSCSI.

Pacjunk, 2021-11-25:
Here is the next dump... I don't know why $0204 is still missing. It still steps through to $0400, and the drive is marked as not compatible.

```
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.740] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=0 (with device)
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.740] [trace] void SCSIDEV::MsgOut() ID 0
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.740] [trace] Message Out Phase
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.740] [trace] void SCSIDEV::MsgOut() ID 0
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.740] [trace] virtual void SCSIDEV::Receive()
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.740] [trace] virtual void SCSIDEV::Receive() length was != 0
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.740] [trace] void SCSIDEV::MsgOut() ID 0
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.740] [trace] virtual void SCSIDEV::Receive()
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.740] [trace] virtual void SCSIDEV::Receive() ctrl.phase: 10 (msgout)
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.741] [trace] Message code IDENTIFY $C0, LUN 0 selected
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.741] [trace] virtual void SASIDEV::Command() Command Phase
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.741] [trace] virtual void SASIDEV::Command() CDB[0]=$3E
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.741] [trace] virtual void SASIDEV::Command() CDB[1]=$00
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.741] [trace] virtual void SASIDEV::Command() CDB[2]=$00
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.741] [trace] virtual void SASIDEV::Command() CDB[3]=$00
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.741] [trace] virtual void SASIDEV::Command() CDB[4]=$00
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.741] [trace] virtual void SASIDEV::Command() CDB[5]=$00
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.741] [trace] virtual void SASIDEV::Command() CDB[6]=$00
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.741] [trace] virtual void SASIDEV::Command() CDB[7]=$00
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.741] [trace] virtual void SASIDEV::Command() CDB[8]=$00
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.741] [trace] virtual void SASIDEV::Command() CDB[9]=$00
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.742] [trace] virtual void SCSIDEV::Execute() Execution phase command $3E
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.742] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $3E
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.742] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing ReadLong10 ($3E)
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.742] [trace] void SASIDEV::Status() Status phase
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.742] [trace] void SASIDEV::Status() Status Phase $00
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.742] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.742] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.742] [trace] void SASIDEV::MsgIn() Starting Message in phase
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.742] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.743] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.743] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.743] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.743] [trace] virtual void SCSIDEV::BusFree() Bus free phase
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.743] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=0 (with device)
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.743] [trace] void SCSIDEV::MsgOut() ID 0
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.743] [trace] Message Out Phase
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.743] [trace] void SCSIDEV::MsgOut() ID 0
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.743] [trace] virtual void SCSIDEV::Receive()
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.744] [trace] virtual void SCSIDEV::Receive() length was != 0
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.744] [trace] void SCSIDEV::MsgOut() ID 0
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.744] [trace] virtual void SCSIDEV::Receive()
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.744] [trace] virtual void SCSIDEV::Receive() ctrl.phase: 10 (msgout)
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:17.744] [trace] Message code IDENTIFY $C0, LUN 0 selected
[... cut ...]
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:18.378] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:18.378] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:18.379] [trace] virtual void SCSIDEV::BusFree() Bus free phase
Nov 25 12:02:18 raspberrypi RASCSI[437]: [2021-11-25 12:02:18.379] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=0 (with device)
```

Pacjunk, 2021-11-25:
The strange thing is that the debug from bluescsi shows $0204!  Strange...

```
Selection
MsgOut2
(246371)Command:3E:0:0:0:0:0:0:2:2:0:ID 2:LUN 0
[*Unknown]
Sts
MsgIn
BusFree
Selection
MsgOut2
(246532)Command:3E:0:0:0:0:0:0:2:3:0:ID 2:LUN 0
[*Unknown]
Sts
MsgIn
BusFree
Selection
MsgOut2
(246678)Command:3E:0:0:0:0:0:0:2:4:0:ID 2:LUN 0
[*Unknown]
Sts
MsgIn
BusFree
Selection
MsgOut2
(246824)Command:3E:0:0:0:0:0:0:2:5:0:ID 2:LUN 0
[*Unknown]
Sts
MsgIn
BusFree
Selection
MsgOut2
(246971)Command:3E:0:0:0:0:0:0:2:6:0:ID 2:LUN 0
[*Unknown]
Sts
MsgIn
BusFree
Selection
MsgOut2
```

rdmark, 2021-11-25:
@uweseimet I tentatively added this to the 21.11 roadmap. If you can figure out a stable solution in the next day or two we can get it into the upcoming release.

uweseimet, 2021-11-25:
@Pacjunk I'm afraid there is nothing more I can do. In case BlueSCSI behaves better somebody else might compare logs and source code in order to proceed. As already mentioned, the fact that $0204 is missing with RaSCSI but not with BlueSCSI may be an important detail, caused by a difference in behavior between BlueSCSI and RaSCSI.

Pacjunk, 2021-11-25:
OK, Thanks for your help

Pacjunk, 2022-01-11:
I have implemented READ/WRITE LONG(10) on the bluescsi, and now the host only sends readlong(10) commands for length 0, 1 and 200. It seems to be happy with that and does not do the command again. My implementation just transfers bytes instead of blocks (as the spec indicated). I can now successfully shadow (mirror) two drives under the operating system! Probably not a lot of use, but good for learning/experimentation.

uweseimet, 2022-09-03:
@Pacjunk I would like to get back to this ticket. I am still wondering how to improve READ/WRITE LONG support. Without explicit support for variable length byte transfers (rascsi essentially can only transfer full blocks) it cannot be properly resolved. The SCSI printer device is also suffering from this and currently uses a work-around.
Anyway, I would like to know which data your BlueSCSI implementation , e.g. for a 200 byte transfer. Are these only dummy data? The specs says that the data are vendor specific and often are checksum data. In case of a 200 byte WRITE LONG command, what does your implementation do with these 200 bytes?

Pacjunk, 2022-09-03:
My code reads/writes and transfers the requested number of bytes from/to the LBA specified. I just wrote another function similar to the one that transfers blocks, but transfers bytes instead. I don't know what data the machine is trying to get, but I supply real data. I think the mirroring feature that requires this will transfer different quantities when required. The 0,1 & 200 byte transfers are only done when the volumes are mounted to test compatibility. Supplying dummy data may work at this stage, but definitely won't work when a drive mirroring operation is required!

uweseimet, 2022-09-03:
@rdmark Please correct me if I am wrong: This means that when a sequence of sectors is copied this is done in chunks of 200 bytes, instead of copying in chunks of multiples of the sector size?

Pacjunk, 2022-09-03:
The 200 bytes is just a compatibility test at mount time. The actual transfers are all different lengths.
If the request is for 200 bytes, I supply 200 bytes. If the request is for 3159 bytes, that is how many I supply. I'm not doing multiples of the requested size.

uweseimet, 2022-09-03:
@Pacjunk So there are no additional data (checksums, vendor-specific data) contained in the data transferred by READ/WRITE LONG in your case? The only difference between copying a drive with regular read/write operations is that sequences of bytes instead of blocks are copied?

Pacjunk, 2022-09-03:
@uweseimet Yes, I just read/write bytes from the file and transfer bytes to/from the system. No other data. What the data contains, I don't know, and don't care!  The drive shadowing (mirroring) operation works successfully on my system, so what I am doing must be sufficient in this case.

uweseimet, 2022-09-03:
@Pacjunk I see. Any rascsi implementation would most likely have to bypass the sector-based cache and directly access the image file. Before each READ LONG the cache woulld have to be flushed in order for the file to be up to date before reading. And any WRITE LONG would have to somehow discard the cache, so that the modified data are immediately available for a subsequent sector-oriented access.
Or the sector-based cache might be discarded completely. as already suggested in the context of https://github.com/akuker/RASCSI/issues/335, so that all read/write operations would always be file-based and thus not restricted to multiples of sector sizes anymore.

Pacjunk, 2022-09-03:
Another advantage of bluescsi - No cache!  (which also means you can turn it off without corrupting images)

uweseimet, 2022-09-03:
Well, I think I already indicated that I would not mind for the cache to be removed completely. @akuker should probably comment on the because he is/was working on https://github.com/akuker/RASCSI/issues/335.

uweseimet, 2022-09-03:
These are the alternatives I would like to propose in order to make progress on this ticket and on the related ones:

1. Completely remove the existing cache. As far as I remember it turned out that there is no performance advantage anyway. Without the cache this ticket, https://github.com/akuker/RASCSI/issues/335 and probably also https://github.com/akuker/RASCSI/issues/807 are more realistic to be resolved. In addition, the code of the printer device, which is not block-oriented, would likely also profit. The existing codebase would become more transparent and better testable.
2. Keep the current cache implementatoin. In this case I have strong doubts that the tickets mentioned above will be resolved at all. Any potential resolution would have to find complicated ways to bypass the cache, making the read/write code even harder to maintain and test.

You won't be surprised that I am in favor of the first alternative ;-).

uweseimet, 2022-09-26:
While improving the code I noticed that there is no cache for SCMO, but only for other mass storage devices. Except for some mode pages SCMO is identical with SCHD/SCRM. In order to evaluate the effectiveness of the cache I used the same image file for SCHD and SCMO and measured the read transfer rates. You would have expected SCHD transfers (of the same data, so that they are cached) to be faster than SCMO transfers, but there was no difference.
@Pacjunk Maybe you can also check this, in case I have missed something?

@akuker By the way, this ticket and others (assigned to you) are still waiting for your feedback.

Pacjunk, 2022-09-28:
@uweseimet Unfortunately I am away for a week (possibly more) and I cannot check at the moment...

uweseimet, 2022-09-28:
@Pacjunk No problem I'll wait :).

uweseimet, 2022-10-11:
@Pacjunk After digging deeper in the code I may have been mistaken regarding the MO device not using the cache. Please do not spend any time on this (yet).

uweseimet, 2022-11-08:
Most likely requires resolving https://github.com/akuker/RASCSI/issues/940, https://github.com/akuker/RASCSI/issues/930 and https://github.com/akuker/RASCSI/issues/807 first. Resolving the latter is most likely a side-effect of resolving https://github.com/akuker/RASCSI/issues/940.