PiSCSI/piscsi #484: Heat question for SN74LS641-1

timmcross, 2021-11-25:
# Info

- Which version of Pi are you using:  3B+
- Which github revision of software: master
- Which board version: 2.4a
- Which computer is the RaSCSI connected to: p4 1.5GHz, Adaptec 2940, WinXP

# Describe the issue
First off, great project!  Built the board from the ground up and worked the first time!

One thing I noticed was that the SN74LS641-1 chips get quite hot during operation.  Is this expected?  Would small heatsinks be recommended? 

Thanks, Tim

rdmark, 2021-11-26:
@timmcross The transceivers do run hot! They're rated for 70C, and anecdotally they run fine naked. If you have room for them, heatsinks probably won't hurt though!

See this post for more details: https://68kmla.org/bb/index.php?threads/rascsi-development-thread.6868/post-426873

Pacjunk, 2021-11-27:
Here's mine. I use "RAM chip heatsinks". They still get quite hot, but anything is better than nothing. Haven't had any problems running for several hours.

[image]
.

hsiboy, 2021-11-30:
Yeah, mine run hot and sing! Depending on what Rascsi is doing, i can hear a faint whine coming from them.

rdmark, 2021-11-30:
Added an FAQ entry for this. Closing out this ticket! https://github.com/akuker/RASCSI/wiki/FAQ

closed 2021-11-30 by rdmark

schreinerman, 2023-01-10:
I'm using the FULLSPEC 2.4a board. Seems to be the heat in the system is generated by an high frequency oscillating ground plane or power supply. In addition the terminator circuit is also not the best and is drawing about 200mA (!) Unfortunately the 5V track width is not that wide. The decoupling capacitors are also missing, so the drivers are getting an unfiltered power input (high frequency noise/oscillation). Silicon and high frequency is generating a lot of heat.... 

So I will try to ad decoupling capacitors, powering terminal power with 3.3V (better 2.8V) via 220 Ohms, bridge ground planes with extra wires to minimise ground plane oscillation