PiSCSI/piscsi #496: [Wild Idea]: Allow RaSCSI to act as a SCSI/USB bridge for modern PCs

marciot, 2021-11-30:
Here is a pie-in-the-sky idea I had the other day.  It would be swell if RaSCSI was able to act as a USB bridge for SCSI devices :grin: The motivation for this even though dedicated USB/SCSI adapters exist, they are rare and very expensive, running several hundred dollars on eBay.

So the idea is that if I had a say a SCSI Zip drive, and I wanted to read it on a modern PC, I could connect a USB cable from my PC to the RaSCSI board and connect the SCSI device I wanted to use to RaSCSI. Then maybe I would check a checkbox next to a device in the web interface so that:

1. The USB port was switched into OTG mode and present itself using the USBMassStorageDeviceClass profile
2. The RaSCSI would begin forwarding SCSI commands between the USB interface and the SCSI device, allowing the PC to mount and read/write to the drive.

At first blush, it would seem like this would be a software only hack, but discussion thread https://github.com/akuker/RASCSI/issues/454 indicates that maybe it would require a hardware modification.

At the least, I would hope that this new use case would give some motivation for including this change on future hardware revisions. It certainly would allow the RaSCSI to do something no other projects are doing and maybe increase the user base for people who merely want to read data off a drive on their modern PCs.

akuker, 2021-11-30:
That's a pretty slick idea. Its definitely do-able, especially if you only have one SCSI device connected. I'm going to tag this as an "enhancement" though ;) 

The first step of just having the Raspberry Pi host be able to mount/read the SCSI device directly would be a huge one as well. From there, you could do a NFS mount (or something similar) to make the files accessible by another host.

uweseimet, 2021-11-30:
@marciot This idea is more or less the counterpart of https://github.com/akuker/RASCSI/issues/497, isn't it?

marciot, 2021-11-30:
> The first step of just having the Raspberry Pi host be able to mount/read the SCSI device directly would be a huge one as well. From there, you could do a NFS mount (or something similar) to make the files accessible by another host.

@akuker: I am not recommending that the Raspberry Pi mount the drive, instead I am recommending it pass-through the SCSI commands directly to the host via the USB. This would potentially allow devices other than drives to be used by the host.

> This idea is more or less the counterpart of #497, isn't it?

@uweseimet: No, as I said, my suggestion does not involve having the Raspberry Pi mount the drive.

So basically there are two different suggestions on the table:
1. Have the Raspberry Pi be able to mount a drive so that it can be accessed from the Pi and/or exported via NFS
1. Have the Raspberry Pi pass-thru the SCSI commands to the USB port so the connected PC host can mount the volume and/or send low level commands to a non-disk device.

These are two very different approaches.

marciot, 2021-11-30:
TBH, approach 2 could probably be tackled by something like the BlueSCSI, as when doing a simple-passthru adapter, Linux and WiFi are kind of redundant.

uweseimet, 2021-11-30:
@marciot Are there any advantages of approach 2. compared to 1. except for the PC being able to also access devices which do not have a filesystem the Pi can mount (and export)? 1. already means a lot of work, but 2. is IMO even more complicated. You say a "simple" pass-through adapter, but I doubt that  it's that simple ;-).
Do you have a (common) use case where 2. would be required and 1. would not work?

marciot, 2021-11-30:
@uweseimet: well, I don't have a specific use case in mind. Just thinking out loud. I agree that 1 would probably bring the most benefit to the most people.

I would think the two approaches would be roughly the same amount of work *if* someone has already written some code to allow a Raspberry Pi to act as USB Mass Storage device. If that is the case, then would only have to write the code to allow RaSCSI to send commands to the drive, which would also be required for the first approach. I know implementing USB Mass Storage devices is pretty common stuff in the microcontroller world, but I don't know how common it is in the RaspPi world.

marciot, 2021-11-30:
@uweseimet: Well, so I guess you can already share a mounted file system as a USB mass storage device, as described here:

https://magpi.raspberrypi.com/articles/pi-zero-w-smart-usb-flash-drive

So arguably if you implemented it the first way, you get the second way for free. It just would not support direct SCSI commands for esoteric devices and you couldn't, for instance, have the host eject the disk in a removable drive.

But I  agree the first method is more beneficial.

uweseimet, 2021-11-30:
@marciot Is the USB mass storage device interface easily mappable to the SCSI interface? If there is no (more or less) one-to-one mapping of USB interface to SCSI interface commands approach 2. may be impossible. There are interfaces that are so different that they are hardly mappable. From what I know mapping SCSI to NVMe is such a case because the NVMe protocol is fundamentally different from the SCSI protocol.

marciot, 2021-11-30:
I know that in the case of a USB Mass Storage Device, the interface is very simple, basically read a block at this address, write a block. This would have a slight advantage over method 1, in that the host could read file systems that the Raspberry Pi could not understand; but it still would not allow custom vendor specific commands.

I have not been able to determine yet is whether there is a SCSI specific USB storage class that would allow arbitrary commands to pass through unchanged. I am surmising that it does because USB SCSI adapters that work with the Zip drive *do* exist.

berlin4apk, 2021-12-12:
For the way 2* I have a direct use Case: I have a old high speed A3 scanner, which need the vendor software on a old windows version.
With the way 2, i cud have a qemu or virtualbox running the vendor software with windows and attach the scanner via rascsi

* Have the Raspberry Pi pass-thru the SCSI commands to the USB port so the connected PC host can mount the volume and/or send low level commands to a non-disk device.

ajacocks, 2022-02-22:
Given the very high cost of USB-SCSI interfaces, this would be a very valuable enhancement to RaSCSI. I have to go through gyrations every time I want to either image a SCSI disk, or to write an OS to one, that I don't have to go through for IDE or MFM devices.

DrReD, 2022-02-23:
Yeah. And old usb2scsi adapters or old PCs with SCSI cards ain't gonna live forever either... :-/

jeroenbrons, 2022-03-03:
maybe this feature could run a iSCSI target which other systems can use as a block device (no USB needed then)

DrReD, 2022-03-03:
It would be great already, if it could read SCSI devices on the Pi
itself. :)

raybellis, 2022-08-24:
There are other non-disk use cases - in my case it would be for audio sample interchange over SCSI.

dericed, 2024-02-17:
I've been fascinated to find this issue as I find it highly needed; I'm specifically referring to @marciot's [approach 2](https://github.com/PiSCSI/piscsi/issues/496#issuecomment-982856037) scenario. In my use-case I'd like the connectivity options to DDS drives to be more accessible (currently I use a Mac G3 with a SCSI PCI board and another with a SCSI<-> [firewire solution](https://www.synchrotech.com/product-1394/firewire-uscsi-conv_01.html) but I would love to use more modern computers). As noted, SCSI adapters are $$$ and hard to obtain, so an open hardware effort would be a great help to many that still need SCSI devices around. @akuker I dug through the sponsor section on this repo and saw general ways to sponsor this project but none to sponsor a particular issue. I'm quite interested in this proposed enhancement so if sponsorship helps, feel welcome to let me know how.

littlejackal, 2024-06-26:
+1 for this feature. My use case is to passthrough non-disk SCSI devices to a Linux host. 

For the sake of this conversation let’s assume that adding a SCSI add-on adapter isn’t desireable or feasible for any number of reasons. 

The envisioned use for me would be to (example) attach a SCSI tape drive to the RaSCSI, plug the RaSCS into the linux host as USB, and allow the system to control it as it would a directly connected tape device. Not something that is accomplished simply by exposing it via an NFS mount. 

I’m handwaving the complexity of actually having the operating system recognize the RaSCSI as a SCSI HBA, but conceptually this functionality if of great use to me.

ewanuno, 2024-09-02:
+1 for this idea. i would like to use scsi hardware samplers with modern computers too.

this would probably involve an iscsi server running on the PI sharing the scsi device over the network, and an iscsi client in an xp virtual machine with the old native sampler control software.

iscsi device sharing for non disk devices has been demonstrated with scsi scanners. 
https://sprocketfox.io/xssfox/2024/05/08/jbos/

a proof of concept for this could be done with pc with a scsi card running an iscsi target.

wmlive, 2024-09-03:
Adding a pointer to https://github.com/svenschnelle/teensyscsi where a working implemenation for USB SCSI Controller utilizing Teensy 4.1 has already been put  into reality three years ago.  There is an associated forum thread https://forum.classic-computing.de/forum/index.php?thread/24720-low-cost-usb-scsi-adapter/ at a German vintage computer forum. Maybe this might serve as inspiration for interested parties.

lucianjames, 2024-11-18:
+1 for this idea for the purpose of using non-storage SCSI devices (scanners in my desired use case)

akuker, 2024-11-18:
I've been looking into this a bit, and I'm thinking of taking a different approach, that I *think* will meet all your use cases.

What I'm thinking is to use the BlueSCSI as a USB/SCSI adapter that will work like those expensive USB/SCSI adapters on ebay.
[image]

This eliminates a lot of the complexity of the Linux kernel/drivers/etc. BlueSCSI also has better performance due to the PIO in the RP2040. 

I'm planning to do the work under this issue if you want to watch it. The ticket currently only calls out mass storage devices, but I'm thinking this can be expanded to also included Scanners
https://github.com/BlueSCSI/BlueSCSI-v2/issues/13

(One note: This feature will only work with the BlueSCSI "Desktop" version with initiator support.)

ZigZagJoe, 2024-11-19:
> I've been looking into this a bit, and I'm thinking of taking a different approach, that I _think_ will meet all your use cases.
> 
> What I'm thinking is to use the BlueSCSI as a USB/SCSI adapter that will work like those expensive USB/SCSI adapters on ebay. ![image](https://private-user-images.githubusercontent.com/34318535/387344337-d8a1d105-ea00-4b89-9e7b-85308a27ac08.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIwNTI2MzIsIm5iZiI6MTczMjA1MjMzMiwicGF0aCI6Ii8zNDMxODUzNS8zODczNDQzMzctZDhhMWQxMDUtZWEwMC00Yjg5LTllN2ItODUzMDhhMjdhYzA4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTE5VDIxMzg1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFiNGIzYTQzNjk5ZjIxZWM5YzJmMGY0YWI4MjgzZDNmYmUyOTZkMzBhMzJmYjFlMzZhMzliZGEwOWRkMDE4ZmQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.YMSVZ98jvRH1TjUQRS4g2E9eF7hDMKgwHbfUkD17yMQ)
> 
> This eliminates a lot of the complexity of the Linux kernel/drivers/etc. BlueSCSI also has better performance due to the PIO in the RP2040.
> 
> I'm planning to do the work under this issue if you want to watch it. The ticket currently only calls out mass storage devices, but I'm thinking this can be expanded to also included Scanners [BlueSCSI/BlueSCSI-v2#13](https://github.com/BlueSCSI/BlueSCSI-v2/issues/13)
> 
> (One note: This feature will only work with the BlueSCSI "Desktop" version with initiator support.)

FWIW, ZuluSCSI already has USB mass storage support implemented for SD card access (I wrote it :) ). It ought to be pretty easy to mate that up to the initiator code that's shared between Zulu and Bluescsi. It's been on the "when I get bored" to-do list for a while.