PiSCSI/piscsi #498: Linux kernel module for RaSCSI hardware

uweseimet, 2021-11-30:
By having a kernel modules that supports the RaSCSI hardware Linux running on the Pi could access SCSI devices via RaSCSI, which would effectively make RaSCSI a SCSI adapter for the Pi.

WarlockD, 2021-12-28:
Made a more proper initiator for the RASCSI so I can pull data off some old SCSI tapes.  Its possable, but I would like the pin configuration to be inline with the SMI interface so you can use DMA to transfer.   That being said, if you don't mind a slow interface it is possable but not on the zero.  I can say with 100% authority that single core bulk data trasfer breaks the wifi module:P

uweseimet, 2022-08-19:
Going to close this ticket as part of a ticket cleanup, because a kernel module would be nice, but is extremely unrealistic.

closed 2022-08-19 by uweseimet