PiSCSI/piscsi #503: Unsupported mode page $01 / ASR-10 "File Operation Error"

lukexi, 2021-12-01:
# Info

- Which version of Pi are you using: Pi 3+
- Which github revision of software: 2.10.1
- Which board version: 2.4
- Which computer is the RaSCSI connected to: Ensoniq ASR-10

# Describe the issue
I've got a new RaSCSI 2.4 to use with my ASR-10, which was previously using a SCSI2SD perfectly. Trying to format an RaSCSI attached disk or read an attached CD ISO results in "FILE OPERATION ERROR" on the ASR-10.

I've tried 512 and 1024 block sizes and multiple varieties of image types with the same result. Termination is on on the RaSCSI and it's the only device.

Here is the log starting with attaching a blank Seagate Barracuda 2GB image and then initiating "Format SCSI Drive" for SCSI Device 6 on the ASR-10, which ends with `virtual void Disk::ModeSense6(SASIDEV*) Unsupported mode page $01`

```
Dec 01 19:31:35 seascape-seastone RASCSI[550]: [2021-12-01 19:31:35.977] [trace] Received DEVICES_INFO command
Dec 01 19:31:35 seascape-seastone RASCSI[550]: [2021-12-01 19:31:35.985] [info] Validating: operation=ATTACH, device id=6, unit=0, type=SCHD, device params='file=Seagate_Barracuda_2GB.hds', vendor='SEAGATE', product='ST32550N', revision='0019', block size=512
Dec 01 19:31:35 seascape-seastone RASCSI[550]: [2021-12-01 19:31:35.987] [info] Executing: operation=ATTACH, device id=6, unit=0, type=SCHD, device params='file=Seagate_Barracuda_2GB.hds', vendor='SEAGATE', product='ST32550N', revision='0019', block size=512
Dec 01 19:31:35 seascape-seastone RASCSI[550]: [2021-12-01 19:31:35.989] [info] Attached SCHD device, ID 6, unit 0
Dec 01 19:31:36 seascape-seastone RASCSI[550]: [2021-12-01 19:31:36.033] [trace] Received SERVER_INFO command
Dec 01 19:31:36 seascape-seastone RASCSI[550]: [2021-12-01 19:31:36.051] [trace] Received DEVICES_INFO command
Dec 01 19:31:36 seascape-seastone RASCSI[550]: [2021-12-01 19:31:36.058] [trace] Received DEVICE_TYPES_INFO command
Dec 01 19:31:36 seascape-seastone RASCSI[550]: [2021-12-01 19:31:36.069] [trace] Received DEFAULT_IMAGE_FILES_INFO command
Dec 01 19:31:36 seascape-seastone RASCSI[550]: [2021-12-01 19:31:36.075] [trace] Received SERVER_INFO command
Dec 01 19:31:36 seascape-seastone RASCSI[550]: [2021-12-01 19:31:36.234] [trace] Received NETWORK_INTERFACES_INFO command
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.781] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.781] [trace] virtual void SASIDEV::Command() Command Phase
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] virtual void SASIDEV::Command() Command Byte 0: $03
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] virtual void SASIDEV::Command() Command Byte 1: $00
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] virtual void SASIDEV::Command() Command Byte 2: $00
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] virtual void SASIDEV::Command() Command Byte 3: $00
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] virtual void SASIDEV::Command() Command Byte 4: $0E
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] virtual void SASIDEV::Command() Command Byte 5: $00
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] virtual void SCSIDEV::Execute() Execution phase command $03
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $03
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing RequestSense ($03)
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] virtual void Disk::RequestSense(SASIDEV*) Status $00, Sense Key $00, ASC $00
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] void SASIDEV::DataIn() Going into Data-in Phase
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 14
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] virtual void SCSIDEV::Send() Move to next phase datain (6)
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] void SASIDEV::Status() Status phase
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] void SASIDEV::Status() Status Phase $00
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] void SASIDEV::MsgIn() Starting Message in phase
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.782] [trace] virtual void SCSIDEV::BusFree() Bus free phase
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.783] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.783] [trace] virtual void SASIDEV::Command() Command Phase
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.783] [trace] virtual void SASIDEV::Command() Command Byte 0: $12
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.783] [trace] virtual void SASIDEV::Command() Command Byte 1: $00
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.783] [trace] virtual void SASIDEV::Command() Command Byte 2: $00
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.783] [trace] virtual void SASIDEV::Command() Command Byte 3: $00
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.783] [trace] virtual void SASIDEV::Command() Command Byte 4: $24
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.783] [trace] virtual void SASIDEV::Command() Command Byte 5: $00
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.783] [trace] virtual void SCSIDEV::Execute() Execution phase command $12
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.783] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $12
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.783] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Inquiry ($12)
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.783] [trace] void SASIDEV::DataIn() Going into Data-in Phase
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.783] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 36
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.784] [trace] virtual void SCSIDEV::Send() Move to next phase datain (6)
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.784] [trace] void SASIDEV::Status() Status phase
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.784] [trace] void SASIDEV::Status() Status Phase $00
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.784] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.784] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.784] [trace] void SASIDEV::MsgIn() Starting Message in phase
Dec 01 19:31:54 seascape-seastone RASCSI[550]: [2021-12-01 19:31:54.784] [trace] void SASIDEV::MsgIn() Transitioning to Send()
[... 129 lines cut ...]
[... cut ...]
Dec 01 19:31:59 seascape-seastone RASCSI[550]: [2021-12-01 19:31:59.338] [trace] void SASIDEV::Status() Status phase
Dec 01 19:31:59 seascape-seastone RASCSI[550]: [2021-12-01 19:31:59.338] [trace] void SASIDEV::Status() Status Phase $02
Dec 01 19:31:59 seascape-seastone RASCSI[550]: [2021-12-01 19:31:59.338] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
Dec 01 19:32:03 seascape-seastone RASCSI[550]: [2021-12-01 19:32:03.336] [trace] virtual void SCSIDEV::BusFree() Bus free phase
```

lukexi, 2021-12-01:
Here is a similar log for attempting to load a Roland ISO image attached at SCSI 6
```
pi@seascape-seastone:~ $ journalctl -u rascsi -f
-- Logs begin at Wed 2021-12-01 20:39:25 GMT. --
Dec 01 20:41:42 seascape-seastone RASCSI[548]: [2021-12-01 20:41:42.996] [info] Detached SCHD device with ID 6, unit 0
Dec 01 20:41:54 seascape-seastone RASCSI[548]: [2021-12-01 20:41:54.973] [info] Validating: operation=ATTACH, device id=6, unit=0, type=SCCD, device params='file=Roland_-_L-CD1_S-50_Factory_Library.iso', vendor='', product='', revision='', block size=0
Dec 01 20:41:54 seascape-seastone RASCSI[548]: [2021-12-01 20:41:54.976] [info] Executing: operation=ATTACH, device id=6, unit=0, type=SCCD, device params='file=Roland_-_L-CD1_S-50_Factory_Library.iso', vendor='', product='', revision='', block size=0
Dec 01 20:41:54 seascape-seastone RASCSI[548]: [2021-12-01 20:41:54.977] [info] Attached read-only SCCD device, ID 6, unit 0
Dec 01 20:42:19 seascape-seastone RASCSI[548]: [2021-12-01 20:42:19.694] [trace] Received SERVER_INFO command
Dec 01 20:42:19 seascape-seastone RASCSI[548]: [2021-12-01 20:42:19.711] [trace] Received DEVICES_INFO command
Dec 01 20:42:19 seascape-seastone RASCSI[548]: [2021-12-01 20:42:19.716] [trace] Received DEVICE_TYPES_INFO command
Dec 01 20:42:19 seascape-seastone RASCSI[548]: [2021-12-01 20:42:19.726] [trace] Received DEFAULT_IMAGE_FILES_INFO command
Dec 01 20:42:19 seascape-seastone RASCSI[548]: [2021-12-01 20:42:19.731] [trace] Received SERVER_INFO command
Dec 01 20:42:19 seascape-seastone RASCSI[548]: [2021-12-01 20:42:19.948] [trace] Received NETWORK_INTERFACES_INFO command
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.111] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.111] [trace] virtual void SASIDEV::Command() Command Phase
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.111] [trace] virtual void SASIDEV::Command() CDB[0]=$03
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.111] [trace] virtual void SASIDEV::Command() CDB[1]=$00
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.111] [trace] virtual void SASIDEV::Command() CDB[2]=$00
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.111] [trace] virtual void SASIDEV::Command() CDB[3]=$00
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.111] [trace] virtual void SASIDEV::Command() CDB[4]=$0E
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.111] [trace] virtual void SASIDEV::Command() CDB[5]=$00
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.111] [trace] virtual void SCSIDEV::Execute() Execution phase command $03
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.111] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $03
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.111] [trace] virtual bool SCSICD::Dispatch(SCSIDEV*) Calling base class for dispatching $03
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.111] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing RequestSense ($03)
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.111] [trace] virtual void Disk::RequestSense(SASIDEV*) Status $00, Sense Key $00, ASC $00
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.111] [trace] void SASIDEV::DataIn() Going into Data-in Phase
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.112] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 14
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.112] [trace] virtual void SCSIDEV::Send() Move to next phase datain (6)
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.112] [trace] void SASIDEV::Status() Status phase
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.112] [trace] void SASIDEV::Status() Status Phase $00
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.112] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.112] [trace] virtual void SCSIDEV::Send() Move to next phase status (8)
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.112] [trace] void SASIDEV::MsgIn() Starting Message in phase
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.112] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.112] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.112] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.112] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.112] [trace] virtual void SCSIDEV::BusFree() Bus free phase
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.112] [trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.113] [trace] virtual void SASIDEV::Command() Command Phase
[... cut ...]
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.119] [trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.119] [trace] void SASIDEV::MsgIn() Transitioning to Send()
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.119] [trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
Dec 01 20:42:34 seascape-seastone RASCSI[548]: [2021-12-01 20:42:34.120] [trace] virtual void SCSIDEV::BusFree() Bus free phase
```

uweseimet, 2021-12-01:
@lukexi Your issue can only be addressed if the data to be returned by the missing mode pages are known. Otherwise it is impossible to support these pages. Do you have any information on the structure of these mode pages?

lukexi, 2021-12-01:
If I understand right 01 is "Read-Write Error Recovery" and 15 is "Completed" or "Extended", according to https://www.seagate.com/files/staticfiles/support/docs/manual/Interface%20manuals/100293068j.pdf (Page 59) and https://en.wikipedia.org/wiki/SCSI_mode_page — would it be enough to look at how [SCSI2SD](http://www.codesrc.com/gitweb/index.cgi?p=SCSI2SD-V6.git;a=summary) handles them, since it works very nicely with the ASR-10?

uweseimet, 2021-12-01:
@lukexi If somebody should implement these pages it's best (in the case of well-documented pages) to use an official documentation. Looking at other software instead of the specification will result in potentially copying wrong behavior.

lukexi, 2021-12-02:
C++ isn't the issue, understanding SCSI is : ) 

(especially since I can't seem to access any of the documents on t10.org... will do my best though!)

@rdmark suggested these might be spurious errors as well — I assume someone more familiar with RaSCSI and SCSI can suss that out quicker than I can at the moment (from what I could tell from perusing SCSI2SD it handles neither of these mode pages specially, though I could be wrong...).

lukexi, 2021-12-02:
Okay, I think I've found the first issue — according to
[Oracle](https://docs.oracle.com/en/storage/tape-storage/storagetek-sl150-modular-tape-library/slorm/mode-sense-6-byte-1ah-and-mode-sense-10-byte-5ah.html#GUID-AD509871-D1C6-49A8-8B83-E45F0DCEB0EC) and [IBM](https://www.ibm.com/docs/en/ts4500-tape-library?topic=commands-mode-sense-6-x1a) we should be truncating responses to Mode Sense to the requested data length. Instead here https://github.com/akuker/RASCSI/blob/master/src/raspberrypi/devices/disk.cpp#L796 RaSCSI is erroring out if we have more data to send than the request.

Deleting lines 796-802 to remove that check + the attempt to change length to fit our data, and truncating instead (i.e. returning the same length requested) lets me get quite a bit further, with the format seeming to complete after a few minutes with no errors on RaSCSI's side, though then failing again on the ASR with a new message "Bad Device ID" that I'll start investigating now.

uweseimet, 2021-12-02:
@lukexi Yes, this may indeed be an issue with how RaSCSI handles the length. There is also an indication about an issue with the length in the logs. Would be great if you could create a PR as soon as you have identified/fixed the issue. Regarding the T10 documents, the final specifications need a registration, but you can download the latest drafts, which are totally fine for our purposes.

uweseimet, 2021-12-14:
@lukexi Any news on this? You sounded as if you could prepare a PR in order to address at least a part of this issue.

lukexi, 2021-12-16:
Hi Uwe, I haven't had a time to do a second dive yet — I'm hoping to figure out the final issue on the ASR-10 (since what I've done so far seems to only half-fix it) before I send a pull request. Should have some time to dig in again in the next week or two!

uweseimet, 2022-01-18:
@lukexi Any news? Note that your issue may be related to https://github.com/akuker/RASCSI/issues/614. There currently is a bugfix branch fix_mode_sense. It would be very helpful if you could test against this branch asap.

uweseimet, 2022-01-20:
@lukexi The latest MODE SENSE changes are now available in the develop branch. Can you please test with this branch and provide feedback?

lukexi, 2022-01-21:
@uweseimet hello!! I had a chance to test the develop branch just now and while I can confirm that it provides the same progress my experimental patch did, the ASR still fails with "Bad Device ID" after seeming to make it through the formatting process, suggesting either that formatting wasn't successful or that the disk isn't being read successfully (according to https://www.chickensys.com/ensoniq/epsasr/005.html), so something's still not quite right...

lukexi, 2022-01-21:
yep, I'm running the trace right now : ) — will post as soon as it's done. (I don't think there's an option to just create the filesystem on the ASR, it's part of the formatting process as far as I know)

lukexi, 2022-01-22:
A bit more information — testing with a 1 megabyte disk image, the format seems to complete successfully (DISK COMMAND COMPLETED) — great! But then fails with "FILE OPERATION ERROR" when trying to write either the OS or a simple instrument file to the SCSI drive.

Log of trying to write a file to the disk
```
[trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
[trace] virtual void SASIDEV::Command() Command Phase
[trace] virtual void SASIDEV::Command() CDB[0]=$00
[trace] virtual void SASIDEV::Command() CDB[1]=$00
[trace] virtual void SASIDEV::Command() CDB[2]=$00
[trace] virtual void SASIDEV::Command() CDB[3]=$00
[trace] virtual void SASIDEV::Command() CDB[4]=$00
[trace] virtual void SASIDEV::Command() CDB[5]=$00
[trace] virtual void SCSIDEV::Execute() Execution phase command $00
[debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $00
[debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing TestUnitReady ($00)
[trace] bool Disk::CheckReady() Disk is ready
[trace] void SASIDEV::Status() Status phase
[trace] void SASIDEV::Status() Status Phase $00
[trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[trace] virtual void SCSIDEV::Send() Move to next phase status (8)
[trace] void SASIDEV::MsgIn() Starting Message in phase
[trace] void SASIDEV::MsgIn() Transitioning to Send()
[trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[trace] void SASIDEV::MsgIn() Transitioning to Send()
[trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
[trace] virtual void SCSIDEV::BusFree() Bus free phase
[trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
[trace] virtual void SASIDEV::Command() Command Phase
[trace] virtual void SASIDEV::Command() CDB[0]=$12
[trace] virtual void SASIDEV::Command() CDB[1]=$00
[trace] virtual void SASIDEV::Command() CDB[2]=$00
[trace] virtual void SASIDEV::Command() CDB[3]=$00
[trace] virtual void SASIDEV::Command() CDB[4]=$04
[trace] virtual void SASIDEV::Command() CDB[5]=$00
[trace] virtual void SCSIDEV::Execute() Execution phase command $12
[debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $12
[debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Inquiry ($12)
[trace] void SASIDEV::DataIn() Going into Data-in Phase
[trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 4
[trace] virtual void SCSIDEV::Send() Move to next phase datain (6)
[trace] void SASIDEV::Status() Status phase
[trace] void SASIDEV::Status() Status Phase $00
[trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[trace] virtual void SCSIDEV::Send() Move to next phase status (8)
[trace] void SASIDEV::MsgIn() Starting Message in phase
[trace] void SASIDEV::MsgIn() Transitioning to Send()
[trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[trace] void SASIDEV::MsgIn() Transitioning to Send()
[trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
[trace] virtual void SCSIDEV::BusFree() Bus free phase
[trace] virtual void SCSIDEV::Selection() Selection Phase ID=6 (with device)
[trace] virtual void SASIDEV::Command() Command Phase
[trace] virtual void SASIDEV::Command() CDB[0]=$28
[trace] virtual void SASIDEV::Command() CDB[1]=$00
[trace] virtual void SASIDEV::Command() CDB[2]=$00
[trace] virtual void SASIDEV::Command() CDB[3]=$00
[trace] virtual void SASIDEV::Command() CDB[4]=$00
[trace] virtual void SASIDEV::Command() CDB[5]=$05
[trace] virtual void SASIDEV::Command() CDB[6]=$00
[trace] virtual void SASIDEV::Command() CDB[7]=$00
[trace] virtual void SASIDEV::Command() CDB[8]=$01
[trace] virtual void SASIDEV::Command() CDB[9]=$00
[trace] virtual void SCSIDEV::Execute() Execution phase command $28
[debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
[debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Read10 ($28)
[debug] virtual void Disk::Read10(SASIDEV*) READ(10) command record=$00000005 blocks=1
[trace] virtual int Disk::Read(const DWORD*, BYTE*, uint64_t)
[trace] bool Disk::CheckReady() Disk is ready
[trace] bool DiskTrack::ReadSector(BYTE*, int) const reading sector: 5
[trace] void Disk::Read(SASIDEV*, uint64_t) ctrl.length is 512
[... 826 lines cut ...]
[trace] virtual void SASIDEV::Command() CDB[7]=$00
[trace] virtual void SASIDEV::Command() CDB[8]=$01
[trace] virtual void SASIDEV::Command() CDB[9]=$00
[trace] virtual void SCSIDEV::Execute() Execution phase command $28
[debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
[debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Read10 ($28)
[debug] virtual void Disk::Read10(SASIDEV*) READ(10) command record=$00000007 blocks=1
[trace] virtual int Disk::Read(const DWORD*, BYTE*, uint64_t)
[trace] bool Disk::CheckReady() Disk is ready
[trace] bool DiskTrack::ReadSector(BYTE*, int) const reading sector: 7
[trace] void Disk::Read(SASIDEV*, uint64_t) ctrl.length is 512
[trace] void SASIDEV::DataIn() Going into Data-in Phase
[trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 512
[trace] virtual void SCSIDEV::Send() Move to next phase datain (6)
[trace] void SASIDEV::Status() Status phase
[trace] void SASIDEV::Status() Status Phase $00
[trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[trace] virtual void SCSIDEV::Send() Move to next phase status (8)
[trace] void SASIDEV::MsgIn() Starting Message in phase
[trace] void SASIDEV::MsgIn() Transitioning to Send()
[trace] virtual void SCSIDEV::Send() sending handhake with offset 0, length 1
[trace] void SASIDEV::MsgIn() Transitioning to Send()
[trace] virtual void SCSIDEV::Send() Move to next phase msgin (9)
[trace] virtual void SCSIDEV::BusFree() Bus free phase
```

uweseimet, 2022-01-22:
@lukexi I'm afraid there is no SCSI FORMAT command anywhere, unless I missed it. In addition, the logs do not appear to contain show any error.

uweseimet, 2022-08-27:
@lukexi Can you please check my last comment from January and double-check? Regarding mode page 1, with the current release 22.08.02 or the develop branch it should be returned. Can you please verify?

uweseimet, 2022-09-01:
@lukexi Just a reminder: If there is no feedback we will have to close this ticket because it is unresolvable without feedback.

lukexi, 2022-09-01:
Thanks Uwe! I had it on my todo list, I'll get down to check it out on the ASR it in the next few days!

uweseimet, 2022-09-01:
@lukexi Sounds good!

lukexi, 2022-09-04:
@uweseimet I just got my Pi rebuilt on the latest develop branch and we're still ending the format with "Bad Device ID" on the ASR-10. I'll run another format with tracing enabled and post it here when done!

lukexi, 2022-09-04:
With [trace] level debugging, we get a different error on the ASR shortly after beginning formatting, "Unconnected SCSI device". Here's the end of that log with a visible error...
```
[2022-09-04 04:20:11.318] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $2A
[2022-09-04 04:20:11.318] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = Disk; DWORD = unsigned int] Executing Write10 ($2A)
[2022-09-04 04:20:11.318] [trace] bool Disk::CheckAndGetStartAndCount(uint64_t&, uint32_t&, Disk::access_mode) READ/WRITE/VERIFY/SEEK command record=$00000173 blocks=1
[2022-09-04 04:20:11.318] [trace] void PrimaryDevice::CheckReady() Device is ready
[2022-09-04 04:20:11.318] [trace] virtual void ScsiController::DataOut() Data out phase
[2022-09-04 04:20:11.318] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 3
[2022-09-04 04:20:11.318] [trace] void ScsiController::Receive()
[2022-09-04 04:20:11.318] [trace] void ScsiController::Receive() Length is 512 bytes
[2022-09-04 04:20:11.323] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 3
[2022-09-04 04:20:11.323] [trace] void ScsiController::Receive()
[2022-09-04 04:20:11.323] [trace] void ScsiController::Receive() ctrl.phase: 7 (dataout)
[2022-09-04 04:20:11.323] [trace] virtual void Disk::Write(const DWORD*, BYTE*, uint64_t)
[2022-09-04 04:20:11.323] [trace] virtual void ScsiController::Status() Status Phase $00
[2022-09-04 04:20:11.323] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 3
[2022-09-04 04:20:11.323] [trace] void ScsiController::Send() Sending handhake with offset 0, length 1
[2022-09-04 04:20:11.323] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 3
[2022-09-04 04:20:11.323] [trace] void ScsiController::Send() Move to next phase status (8)
[2022-09-04 04:20:11.323] [trace] virtual void ScsiController::MsgIn() Message In phase
[2022-09-04 04:20:11.323] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 3
[2022-09-04 04:20:11.323] [trace] virtual void ScsiController::MsgIn() Transitioning to Send()
[2022-09-04 04:20:11.323] [trace] void ScsiController::Send() Sending handhake with offset 0, length 1
[2022-09-04 04:20:11.323] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 3
[2022-09-04 04:20:11.323] [trace] virtual void ScsiController::MsgIn() Transitioning to Send()
[2022-09-04 04:20:11.324] [trace] void ScsiController::Send() Move to next phase msgin (9)
[2022-09-04 04:20:11.324] [trace] virtual void ScsiController::BusFree() Bus free phase
[2022-09-04 04:20:11.324] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 3
[2022-09-04 04:20:11.324] [trace] virtual void ScsiController::Selection() Selection Phase Target ID=0
[2022-09-04 04:20:11.502] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 3
[2022-09-04 04:20:11.502] [trace] virtual void ScsiController::Command() Command Phase
[2022-09-04 04:20:14.502] [error] virtual void ScsiController::Command() No command bytes received
[2022-09-04 04:20:14.502] [debug] Error status: Sense Key $0B, ASC $00, ASCQ $00
[2022-09-04 04:20:14.502] [trace] virtual void ScsiController::Error(scsi_defs::sense_key, scsi_defs::asc, scsi_defs::status) Error (to status phase)
[2022-09-04 04:20:14.502] [trace] virtual void ScsiController::Status() Status Phase $02
[2022-09-04 04:20:14.502] [trace] virtual BUS::phase_t ScsiController::Process(int) Initiator ID is 3
[2022-09-04 04:20:14.502] [trace] void ScsiController::Send() Sending handhake with offset 0, length 1
[2022-09-04 04:20:17.502] [trace] virtual void ScsiController::BusFree() Bus free phase
```

lukexi, 2022-09-04:
Dropping to [debug] fixes that, so we can at least see a debug trace which looks like this
```
[2022-09-04 04:24:01.283] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $03
[2022-09-04 04:24:01.283] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = PrimaryDevice; DWORD = unsigned int] Executing RequestSense ($03)
[2022-09-04 04:24:01.284] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $12
[2022-09-04 04:24:01.284] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = PrimaryDevice; DWORD = unsigned int] Executing Inquiry ($12)
[2022-09-04 04:24:01.285] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $03
[2022-09-04 04:24:01.286] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = PrimaryDevice; DWORD = unsigned int] Executing RequestSense ($03)
[2022-09-04 04:24:01.287] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $25
[2022-09-04 04:24:01.287] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = Disk; DWORD = unsigned int] Executing ReadCapacity10 ($25)
[2022-09-04 04:24:01.288] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $00
[2022-09-04 04:24:01.288] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = PrimaryDevice; DWORD = unsigned int] Executing TestUnitReady ($00)
[2022-09-04 04:24:01.289] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $28
[2022-09-04 04:24:01.289] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = Disk; DWORD = unsigned int] Executing Read10 ($28)
[2022-09-04 04:24:02.139] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $00
[2022-09-04 04:24:02.139] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = PrimaryDevice; DWORD = unsigned int] Executing TestUnitReady ($00)
[2022-09-04 04:24:02.140] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $1A
[2022-09-04 04:24:02.140] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = ModePageDevice; DWORD = unsigned int] Executing ModeSense6 ($1A)
[2022-09-04 04:24:02.141] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $04
[2022-09-04 04:24:02.141] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = Disk; DWORD = unsigned int] Executing FormatUnit ($04)
[2022-09-04 04:24:02.141] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $1A
[2022-09-04 04:24:02.142] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = ModePageDevice; DWORD = unsigned int] Executing ModeSense6 ($1A)
[2022-09-04 04:24:02.142] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $12
[2022-09-04 04:24:02.142] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = PrimaryDevice; DWORD = unsigned int] Executing Inquiry ($12)
[2022-09-04 04:24:02.144] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $25
[2022-09-04 04:24:02.144] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = Disk; DWORD = unsigned int] Executing ReadCapacity10 ($25)
[2022-09-04 04:24:02.145] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $2A
[2022-09-04 04:24:02.145] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = Disk; DWORD = unsigned int] Executing Write10 ($2A)
[2022-09-04 04:24:02.150] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $2A
[2022-09-04 04:24:02.150] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = Disk; DWORD = unsigned int] Executing Write10 ($2A)
[2022-09-04 04:24:02.158] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $2A
[2022-09-04 04:24:02.158] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = Disk; DWORD = unsigned int] Executing Write10 ($2A)
[2022-09-04 04:24:02.171] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $2A
[2022-09-04 04:24:02.171] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = Disk; DWORD = unsigned int] Executing Write10 ($2A)
[2022-09-04 04:24:02.180] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $2A
[2022-09-04 04:24:02.180] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = Disk; DWORD = unsigned int] Executing Write10 ($2A)
[2022-09-04 04:24:02.189] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $2A
[~48000 repeats]
[2022-09-04 04:26:17.400] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = Disk; DWORD = unsigned int] Executing Write10 ($2A)
[2022-09-04 04:26:17.406] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $2A
[2022-09-04 04:26:17.406] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = Disk; DWORD = unsigned int] Executing Write10 ($2A)
[2022-09-04 04:26:17.411] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $2A
[2022-09-04 04:26:17.411] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = Disk; DWORD = unsigned int] Executing Write10 ($2A)
[2022-09-04 04:26:17.417] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $2A
[2022-09-04 04:26:17.417] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = Disk; DWORD = unsigned int] Executing Write10 ($2A)
[2022-09-04 04:26:17.422] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $2A
[2022-09-04 04:26:17.422] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = Disk; DWORD = unsigned int] Executing Write10 ($2A)
[... cut ...]
[2022-09-04 04:27:19.322] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = PrimaryDevice; DWORD = unsigned int] Executing TestUnitReady ($00)
[2022-09-04 04:27:19.323] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $28
[2022-09-04 04:27:19.323] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = Disk; DWORD = unsigned int] Executing Read10 ($28)
[2022-09-04 04:27:19.330] [debug] ++++ CMD ++++ void ScsiController::Execute() Executing command $28
[2022-09-04 04:27:19.330] [debug] bool Dispatcher<T>::Dispatch(T*, DWORD) [with T = Disk; DWORD = unsigned int] Executing Read10 ($28)
```

uweseimet, 2022-09-04:
@lukexi Thank you for the logs. Unless I have missed something the logs contain no error messages, and also do not contain the FORMAT command and also no MODE SENSE command for page 1. Can you please attach a log where these commands are visible? Maybe we should concentrate on the MODE SENSE for pages $01 and $15. My claim was that there should not be an error message when page $01 is requested, that this is what I would like to verify. Page $15 is not supported, so an error message is expected here. The commands for $28 (READ) and $2A (WRITE) are not that interested, they just read and write sectors.

lukexi, 2022-09-04:
I see "Executing FormatUnit ($04)" in the [debug] log, line 18? I'll try running some more operations to get page 01 to trigger today!

lukexi, 2022-09-04:
And yes, there are no RaSCSI errors reported in the [debug] log, the error is on the ASR-10 at the end of the Formatting process, just after the final Read commands which I assume are checking that the format operation succeeded and it's not liking what it sees, suggesting RaSCSI is either not writing something correctly or sending back bad data during the reads I suppose.

uweseimet, 2022-09-04:
@lukexi I'm afraid I cannot help with the format operation problem, if the rascsi logs do not reveal anything unusual. But if the mode page problem is gone (which appears to be the case) that's a step forwared. I suggest that you create a new ticket with the remaining (format) issue, so that this ticket can be closed.
A new, more precise and up to date ticket about the remaining issue would make it easier for somebody to work on it.
Do you agree with this approach?

lukexi, 2022-09-04:
Yes very much so! I'll do a run through today and capture some precise logs and make an issue for each. Thanks for all your help!

closed 2022-09-04 by lukexi

uweseimet, 2022-09-04:
@lukexi Sounds good. Thank you for your feedback!

uweseimet, 2022-09-04:
@lukexi By the way, please use the very latest develop branch for any testing.