PiSCSI/piscsi #504: Incompatibility: Yamaha EX5R synth/sampler

bzeiss, 2021-12-02:
Likely one of the more rare use cases, I thought it might be useful to document the findings anyway.

# Info

- Which version of Pi are you using: Raspberry Pi 3
- Which github revision of software: November 2021 Release
- Which board version: 2.4
- Which computer is the RaSCSI connected to: Yamaha EX5R synthesizer/sampler

# Describe the issue

There is an incompatibility issue with the Yamaha EX5R synth/sampler. 

Setup:
- RASCSI is connected through an internal SCSI cable. SCSI2SD works fine with the same cable and connection setup.
-  Termination is enabled on the ASIB-1 SCSI expansion module for the synth. This is the setting that has worked correctly with SCSI2SD.
- Termination on the RASCSI is enabled
- One hard disk is configured: 1000MB generic hdd. I have tried as well: named hard drives with sizes from 500mb to 2000mb (max is 2gb according to what I have found). Removable drives don't work as well.

Steps:
- Go to disk
- Go to device format
- Select HD0 via rotary wheel

Results:

1) Device HD0 selected via rotary wheel yields "illegal disk" after a while:

`Dec  2 19:19:38 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:38.781] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $00
Dec  2 19:19:38 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:38.781] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing TestUnitReady ($00)
Dec  2 19:19:38 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:38.783] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $25
Dec  2 19:19:38 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:38.783] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing ReadCapacity10 ($25)
Dec  2 19:19:38 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:38.784] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Dec  2 19:19:38 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:38.784] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Read10 ($28)
Dec  2 19:19:38 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:38.784] [debug] virtual void Disk::Read10(SASIDEV*) READ(10) command record=$00000001 blocks=1
Dec  2 19:19:38 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:38.792] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $00
Dec  2 19:19:38 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:38.792] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing TestUnitReady ($00)
Dec  2 19:19:38 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:38.793] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $25
Dec  2 19:19:38 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:38.794] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing ReadCapacity10 ($25)
Dec  2 19:19:38 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:38.796] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Dec  2 19:19:38 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:38.796] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Read10 ($28)
Dec  2 19:19:38 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:38.796] [debug] virtual void Disk::Read10(SASIDEV*) READ(10) command record=$00000001 blocks=1
Dec  2 19:19:39 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:39.038] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $00
Dec  2 19:19:39 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:39.038] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing TestUnitReady ($00)
Dec  2 19:19:39 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:39.039] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $25
Dec  2 19:19:39 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:39.039] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing ReadCapacity10 ($25)
Dec  2 19:19:39 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:39.041] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Dec  2 19:19:39 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:39.041] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Read10 ($28)
Dec  2 19:19:39 rascsi-ex5r RASCSI[508]: [2021-12-02 19:19:39.042] [debug] virtual void Disk::Read10(SASIDEV*) READ(10) command record=$00000001 blocks=1
`

2) If I ignore the illegal disk warning and try to format the HD0 disk anyway:

`Dec  2 19:21:01 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:01.607] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $00
Dec  2 19:21:01 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:01.607] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing TestUnitReady ($00)
Dec  2 19:21:01 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:01.609] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $25
Dec  2 19:21:01 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:01.609] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing ReadCapacity10 ($25)
Dec  2 19:21:01 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:01.610] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $28
Dec  2 19:21:01 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:01.611] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Read10 ($28)
Dec  2 19:21:01 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:01.611] [debug] virtual void Disk::Read10(SASIDEV*) READ(10) command record=$00000001 blocks=1
Dec  2 19:21:01 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:01.731] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $2A
Dec  2 19:21:01 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:01.731] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Write10 ($2A)
Dec  2 19:21:01 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:01.731] [debug] virtual void Disk::Write10(SASIDEV*) WRITE(10) command record=$00000001 blocks=1
Dec  2 19:21:01 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:01.849] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $2A
Dec  2 19:21:01 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:01.850] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Write10 ($2A)
Dec  2 19:21:01 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:01.850] [debug] virtual void Disk::Write10(SASIDEV*) WRITE(10) command record=$00000100 blocks=1
Dec  2 19:21:01 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:01.975] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $2A
Dec  2 19:21:01 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:01.975] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Write10 ($2A)
Dec  2 19:21:01 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:01.975] [debug] virtual void Disk::Write10(SASIDEV*) WRITE(10) command record=$00000102 blocks=8
Dec  2 19:21:01 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:01.985] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $2A
Dec  2 19:21:01 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:01.985] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Write10 ($2A)
Dec  2 19:21:01 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:01.985] [debug] virtual void Disk::Write10(SASIDEV*) WRITE(10) command record=$000001FC blocks=8
Dec  2 19:21:02 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:02.114] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $2A
Dec  2 19:21:02 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:02.114] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing Write10 ($2A)
Dec  2 19:21:02 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:02.114] [debug] virtual void Disk::Write10(SASIDEV*) WRITE(10) command record=$0000010A blocks=8
Dec  2 19:21:05 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:05.606] [error] virtual void SCSIDEV::Receive() Not able to receive all data. Going to error
Dec  2 19:21:05 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:05.609] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $03
Dec  2 19:21:05 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:05.609] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing RequestSense ($03)
Dec  2 19:21:05 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:05.611] [debug] ++++ CMD ++++ virtual void SCSIDEV::Execute() Executing command $03
Dec  2 19:21:05 rascsi-ex5r RASCSI[508]: [2021-12-02 19:21:05.611] [debug] virtual bool Disk::Dispatch(SCSIDEV*) Executing RequestSense ($03)
`

rdmark, 2021-12-02:
@bzeiss Can you please provide trace level logs as well?

bzeiss, 2021-12-02:
Sure, I'm on it already.

uweseimet, 2021-12-02:
The log effectively says that there is no issue when writing a single sector, but when writing 8 sectors RaSCSI claims that the device does not send enough data.
@bzeiss I added additional logging for a further analysis. Can you please create a log on trace level based on the "issue_504" branch? Only a small port of the log is relevant, namely about 100 lines before and after the message that says "Not able to receive all data. Going to error"

bzeiss, 2021-12-03:
Absolutely, I'm happy to help any way I can.

trace log cut about +/- 100 lines around "going to error" (I tried to find some start point that looked reasonable):

[scsi-error-going-to-error.txt](https://github.com/akuker/RASCSI/files/7645819/scsi-error-going-to-error.txt)

full trace log:

[scsi-error.txt](https://github.com/akuker/RASCSI/files/7645817/scsi-error.txt)

bzeiss, 2021-12-03:
I have three RASCSI devices and lots of Pis to test with. I have my doubts that this is a hardware error, but I will double check nonetheless. Will report back later.

uweseimet, 2021-12-03:
@bzeiss Sounds good. If you can reproduce the problem also with a different hardware combination it is likely that your Yamaha's SCSI signal handling slightly differs from the signal handling of other (working) computer hardware, and that the RaSCSI software (in contrast to SCSI2SD) has issues with this different signal handling, resulting in RaSCSI assuming an early end of the data transfer. In this case it might either be that your Yamaha is worng (regarding the usage of bus signals) or the RaSCSI software is.

bzeiss, 2021-12-03:
So I have tried pi3, pi zero 2s and the three rascsi 2.4 hats in different kinds of combinations and also a different cable. I also went back and created a fresh sd card installation based on buster 32-bit rather than bullseye. The effect is still the same, and the same device format option on the EX5R tends to send different amounts of bytes which I do agree is a weird non-determinsm:

```
ez 03 11:35:48 raspberrypi RASCSI[470]: [2021-12-03 11:35:48.530] [trace] virtual void SCSIDEV::Receive() length is 512
Dez 03 11:35:51 raspberrypi RASCSI[470]: [2021-12-03 11:35:51.889] [error] virtual void SCSIDEV::Receive() Not able to receive 512 data, only received 510. Going to error
Dez 03 11:35:51 raspberrypi RASCSI[470]: [2021-12-03 11:35:51.889] [trace] virtual void SCSIDEV::Error(ERROR_CODES::sense_key, ERROR_CODES::asc) Error (to status phase)
```

Those ASIB-1 SCSI modules from Yamaha are known to be really slow (like almost floppy-disk slow) and weird. I think it's quite possible that there are small deviations from the standard SCSI implementation although usually standard MO drives or harddisks did work. I have no knowledge of the SCSI protocol whatsoever, but I do wonder though whether this is maybe some kind of sparse write where the missing bytes have to be filled up with zeros by the receiver or something like RASCSI giving up too quickly waiting for the data for the slow yamaha scsi module (as opposed to old mechanical SCSI hard disks) or something like that. I may know someone who has worked on a SCSI software for the Yamaha A-samplers. Those devices use the same SCSI module. Maybe he knows something about possible quirks of the Yamaha SCSI implementation. I'll try to get in touch with him.

Also I'll try to dump my scsi2sd partition to an image and try whether reading that disk works.

uweseimet, 2021-12-03:
@bzeiss No, there is nothing like a sparse write. All bytes have to be transferred. Also, slow devices don't matter because there is a handshake based on SCSI signals. As long as a transfer does not take too long and a device explicitly reports a timeout (which is not the case here) speed does not matter.
All in all from what we know by now there appears to be an issue with the handling of SCSi bus signals, resulting in incomplete transfers. Can you completely recompile RaSCSI after setting the DEBUG environment variable to 1, please? There are some assertions in the RaSCSI code that point out unexpected bus signals.

bzeiss, 2021-12-03:
TL;DR:
It started to work once I started writing new images to the sd card using dd or fill them up with shred. No bug in my opinion.

Long story:

Well this is weird: I compiled with the DEBUG flag and before taking a closer look a new logs, I did what I wrote above and copied back an image of my mo drive of the scsi2sd sd card (using dd, blocksize+sector count by looking that up in the scsi2sd xml file). I saved this as .hdr file to the rascsi sd images folder and I could use it just fine, i.e., read the files on it, even save them back and now comes the weird part: even formating that image worked without the problems above. 

This caused me to double check whether .hdr files really would not work and I did the following:

- created a new .hdr file through the web interface of size 500mb which could then be formatted fine. 
- Created a new 500MB harddisk --> formatting fine. 
- Tried formatting the same harddisk image that previsouly failed --> formatting still fails. 
- Created a new 1500MB harddisk --> formatting fine. 
- Created a new 2000MB harddisk --> formatting fine.

If I hadn't used multiple sd cards during testing, I would think that there is something wrong with the sd card. I found this to be very weird. So I decided to overwrite the non-working image with random data using shred. After that, this bad image could be formatted as well.

So I guess what actually made this work was actually writing the data (or just data?) to the sdcard.

bzeiss, 2021-12-03:
On my second bullseye image, it still doesn't work with shred or dd when creating images. I'm starting to think that the DEBUG environment variable did something.

uweseimet, 2021-12-03:
@bzeiss In debug mode RaSCSI no optimized code is generated. Are you familiar with configuring compiler options? It would be interesting to learn whether this is a GCC optimizer issuue. The regular build optimizes with -O3, the debug build uses -O0. Can you try with -O2 and -O1 by changing the respective options directly in the Makefile before recompiling everything?
In debug mode RaSCSI is also slower due to the lack of optimization, which might also have an effect. Are you running a 32 bit or a 64 bit OS?

bzeiss, 2021-12-03:
So, I'm a bit puzzled. These are the results when trying to format the drive on 64bit bullseye, Pi3. Each with multiple tries:
- DEBUG --> works
- -O3 in Makefile CFLAGS and CXXFLAGS --> scsi error
- -O2 in Makefile CFLAGS and CXXFLAGS --> scsi error
- -O1 in Makefile CFLAGS and CXXFLAGS --> scsi error
- -O0 in Makefile CFLAGS and CXXFLAGS --> works

uweseimet, 2021-12-03:
@bzeiss So there are issues as soon as optimzed code is generated. Can you try the clang compiler instead of gcc? Just replace the compiler name in the Makefile after installing clang.

bzeiss, 2021-12-04:
I have tried clang-11 and clang++-11 with -O3 and -O2. Still scsi error. Do you think it might be worth playing around with the systimer stuff? I have seen there is a gpio delay built into the receive function somewhere. Depending on how such delays are implemented, I can imagine such "empty" behavior is something where an optimizer might try to optimize too much.

uweseimet, 2021-12-04:
@bzeiss As already mentioned SCSI itself uses handshakes and is timing independent. I don't know details on timers being used in the gpio-related code, but any usage of timers may indeed be relevant here. @akuker might be familar with this code, he can probably comment.

bzeiss, 2021-12-04:
So, I have managed to create the scsi error on -O0 (which previously worked). In gpiobus.cpp, I have commented out the GPIO_DATA_SETTLING sleep in ReceiveHandShake:
```
                        // Wait until the signal line stabilizes
//                      SysTimer::SleepNsec(GPIO_DATA_SETTLING);
````
after seeing how the wait in SleepNsec takes place:
```
        // Start
        DWORD start = armtaddr[ARMT_FREERUN];

        // Loop until timer has elapsed
        while ((armtaddr[ARMT_FREERUN] - start) < diff);
```
While not reading the code in detail, it kind of looked like a typical optimization target. Probably not a valid proof for anything, but it might be something. I'll probably try to replace the sleep code with something different tomorrow.

bzeiss, 2021-12-04:
So I made it work with -O3 (essentially, with all the defaults in the Makefile). I started to mess with SleepNsec and thought that the while loop is optimized. So I have added stuff like __asm__ __volatile__ ("nop;"); or so into the while loop which supposedly let's the compiler think that there is code with side effects in there which can't be optimized. It didn't have any positive effect on the problem though. So I'm not sure that loop optimization is really the reason at this point. However, after being puzzled by that, I reverted those changes and started to mess with GPIO_DATA_SETTLING, set it to 200 first which didn't help and then tried out 1000ns which surprisingly worked. I will test this for stability and will try to find the value where it starts to work.

bzeiss, 2021-12-05:
So, I have tried to find a lower GPIO_DATA_SETTLING by going in 50ns steps. 350ns does not work. 400ns does seem work stable after a few brief tests with both a Pi3 and a Pi Zero 2 with formatting, reading and writing typical workloads. It would be interesting to know whether setting this higher has side effects of any sort.

uweseimet, 2021-12-05:
@bzeiss Does this just mean that the Pi always waits that long before it continues processing, or is this just the maximum time the Pi waits? In other words, if a signal is stable within less time, does the Pi detect that and continues processing without waiting longer? If not I would expect the transfer rates to go down.
As already indicated, somebody else,e.g. @akuker, would have to make a decision here. I'm not familiar enough with the low level signal handling.

bzeiss, 2021-12-05:
The way I read the code, it's a fixed waiting time based on an arm timer. So I suspect as well it might possibly end up influencing transfer time of devices that can make use of it. I have no idea on what 300 more nanoseconds mean in that context though. For the EX5/EX5R, it likely doesn't matter much, it's much too slow anyway. In my opinion, it would be interesting to have a comparing benchmark and additional tests using a scsi host with the highest known throughput rates. I have a pentium 4 setup with an adaptec pci scsi card running linux. I could try to do comparison benchmarks with this one and see whether there are any noticeable differences if that helps.

uweseimet, 2021-12-05:
@bzeiss What does the SCSI specification say about the time it may take for a signal to become stable? I'm sure that there is a clear statement on that somewhere in the specs. If the specification says that this time can be longer than what RaSCSI currently assumes, a change in RaSCSI would IMO be required.

uweseimet, 2021-12-05:
@bzeiss Completely different question, out of curiosity: Can it be that you know Ina Schieferdecker?

bzeiss, 2021-12-05:
Have to look at the standards and how to access them. I'm not familiar with them.

uweseimet, 2021-12-05:
@bzeiss https://www.manualslib.com/manual/1157393/Seagate-St11200.html?page=20 lists some delays. I'm not sure whether what causes your issue is the bus settle delay. If it is, it should indeed be 400 ns.

akuker, 2021-12-07:
Sorry for the delay in getting back to you guys....

A while back, when I was making changes in this area, I noticed that this value didn't match the standard. But, it seemed to work in all of my testing, so I left it alone.

> I'm sure that there is a clear statement on that somewhere in the specs. If the specification says that this time can be longer than what RaSCSI currently assumes, a change in RaSCSI would IMO be required.

@bzeiss, your testing seems to prove that this "magic number" should have really been 400ns. This matches the "Bus settle delay" value in the referenced Seagate document.

I'm curious what functional impacts this has on RaSCSI's performance and if it negatively affects other platforms. 

We've got some more testing to do to make sure we're not breaking things, but I think `GPIO_DATA_SETTLING` should be replaced with `SCSI_DELAY_BUS_SETTLE_DELAY_NS`

bzeiss, 2021-12-07:
Let me know if I can help. As written above, I have a Linux Pentium 4 system with an Adaptec AHA-2940. I could do some read/write benchmark to a RACSI hdd with both 100ns and 400ns to see whether there is any difference on a system that should probably perform as good as it will probably get for RASCSI (sysbench or gnome-disks maybe?).

akuker, 2021-12-07:
@bzeiss - It would be fantastic if you could capture some benchmarks to see what the performance hit is. I probably won't have time until this weekend :-(

bzeiss, 2021-12-09:
I'm on it, I was a little irritated by the firsts with the standard 100ns configuration with gnome-disks (which was in my opinion too slow with an average of 1.3mb/s read rate and 700kb/s write rate). That's why I setup sysbench with all kinds of benchmarks (sequential read, write, read+write and random read, write, read+write) with a test size of 1gb. I hope this will yield more useful results for a baseline.

akuker, 2021-12-10:
[image]

Here are my test results on a PowerMac G4 (Quicksilver) using SCSI Director 4.0 Pro. 

```
diff --git a/src/raspberrypi/gpiobus.h b/src/raspberrypi/gpiobus.h
index cc1dcfe..8671da2 100644
--- a/src/raspberrypi/gpiobus.h
+++ b/src/raspberrypi/gpiobus.h
@@ -433,7 +433,7 @@
 //     Constant declarations (bus control timing)
 //
 //---------------------------------------------------------------------------
-#define GPIO_DATA_SETTLING 100                 // Data bus stabilization time (ns)
+//#define GPIO_DATA_SETTLING 100                       // Data bus stabilization time (ns)
 // SCSI Bus timings taken from:
 //     https://www.staff.uni-mainz.de/tacke/scsi/SCSI2-05.html
 #define SCSI_DELAY_ARBITRATION_DELAY_NS          2400
@@ -459,6 +459,7 @@
 #define SCSI_DELAY_FAST_HOLD_TIME_NS               10
 #define SCSI_DELAY_FAST_NEGATION_PERIOD_NS         30
 
+#define GPIO_DATA_SETTLING  SCSI_DELAY_BUS_SETTLE_DELAY_NS
 // The DaynaPort SCSI Link do a short delay in the middle of transfering
 // a packet. This is the number of uS that will be delayed between the
 // header and the actual data.
```

akuker, 2021-12-10:
There doesn't seem to be a meaningful speed impact (from my results, anyway). I'm going to blame the slight variability on the non-determinism of the Pi's Linux kernel. Some numbers are higher, some lower. (And it changes when I re-run the test)

I propose that we push this change into the `develop` branch. What are your thoughts @uweseimet  and @bzeiss ?

uweseimet, 2021-12-10:
@akuker Looks as if the change does not make a relevant difference, indeed. Any chance to get higher transfer rates with an improved hardware? With my Atari TT the RaSCSI transfer speed is about 65% of a real SCSI drive. With a real drive the Mac should be even faster than the Atari.

bzeiss, 2021-12-10:
Interesting those benchmarks look a lot like what I have seen. Will post my results probably tomorrow.

akuker, 2021-12-10:
> Any chance to get higher transfer rates with an improved hardware?

I'm open to suggestions on how. Baremetal might give a performance boost, since you're not going to have the Linux Kernel overhead. Or, it might be possible to use a separate microcontroller to manage the SCSI bus transactions.

I haven't really dug into to identify what is the bottleneck/limiting agent. It's pure speculation on my part.

uweseimet, 2021-12-10:
@akuker If this was kernel overhead I would expect a Pi 4 to be much faster (and not just 100 KB/s) than the Zero WH, because the Pi 4 CPU is faster This is why I assume it is somehow hardware-related. Are there any limitations imposed by the GPIO port hardware?

bzeiss, 2021-12-11:
Weird, for me the 400ns even improve throughput a little (Pentium 4 2.something ghz, Adaptec 2940, RASCSI 2.4, Pi3b). My values are still too slow from what I gather from other benchmarks and I'll investigate on why, but still it's somewhat interesting I think.

[rascsi-16kib-400ns.log](https://github.com/akuker/RASCSI/files/7696836/rascsi-16kib-400ns.log)
[rascsi-16kib-develop.log](https://github.com/akuker/RASCSI/files/7696837/rascsi-16kib-develop.log)

I'll probably try this again on my pi4 and a 32bit os rascsi to see whether this changes anything.

uweseimet, 2021-12-11:
@bzeiss One has to be careful when comparing the benchmarks. Different benchmarks on different platforms often yield different results. Some may be fooled by (hard disk drive) caches, others by filesystem caches, and the number of contiguous blocks usually also differs.

bzeiss, 2021-12-11:
I know, though I think it should be somewhere in the same rough ballpark. In fact, with the pi4, I do get more similar results at 16kib block size. It seems it makes more of a difference than I really anticipated. In any case, I also don't think that there should be a noticeable performance penality.

[rascsi-16kib-400ns.log](https://github.com/akuker/RASCSI/files/7697679/rascsi-16kib-400ns.log)

uweseimet, 2021-12-11:
@bzeiss Looks as if your benchmark is based on files. This does not really yield meaningful results because the filesystem characteristics (cache, cluster size, allocation table management) have an impact. It's better to directly read sectors. The lower the access level, the better.
The simple benchmarks I ran with my Atari were made on the BIOS level, which is right on top of the SCSI level. No filesystem management was involved, and the Pi model did not make a big difference. My guess is that the Pi model dependency you see is caused by the filesystem management overhead.

uweseimet, 2021-12-11:
@bzeiss I think I still have to get used to the fact that the filesystem in this case on not on the Pi, but on the PC ;-). So even though a filesystem-based benchmark is not reliable for measuring transfer rates cannot explain why you see a performance dependency on the Pi model while I don't.

bzeiss, 2021-12-11:
It's a PC doing nothing else. But anyway, I think you are right that this isn't yielding much additional information than we already have with respect to this issue.

akuker, 2021-12-12:
Just as an FYI: Discord user frogofwar tested out the baremetal version of RaSCSI and reported the following reults:

> 1334k/1037k using the bare metal version on pi4

uweseimet, 2021-12-12:
@akuker What do these two values mean? One for read, one for write throughput, or one baremetal and the other with Raspberry Pi OS?

akuker, 2021-12-13:
Sorry... these are with the bare metal version of RaSCSI:
Read: 1334k / sec
Write: 1037k / sec

uweseimet, 2021-12-13:
@akuker Well, in order to make any judgment one would also need the values for the same hardware (computer) but with RaSCSI running on Raspberry Pi.

akuker, 2021-12-20:
@bzeiss - could you try out the `akuker_adjust_wire_settle_timing` branch?

bzeiss, 2021-12-20:
@bzeiss Unfortunately, I have some trouble with the testing. Given the upcoming holidays, I am no longer in my appartment where the EX5R is. I have taken a RASCSI device with me, so I could easily build and run it, but I can't verify it with the device until the end of January. 

However, I will probably receive a new toy to my holiday place tomorrow, a Yamaha A4000, with the same ASIB-1 SCSI interface as the EX5R which may or may not have  the same trouble with the 100ns delay. I can tell more about that tomorrow.

That said, I have reviewed the code change and don't see why this shouldn't work with the EX5R as you essentially just removed GPIO_DATA_SETTLING and replaced it with SCSI_DELAY_BUS_SETTLE_DELAY_NS which is exactly the 400ns which I have confirmed to be working with the EX5R and the Linux PC that I used for benchmarking. So if there are no known problems with other devices like Mac etc., I think this should be fine.

closed 2021-12-20 by uweseimet