PiSCSI/piscsi #507: RASCSI on 64bit bullseye

bzeiss, 2021-12-05:
# Info

- Which version of Pi are you using: pi zero 2 / pi4
- Which github revision of software: November 2021 release
- Which board version: fullspec 2.4
- Which computer is the RaSCSI connected to: any

# Describe the issue

I have been experimenting in making RASCSI run on 64bit bullseye. Out of the box, a few small changes seem to be necessary that I want to point out (in case they are not yet known):

1. By default, the Makefile uses 
```
CROSS_COMPILE ?= arm-linux-gnueabihf-
````
With vanilla 64bit bullseye, the cross compiler isn't installed by easy_install.sh. If it is installed using "apt install gcc-arm-linux-gnueabihf", the compilation will work until the linking starts. The linker then won't find some libraries like the protobuf library. For my purposes, I have fixed it by just uncommenting the CROSS_COMPILE line. After that, the normal gcc and g++ are used which will do fine when linking.

2. rasctl protobuf is broken:
```
rasctl -l
terminate called after throwing an instance of 'std::system_error'
  what():  Unknown error -1
Abgebrochen
```
This can be fixed by adding -pthread to the rasctl linking.
```
$(BINDIR)/$(RASCTL): $(SRC_PROTOBUF) $(OBJ_RASCTL) | $(BINDIR)
        $(CXX) $(CXXFLAGS) -o $@ $(OBJ_RASCTL) -lprotobuf -lstdc++fs -pthread
```
3. The compiler flag -DSPDLOG_FMT_EXTERNAL needs to be removed. I think this one has been changed already.

uweseimet, 2021-12-05:
@rdmark I don't have a bullseye 64 bit system yet (only 32 bit). You already have it, don't you? Can you please verify? Looks as if 32 and 64 bit bullseye differ more than the previous 32 and 64 bit buster versions.

bzeiss, 2021-12-05:
I have been googling for that specific error and it seems that error seems to be a protobuf message. See: 

https://github.com/protocolbuffers/protobuf/issues/4958

I simply tried it and it fixed the issue for me. I'm not saying this is the correct solution, it just did the job for me. Linking rasctl worked with -pthread and I can attached and detach images with it afterwards.

uweseimet, 2021-12-05:
This message looks like an uncaught C++ exception. RaSCSI uses exceptions for I/O errors, not only related to protobuf. In case you can reproduce it a stacktrace would be helpful to see the code line where this happened. Can you compile with -g, then start rascsl with gdb and run "-l"? After the crash please execute "bt" within gdb to see the stacktrace. Then we should know more about the actual cause of the error and at least fix the issue with the (most likely) uncaught exception.

uweseimet, 2021-12-05:
@bzeiss I just installed a fresh 64 bit bullseye and with the develop branch could not reproduce your compiler issue. The standard compiler being used by the Makefile "g++", not the cross compiler, because the CROSS_COMPILE environment variable is not set by default.
I can reproduce the rasctl crash, though, and will now have a look at it.

rdmark, 2021-12-05:
@uweseimet No, all my Pi:s are running 32 bit systems right now, sorry.

uweseimet, 2021-12-05:
@bzeiss After looking at the rasctl stack backtrace and dong a google search I assume that libprotobuf at least on some platforms indeed requires -lpthread. I just created a feature branch for this change.

bzeiss, 2021-12-05:
Ok great, I'll try to figure out what is happening on my system with that CROSS_COMPILE environment variable. I don't have it set by default as well, it's still getting used.

bzeiss, 2021-12-05:
So I'm on the v21.11.01 release branch:
```
pi@rascsi-ex5r:~/RASCSI $ grep CROSS_COMPILE * -R
src/raspberrypi/Makefile:##   CROSS_COMPILE=arm-linux-gnueabihf- : Specify which compiler
src/raspberrypi/Makefile:##                 ARM=x86_64 CROSS_COMPILE=x86_64-linux-gnu-cpp
src/raspberrypi/Makefile:CROSS_COMPILE ?= arm-linux-gnueabihf-
src/raspberrypi/Makefile:CC = $(CROSS_COMPILE)gcc
src/raspberrypi/Makefile:CXX = $(CROSS_COMPILE)g++
```
so we have the CROSS_COMPILE variable only in the Makefile. The last three lines mean: if the environment variable CROSS_COMPILE is not set, the Makefile will set the variable CROSS_COMPILE to "arm-linux-gnueabihf" and therefor CC and CXX will be prefixed with "arm-linux-gnueabihf" (see: http://www.gnu.org/software/make/manual/html_node/Setting.html). So unless you have an environment variable CROSS_COMPILE= (with empty contents) set in your environment, I think you should also get the following error in my opinon:

```
mkdir -p obj/fullspec
arm-linux-gnueabihf-g++ -O0 -g -Wall -DDEBUG -std=c++17 -Wno-psabi -iquote . -D_FILE_OFFSET_BITS=64 -MD -MP -DFMT_HEADER_ONLY -DCONNECT_TYPE_FULLSPEC -c rascsi.cpp -o obj/fullspec/rascsi.o
make: arm-linux-gnueabihf-g++: Datei oder Verzeichnis nicht gefunden
make: *** [Makefile:153: obj/fullspec/rascsi.o] Fehler 127
```

uweseimet, 2021-12-05:
@bzeiss Yes, you are right. I missed that on my system CROSS_COMPILE was set to empty in my standard profile, and when setting up the new OS the first thing I did was copying my profile to it. I'm sorry for that.
I don't know, though, what the right solution would be for somebody not using easyinstall, except for explicity unsetting CROSS_COMPILE. When using easyinstall on bullseye 64 bit, easyinstall can do this.
@rdmark What's your opinion on that?

bzeiss, 2021-12-05:
why not the other way round? 
Makefile:
```
CROSS_COMPILE ?=
```
So it gets initialized to no prefix if no environment variable is set. On the other hand, if someone wants to explicitly cross compile, he must set the CROSS_COMPILE prefix in his environment to his target platform. The Makefile in this case will use the environment variable for the prefix. Seems more logical to me.

uweseimet, 2021-12-05:
I'm not sure whether there is a compiler on a 32 bit Pi OS without any prefix. @rdmark Maybe you can check whether the proposed change also works for 32 bit? I agree with the proposed setup being more logical than the current one.

rdmark, 2021-12-06:
@uweseimet I tried both proposed modifications to the Makefile with success on my 32bit bullseye. No compiler warnings.

uweseimet, 2021-12-06:
@rdmark OK, I updated the issue_507 branch accordlingly

rdmark, 2021-12-06:
PR tested and merged.
@bzeiss Can you have a look at the latest develop code and confirm that your issues have been resolved now?

bzeiss, 2021-12-06:
Tried it on a fresh git clone of develop (dependent packages already installed though). The compilation uses the correct compiler now and rasctl works fine as well. So the issues I have reported are resolved.

closed 2021-12-06 by uweseimet

rdmark, 2021-12-06:
@bzeiss That's great! However, when you say that dependent package are already installed, does this mean that there are dependencies that aren't in the list on the wiki or the easyinstall script?

bzeiss, 2021-12-07:
@rdmark it's just a remark that I haven't rechecked anything regarding the packages and that I didn't retest on a vanilla installation. I can't recall that anything was missing.